ghaerr/elks #329: ELKS not setting date during boot

Mellvik, 2020-02-13:
[commit dc3baae3]
ELKS does not pick up the current date from the (physical) machine during boot. Date is set to 0 (1-1-70).

Further, the date -s command seems broken, it does not accept any reasonable variant of the format given in the usage message. See picture.
[image]

ghaerr, 2020-02-13:
@Mellvik: Thank you for your testing.

Yes, I'm aware of both these issues, and had I been able to get date to accept a new date, I would have been able to look further into FAT date issues in my previous commits.

Mellvik, 2020-02-17:
Sys_utils/clock.c does clock initalization (and maintenance if needed). It has been disabled because it does direct port I/O (according to the makefile). It also needs some rework to fit into the new compiler regime. 
If this is not on anyone's agenda I'd like to take stab at it. 
IMHO all that is needed for now is to set the daytime from HW @ system startup, no running maintenance (adjustments) involving adjtime and cron. Opinions solicited.

-- Mellvik

jbruchon, 2020-02-23:
Fixed by #364

closed 2020-02-23 by jbruchon