ghaerr/elks #334: elksemu fails with "waitpid failed" error on some Linuxes?

tkchia, 2020-02-15:
Reported by @georgp24 (https://github.com/tkchia/gcc-ia16/issues/50#issuecomment-586609217):

> I also tried elksemu as you mentioned.
> `elksemu/elksemu ./helloa`
> This did just output:
> `waitpid failed`

tkchia, 2020-02-15:
Hello @georgp24, may I know which version of which Linux distribution you are using?

mfld-fr, 2020-02-15:
Quick test:
```
mfld@home /mnt/data/home/mfld/advantech/elks-master/target $ ../elksemu/elksemu bin/sh
$ echo 'Hello @tkchia!'
Hello @tkchia!
$ 
```
Config:
```
Linux home 4.4.87-gentoo-r1 #6 SMP Mon Sep 3 20:43:34 CEST 2018 x86_64 Intel(R) Xeon(R) CPU E5450 @ 3.00GHz GenuineIntel GNU/Linux
```

georgp24, 2020-02-15:
My Linux distro is:
``` openSUSE 13.2 (x86_64)
VERSION = 13.2 ```

georgp24, 2020-02-16:
I get:
```
georg@linux-lj8b:~/elks/target> ../elksemu/elksemu bin/sh
waitpid failed
georg@linux-lj8b:~/elks/target>
```

tkchia, 2020-02-16:
Hello @georgp24,

OK, I will see if I can somehow set up an openSUSE box and reproduce the problem...

Thank you!

georgp24, 2020-02-16:
Thank you tkchia, however, the current version of SUSE is 15.1 and not 13.2 which I have. Also, I ought to download quite a number of updates. This version worked fine for all my purposes, so I did not bother to update it.
It may be that the current version of openSUSE does not show this problem.

georgp24, 2020-02-16:
I updated my SUSE version to the latest one, 15.1. Run make clean and make all. Now elksemu works! Thank you.

tkchia, 2020-02-16:
Hello @georgp24,

Thanks for your report!  I guess that means I can close this issue then.  (Though I am still curious why it was failing on openSUSE 13.2.)

Thank you!

closed 2020-02-16 by tkchia

mfld-fr, 2020-02-16:
Thank you @tkchia for your support on ELKSEMU !

tkchia, 2020-02-17:
Hello @georgp24, hello @mfld-fr,

I have reproduced the problem --- in fact, this problem _plus_ a "Segmentation fault" problem which was masked by the "waitpid failed" problem --- on an openSUSE 13.2 VM.  I have submitted a pull request to fix both problems (https://github.com/elks-org/elks/pull/346).

Thank you!