ghaerr/elks #341: Thoughts on future

ghaerr, 2020-02-16:
After thinking about @mfld-fr’s comment on ‘better advertising’ the following came to mind.

I’m big on better advertising also, as a means to get more people interested in using or developing for the ELKS project. ELKS will grow with more user involvement, giving increased project visibility. Over the past few years, ELKS has steadily returned from times of not much activity beforehand. Things are now proceeding much more quickly, and ELKS is working quite well, for education as well as use on user’s 8086 systems.

Our major and minor numbering scheme could reflect our progress by incrementing the major number on major accomplishments, then adding minor releases with planned incremental improvements when they happen.  There’s so much that has been added in this release, it would be a shame to not use this opportunity to showcase a major ELKS release. To wit:

Major Enhancements in 0.3.0
-	Switch to modern gcc-ia16 compiler

Major End User Enhancements in 0.4.0
-	FAT filesystem dramatically improved
-	FAT boot
-	Prebuilt disk images for all supported MINIX and FAT sizes
-	Nano-X graphics support
-	Enhanced busyelks commands available

Major Developer Enhancements in 0.4.0
-	Continuous Integration support
-	Build as normal user
-	OSX supported
-	Standard filesystem tool support with external or ELKS own tools
-	ELKSEMU working
-	ELKS PB allowing fast disk type detection with no kernel probing
-	Probe booter for device debugging

Minor Developer Enhancements in 0.4.0
-	Lots of libc and gcc-ia16 improvements
-	Configurable process dynamic data size
-	Improved boot loaders
-	Lots of code cleanup

Possible candidates for 0.4.x releases, incremented when completed:

End User:
-	Disk sets for different ELKS uses (Game, BusyElks, Developer, Debug, Networking, etc)
-	Package manager/Config integration by tag specification
-	Kernel probing design and implementation for wider disk support
-	Support for SSD FAT devices

Developer:
-	Unified boot loader and SYSINIT/INITSEG cleanup
-	FAT filesystem completion (FAT32, backup FAT table, etc)

This is just a first draft, would be happy work more on it should there be interest.

mfld-fr, 2020-02-17:
Mmm... I agree that the next release will contain more than bug fixes. But I also think that, even after several weeks of work, and except the 'package manager', we 'only' improved existing things that were already drafted, or we 'only' filled some implementation gaps.

I would rather increment to 0.4.0 whenever something tremendous is introduced, like the move from DEV86 to GCC-IA16 that really changed the game.

mfld-fr, 2020-02-17:
Other points related to next release description merged to #307.

closed 2020-02-17 by mfld-fr