ghaerr/elks #342: ls -li (list inode #) broken

ghaerr, 2020-02-17:
I finally had some time to track down the broken `ls -li` command to list a file's inode number.

The problem is that elks/include/arch/stat.h uses a variable-sized `ino_t` type that changes size based on CONFIG_32BIT_INODES. This of course causes problems with applications because the `stat` structure size and `st_ino` field are different sizes, based on the kernel compilation, which is no good.

It's easily repaired by fixing the `st_ino` size in stat.h and replacing ` t_ino` with`u_ino_t`, which is always 32 bits long. I've checked the kernel's readdir function and this will work in both CONFIG_32BIT_INODE cases.

I would submit a patch, but notice that include/arch/stat.h is also present in the `gcc-ia16` build chain, and am not sure whether stat.h comes from there, or whether both locations of the same file need updating. @tkchia, can you comment on whether the toolchain also needs changed? If so, then I suggest that you submit a PR with that and the elks/include/arch/stat.h updated, otherwise I'll do it.

The exact change should be `u_ino_t  st_ino` without ifdefs in stat.h.

mfld-fr, 2020-02-17:
What do you mean by 'stat.h' is in the build chain ?

ghaerr, 2020-02-17:
I mean that ‘grep -r’ showed a copy of stat.h underneath the cross directory also.

mfld-fr, 2020-02-17:
Yes, but in LIBC or in GCC ?

ghaerr, 2020-02-17:
Both, I guess. I'm not sure. I don't understand what copies of stat.h are doing under the `cross` directory:

```
Gregs-MBP2:elks-gh greg$ grep -r ino_t .
./libc/misc/getcwd.c:static ino_t root_ino;
./libc/misc/getcwd.c:   ino_t this_ino;
./libc/misc/getcwd.c:ino_t this_ino;
./libc/misc/getcwd.c:   int slow_search = (sizeof(ino_t) != sizeof(d->d_ino));
./cross/ia16-elf/lib/elkslibc/include/linuxmt/msdos_fs_sb.h:	ino_t dev_ino;               /* "/dev" ino */
./cross/ia16-elf/lib/elkslibc/include/linuxmt/types.h:typedef __u32			u_ino_t;
./cross/ia16-elf/lib/elkslibc/include/linuxmt/types.h:typedef __u32			ino_t;
./cross/ia16-elf/lib/elkslibc/include/linuxmt/types.h:typedef __u16			ino_t;
./cross/ia16-elf/lib/elkslibc/include/linuxmt/types.h:    ino_t			f_tinode;
./cross/ia16-elf/lib/elkslibc/include/linuxmt/msdos_fs.h:	ino_t ino;		       /* ino for the file */
./cross/ia16-elf/lib/elkslibc/include/linuxmt/msdos_fs.h:	ino_t ino; /* inode number. */
./cross/ia16-elf/lib/elkslibc/include/linuxmt/msdos_fs.h:extern ino_t msdos_get_entry(struct inode *dir,loff_t *pos,struct buffer_head **bh,
./cross/ia16-elf/lib/elkslibc/include/linuxmt/msdos_fs.h:    struct msdos_dir_entry **res_de,ino_t *ino);
./cross/ia16-elf/lib/elkslibc/include/linuxmt/msdos_fs.h:extern ino_t msdos_parent_ino(struct inode *dir,int locked);
./cross/ia16-elf/lib/elkslibc/include/linuxmt/fs.h:    ino_t			i_ino;
./cross/ia16-elf/lib/elkslibc/include/linuxmt/fs.h:typedef int (*filldir_t) (char *, char *, size_t, off_t, ino_t);
./cross/ia16-elf/lib/elkslibc/include/linuxmt/fs.h:extern struct inode *__iget(struct super_block *,ino_t);
./cross/ia16-elf/lib/elkslibc/include/arch/stat.h:    u_ino_t	st_ino;
```

mfld-fr, 2020-02-17:
So only on ELKS side, not on GCC. You can see it in cross because @tkchia made the ELKS LIBC 'multilib' capable, so installing each variant of LIBC in cross. He would explain it better than me.

tkchia, 2020-02-17:
Hello @ghaerr, hello @mfld-fr,

Well, it is actually a bit simpler than that: I arranged to "install" the `elks-libc` runtime library and include files in the `cross/` directory.  This way it would be easier for one to build ELKS `a.out` programs --- given the `. ./env.sh` environment --- without having to specify a bunch of `-I`..., `-L`..., and `-T`... options by hand.  The is done by an `install` rule in `libc/Makefile`; also see the `all` rule in the main `Makefile`.

(If I remember correctly, this was in response to someone's issue report, but I cannot recall offhand which...)

So the way to propagate any changes in the `*.h` files to the `cross/` directory is simply to "reinstall" the C library. :-)

Thank you!

ghaerr, 2020-02-17:
@tkchia, thanks for the great explanation, I understand now.

This issue can be closed.

tkchia, 2020-02-17:
Hello @ghaerr,

Thanks!  I do also hope to add "multilib" support for `elks-libc`, some time in the future.  Specifically, I would like to be able to build variants of the C library for different ABIs, e.g. passing function parameters in registers vs. through stack memory.  But again, that is for the future.

Thank you!

mfld-fr, 2020-02-17:
Reminder : an issue can be closed only after the PR fixing it has been merged, not before.

Closing this one as I just fixed in commits 12f1995 and 32f2581 with POSIX compliance.

closed 2020-02-17 by mfld-fr