ghaerr/elks #347: [fs] Inefficient 'dirent' copy to user space in 'readdir'

mfld-fr, 2020-02-18:
Currently the `dirent` structure is copied from the kernel to the user data segment using one 'put' for each member. Bad impact on `ls` performance.

Also an opportunity to clean duplicated `dirent` declaration in kernel, and wrap the POSIX `ino_t` print format in user space.

See also #164 for previous reworking on this part.

ghaerr, 2020-02-18:
Completely disagree with the need for this "fix". There are far bigger issues to fix than worrying about changing a few kernel-to-usermode put calls, and the `dirent` structure shouldn't be opaque and then uncovered in userspace.

Concerned for `ls` performance? Have you looked at `ls`?? It calls `realloc` for *every* filename! It uses FAR more time possibly reallocating memory, calling system `sbrk`, then running memcpy, for every file.

And it calls another `realloc` for two bytes JUST to add a single byte of output for `ls -F`. Talk about terrible performance. You should consider rewriting `ls` if you're actually concerned about `ls` speed, not rewrite this kernel section.

mfld-fr, 2020-02-19:
Print format encapsulated in `PRIino` macro, as C99 does for the portable integer types, and to keep the same logic as that standard. See commit 647ab5d0.

mfld-fr, 2020-02-26:
Will optimize the `dirent` copy in my own fork. Closing this one.

closed 2020-02-26 by mfld-fr