ghaerr/elks #354: Revert disputed commits to allow getting back to work

ghaerr, 2020-02-21:
I don't now how to open a PR to revert a commit, so I am opening a this issue.

The three commits listed below are what caused the decision of myself (@ghaerr) and @tkchia to stop continued contributions to this repository, especially since they were not committed through the normal pull release discussion process. As such, they should be reverted, after which @mfld-fr can open a PR and go through the standard process if desired.

[image]

jbruchon, 2020-02-21:
In your fork, you `git branch`, then `git revert commit1 commit2 ...` with each commit listed to be reverted, which will produce a new commit that reverts them. Then you'd issue a PR for that branch. After the PR is merged into master, you can delete the branch from your local repo.

ghaerr, 2020-02-21:
Thank you. I'll wait and make sure @tkchia supports this approach and then submit a PR.

jbruchon, 2020-02-21:
While you and @tkchia are looking at this, I'd also like the 2 outstanding pull requests reviewed. There are a lot of patches underneath those requests and they should be discussed and potentially merged as soon as possible.

ghaerr, 2020-02-21:
#332 is definitely on hold until after this release, considered too untested to add at this point.

#350 appears quite messed up: it contains lots of other commits, when it's only supposed to contain the 3rd set of additions for BusyELKS from Marcin. Notice that it also pulled over the big disputed conversation from one of the 3 disputed commits - that is definitely not supposed to be part of the commit. I am only guessing that perhaps it got messed up with the recent HEAD readjustments. The 3rd set of additions for BusyELKS would be nice to add, but we need to find a clean copy first.

ghaerr, 2020-02-21:
I now see the problem with #350 is that it's a PR from @mfld-fr 's marcin branch, which he's apparently also using to rebuild his fork when the HEAD changed, so it now includes other stuff as well.

mfld-fr, 2020-02-21:
You can delete the 'marcin' branch, I am waiting first for the situation to stabilize, then I will repush Marcin's patches.

tkchia, 2020-02-21:
Hello @ghaerr,

> I'll wait and make sure @tkchia supports this approach and then submit a PR.

Yes, I am fine with this.

I will also try to jot down my observations on the `ino_t` problem soon (it is a _little_ bit more complicated than I initially thought).

Thank you!

ghaerr, 2020-02-21:
> > I'll wait and make sure @tkchia supports this approach and then submit a PR.
> 
> Yes, I am fine with this.

Great, I'll get a PR ready. I'm a little tight on time this weekend, but will get to it least by Monday.

closed 2020-02-21 by ghaerr