ghaerr/elks #355: Getopt broken

Mellvik, 2020-02-21:
getopt() is currently broken, causing programs like du, grep, miniterm and a few others (like clock, which led me to the issue in the first place) to fail (hang) if options are specified.
The programs using getopt are few, even in busyelks. So before spending time fixing it (probably an IA16 issue), are there any pro/con opinions on using getopt? Fix or ditch?

--Mellvik

marcin-laszewski, 2020-02-21:
Fix. 'getopt' is widely used. It is easier to adopt new programs with it.

Mellvik, 2020-02-21:
OK; it turns out that ./elks/arch/i86/tools/getopt.c works, while the one in use today, in ./libc/misc/getopt.c does not. The one that works is also the simplest. I suggest moving the working version to the libc tree.
Tested: grep, du, clock

ghaerr, 2020-02-21:
@Mellvik, nice find, thank you.

I am quite aware of the many incompatibilities in the many getopt0 implementations, I found a very small getopt() for use in the `mfs` tool that would work on both OSX and Linux. Of course, we're talking about use in ELKS runtime here. 

Good gosh, another tools directory I hadn't even realized existed! I wonder whether the tools directories should be combined as yet another project.

I'd like to find time to take a quick look at now the 3 getopt implementations that we have in ELKS, so as to see whether just one might be able to work, and unlitter the situation a bit more.

jbruchon, 2020-02-21:
Anything that simplifies the hierarchy is supported by me.

jbruchon, 2020-02-23:
Fixed by #364

closed 2020-02-23 by jbruchon