ghaerr/elks #356: Floppy probing very slow [enhancement]

Mellvik, 2020-02-21:
The probing process (in the floppy driver?) to determine actual density of floppies is very slow on real hardware - 5-8 seconds with multiple (5?) full length seeks across the disk. 

Should find a faster way to do this.

ghaerr, 2020-03-02:
Hello @tkchia,

It looks like the probing routine in elks/arch/i86/drivers/block/doshd.c, uses a 'read_sector' routine that "probes" by trying to read a sector 5 times, before moving to try another sector. I'm thinking that read_sector should be passed a retries parameter (1 in the probe cases, MAX_ERR in the normal read cases) to speed the probe up considerably. Also, perhaps 'reset_bioshd' should not be called between probe failures, only for real read sector retries or when retrying the entire probe sequence.

Also, when probing for cylinder, since there's only two cases, 40 and 80, the routine will always run worse case for 40 sector floppies. Perhaps it should just try 80 and if fail assume 40.

It would seem that by using both these methods, the HW probing could be increased by quite a bit, with little worries about it working less reliably.

If you would prefer, I could submit a PR for this, although I don't have actual PC hardware to test with.

Mellvik, 2020-03-02:
Thank you @ghaerr for looking into this.
I have on  my list to look at how freeDos is doing this. Neither DOS nor Freedos do searches when probing the presence (and possibly density), as far as I can tell from physical observation, both stay at track zero.

I’m away from my ‘HW-lab’ till late next week, can do more physical testing then.

—Mellvik

ghaerr, 2020-03-18:
Hello @Mellvik,

Did you mention this issue is now fixed? If so, please close this issue.

Mellvik, 2020-03-19:
Yes, probing is now acceptable. Closing.

closed 2020-03-19 by Mellvik