ghaerr/elks #358: Distinguishing between ino_t and ... ino_t

tkchia, 2020-02-21:
(Follow-up to https://github.com/jbruchon/elks/issues/354 .)  Currently the type `ino_t` is used in ELKS to represent file serial numbers, both
  * for the kernel's internal processing, e.g. in `struct inode`, and
  * to communicate file serial numbers to userland, e.g. in `struct stat` and `struct dirent`.

(It is also used in documentation --- in `Documentation/text/minix_fs.txt` --- but this is a more minor problem.)

With the default ELKS build settings --- which only does 16-bit inodes --- I can write a program which reads the file's owner id, which will compile to something like this:
```
...
0000003B  8D46E2            lea ax,[bp-0x1e]    ; stat("/dev/null", &sb);
0000003E  50                push ax
0000003F  B80400            mov ax,0x4
00000042  50                push ax
00000043  E87E01            call word 0x1c4
00000046  FF76EA            push word [bp-0x16] ; printf("%#o\n", (unsigned)sb.st_uid);
...
000001C4  B81200            mov ax,0x12
000001C7  E93101            jmp word 0x2fb
...
000002E4  CD80              int 0x80
...
000002F4  89E3              mov bx,sp
000002F6  8B5F02            mov bx,[bx+0x2]
000002F9  EBE9              jmp short 0x2e4
```
The problem is that we should be able to run the same _binary_ program on another ELKS system, perhaps one which supports 32-bit inodes, the program should still work.  I.e. if the program invokes syscall `0x12`, **it should reasonably expect that the `.st_uid` value will _still_ be at `[bp-0x16]` on the second ELKS system, not at some other offset into the structure**.

tkchia, 2020-02-21:
So to make the {user ↔ kernel} interface stable enough, one thing that will definitely be needed is a way to distinguish between the kernel's internal inode number type, and the inode number type(s) that it uses to interface with userland.

One possibility is to follow the approach used in Linux.  Namely, define `ino_t` differently in C headers used by the kernel, and C headers used by userland code.  (In particular: Linux 5.5.5 in particular defines `ino_t` to `__kernel_ino_t` for the kernel.  GLibC does not use `__kernel_ino_t` to define `ino_t`.)

This means that from the kernel's viewpoint, a `struct stat` may have an `.st_ino` field that is _not_ of `ino_t` type, but some other type (e.g. `uint16_t` or `uint32_t`) actually expected by the user process.

There may be other possibilities.

tkchia, 2020-02-21:
(From the user's viewpoint, the type of the inode number _must_ be named `ino_t`, [per POSIX](https://pubs.opengroup.org/onlinepubs/009695399/basedefs/sys/types.h.html).  However, this `ino_t` can (and perhaps should) be different from the kernel's internal `ino_t`.)

tkchia, 2020-02-21:
Another thing that will be needed is a way to extend the {user ↔ kernel} interface to allow for 32-bit inode numbers.  Some possibilities:
  * Modify the current `stat` syscall (#18) etc. to always output 32-bit `ino_t`'s --- this causes a one-time compatibility breakage --- and let it stay that way.
  * Split `stat` into two (or more) syscalls, each dealing with a different variant of `struct stat`.
  * Add some sort of "flags" parameter to the `stat` syscall, to allow a user process to tell the kernel what kind of structure the user expects.
  * Some combination of the above.

Likewise for `fstat`, `readdir`, etc.

Let me know what you think of all this.  Thank you!

ghaerr, 2020-02-21:
> One possibility is to follow the approach used in Linux. Namely, define `ino_t` differently in C headers used by the kernel, and C headers used by userland code. (In particular: Linux 5.5.5 defines `ino_t` to `__kernel_ino_t` for the kernel. GLibC does not use `__kernel_ino_t` to define `ino_t`.)

I agree with this approach, not necessarily with the naming though?

> This means that from the kernel's viewpoint, a `struct stat` may have an `.st_ino` field that is _not_ of `ino_t` type, but some other type (e.g. `uint16_t` or `uint32_t`) actually expected by the user process.

I think this was the intent of the (soon-to-be-returning) `u_ino_t` type, which was/is in the outward facing stat.h struct. I'm guessing that the 'u' in the type stands for user.

ghaerr, 2020-02-21:
> Another thing that will be needed is a way to extend the {user ↔ kernel} interface to allow for 32-bit inode numbers. Some possibilities:

Yes, this release should take the one-time step to widen the ABI, to pave the path to work well with FAT32 filesystems, etc.

> * Modify the current `stat` syscall (#18) etc. to always output 32-bit `ino_t`'s --- this causes a one-time compatibility breakage --- and let it stay that way.

This was my implementation in the PR I submitted to fix `ls` that ended up getting rewritten. I believe this simple solution is the best approach for ELKS.

> * Split `stat` into two (or more) syscalls, each dealing with a different variant of `struct stat`.
> * Add some sort of "flags" parameter to the `stat` syscall, to allow a user process to tell the kernel what kind of structure the user expects.

These are overly complicated, and only serve to increase code size while adding little value. ELKS only need for a wider external inode is the FAT32 filesystem, that's it.

> Likewise for `fstat`, `readdir`, etc.

Yes. The `readdir` call currently uses a `long` which would get replaced with the newly named type.

jbruchon, 2020-02-21:
I do want to remind everyone that while we have actual users, the user base is extremely small and "in the loop" on the project. Concerns about existing built software not working with incompatible ABI/API changes are not very important, in my current estimation. Better to break things in the 0.x versions and get them right than to start piling up legacy compatibility stuff. If a crucial structure needs to be modified to avoid exponential headaches down the line, I prefer to see the changes and pain happen immediately, assuming the changes make sense and are as standards-compliant (SUS, POSIX, etc.) as feasible.

It might also be prudent to open the discussion on future support of other platforms than 8086. Is that something which we should consider? What about 68000, 6809, 65816, and such? If those are a possible long-term set of goal targets, we should at least consider what happens with those other platforms if we built for them. I'm perfectly happy with letting FUZIX do all that, if the consensus is 8086 only, but there are a ton of microcontrollers and old computer systems out there that can't ever run Linux. (Should I make an enhancement issue for discussing this?)

ghaerr, 2020-02-21:
>  (Should I make an enhancement issue for discussing this?)

Not yet, IMO. ELKS is just barely getting to the point where it works, from a standpoint of running working applications. There is tons of cleanup needed in the application tree, and I want to consider both smaller (busyElks) and larger (possible wholesale replacement of many commands with modern `binutils-ia16`) versions. We need to get through at least a couple more big version updates before considering new processor types, or shared libraries, IMO.

mfld-fr, 2020-02-22:
@tkchia : Nice to see that after some more thinking, you are coming to the same finding I had : to ensure ABI stability, you would have to maintain 2 different set of structures, one in the kernel space, and another in the user space, and to copy between them when crossing the boundary. At least if you want to keep the kernel structures optimized to save space. You could ever put a 32-bit member everywhere also in kernel space, but the impact would be multiplied by the number of structure instances that contain that member. I let you counting them in the sources as I did...

On the side, as you should be POSIX compliant for application source portability (API stability), the structures on the user side should use the POSIX names, whatever ABI stability or not. It won't save you anyway from lack of POSIX support in C99...

I made the choice to drop the ABI constraint, first to keep ELKS most simple but consistent for educational & training purpose, and second to keep ELKS small and most efficient for running on old HW, because both are the actual ELKS usage today. Never used for production.

I am therefore still thinking that keeping ABI stability is not a real need. Users either recompile the whole sources, or take the whole image. I never saw a report since 2015 where a user executes applications on another medium and built with another kernel configuration.

It looks like you took the direction of keeping that design constraint, and so accepting the additional workload it implies. Well... I wish you then to be able to sustain it for a long time, because I won't support it.

One more word to @ghaerr : even I had a rough communication as a maintainer to enforce what I think are the right project goals, I will never accept anyone, even with strong disagreement, to be injurious like you did in so many latest posts, up to the point to tell me I could 'lost my mind'.

You can attack my thinking, your can attack my position, but never my being. I never criticized what you are, just what you did or explained you want to do. I refuse to collaborate with people with such wrong behavior, so I will not respond to your latest questions until I got your apologies.

tkchia, 2020-02-22:
Hello @mfld-fr,

> I am therefore still thinking that keeping ABI stability is not a real need. Users either recompile the whole sources, or take the whole image. I never saw a report since 2015 where a user executes applications on another medium and built with another kernel configuration.

Well, actually --- you could have asked!  Just because people are not doing _A_, it does not mean that people will never _want_ to do _A_.  (Because, indeed, there are many other possible reasons for not doing _A_.)

So, there is no need to [assume](https://www.xkcd.com/1339/) that nobody would ever, ever _want_ to build an ELKS program on one system and run it on someone else's ELKS system.  You could simply directly ask the community, "hey, do you think you would want to build an ELKS program on one system and have it run elsewhere?"  And you would have a direct answer to your question.

Really, it is not about you, it is not about me, not about @ghaerr, and probably not about @jbruchon either.  I think, as long as this project is to remain a community project, then it will have to be aligned with what the community as a whole wants --- otherwise it ceases to be a community project.

And I think the best way to know what the community wants is to ask, and listen carefully.  And if I do something erroneous, and I do not listen, I _want_ the community to call me out on it.  That is the correct way.

Thank you!

mfld-fr, 2020-02-22:
@tkchia : Okay for a 'community' driven project (I am still quite doubtful about the 'community')... let me wait and watch what happens in the next months, in both terms of user experience and project maintenance... I would then revise my opinion on what really happened.

ghaerr, 2020-02-22:
> I refuse to collaborate with people with such dreadful behavior, so I will not respond to your latest questions until I got your apologies.

First: I apologize. Responding in such a manner will not happen again.

I will admit to being in extreme frustration, by seeing through the process of direct commits and sole write access, that my opinion and that of several others in the community, was disregarded and there was nothing anyone could do about it. After then seeing that the discussed negative consequences of that decision were followed by more direct commits, I suppose I lost my mind. Nonetheless, I should have handled it differently.

> @tkchia : Nice to see that after some more thinking, you came to the same finding I had : to ensure ABI stability, you would have to maintain 2 different set of structures, one in the kernel space, and another in the user space, and to copy between them when crossing the boundary.

The idea of perhaps copying between different sized structures to ensure ABI stability is not the same finding as coding an external interface with a varying size type, which is never an ABI. That could be called an application 'source' interface, I suppose.

> I am therefore still thinking that keeping ABI stability is not a real need. Users either recompile the whole sources, or take the whole image. I never saw a report since 2015 where a user executes applications on another medium and built with another kernel configuration.

Since the cost of maintaining this particular ABI is extremely low versus the gains earned and future application portability increased, why force the decision of recompilation and also introduce likely application portability problems?

> I made the choice to drop the ABI constraint, first to keep ELKS most simple but consistent for educational & training purpose, and second to keep ELKS small and most efficient for running on old HW, because both are the actual ELKS usage today. Never used for production.

Neither my nor your opinion on the matter of the ABI discussion measurably changes the efficiency of ELKS on old or any HW. 

With regards to production use, or educational and training, my opinion is that ELKS currently works poorly in all three cases: there are just too many bugs, the source code and many constructs are too complicated, there are too many ifdefs, and the system, but most particularly the command set, is so buggy that it is barely usable. It needs a lot of work, and we are doing that work. I believe source code readability and simplicity are very important in achieving the goal of education and training 'efficiency'.

> It looks like you took the direction of keeping that design constraint, and so accepting the additional workload it implies. Well... I wish you then to be able to sustain it for a long time, because I won't support it.

Can you let us know what additional workload is actually implied here, I don't see it. I actually believe that reversing that decision implies additional workload.

ghaerr, 2020-02-22:
> (From the user's viewpoint, the type of the inode number _must_ be named `ino_t`, [per POSIX](https://pubs.opengroup.org/onlinepubs/009695399/basedefs/sys/types.h.html). However, this `ino_t` can (and perhaps should) be different from the kernel's internal `ino_t`.)

@tkchia: I missed this before writing my reply yesterday about `u_ino_t`. 

> This means that from the kernel's viewpoint, a struct stat may have an .st_ino field that is not of ino_t type, but some other type (e.g. uint16_t or uint32_t) actually expected by the user process.

The two above statements of yours now seem in possible conflict, does `stat` need to contain an `ino_t` type or not? 

If so, then an ABI dictates that kernel `ino_t` be named something else. If not, then `u_ino_t`, or perhaps `uint32_t` would work.

tkchia, 2020-02-22:
Hello @ghaerr,

> does [`struct`] `stat` need to contain an `ino_t` type or not?

Yes --- but only from the userland viewpoint.  If an application program includes `<sys/stat.h>`, then the `.st_ino` field of a `struct stat` must be `ino_t` --- that is all.

(So the kernel code internally is free to give a different name to the `.st_ino` field's type, if need be, as long as it ends up being the same binary type.  As far as I can tell, this is the approach used by Linux.)

> If so, the n an ABI dictates that kernel `ino_t` be named something else.

I guess that is another possible way, though it is not strictly necessary for POSIX compatibility.  (This solution means that all the possible kernel-internal uses of `ino_t` will have to be replaced by, say, `k_ino_t`.)

Thank you!

tkchia, 2020-02-27:
@ghaerr's merged commits (https://github.com/jbruchon/elks/pull/382) have fixed this issue.  Thank you all!

closed 2020-02-27 by tkchia