ghaerr/elks #359: Set datetime from RTC

Mellvik, 2020-02-21:
Now that getopt works, the modified clock.c also works. I haven't made it into the github world yet, the modified source is here:
https://github.com/Mellvik/elks/blob/master/elkscmd/sys_utils/clock.c

Local branch commit: https://github.com/Mellvik/elks/commit/eea20594680956def76da5ad0e65f257569407f5

Also need to add `clock -s` to `/etc/rc.d/rc.sys`
Comments welcome, is there a good reason to keep BCC backwards compatibility (asm)?

tkchia, 2020-02-21:
Hello @Mellvik,

> is there a good reason to keep BCC backwards compatibility (asm)?

None that I know of, but I will see what our other contributors say.  Thank you!

ghaerr, 2020-02-21:
@Mellvik: Thank you for getting this all working! I'll take a look at your source and update rc.sys accordingly. I may need until Monday to do this given my time restraints this weekend.

@tkchia: The ifdef BCC_ASM is littered throughout lots of libc C files, and I'm not surprised to see an implementation of `inb` stuffed into an application's C source. Yes, IMO opinion these should all be cleaned up (deleted from C sources, and probably re-ported to `as-ia16` format in libc).

ghaerr, 2020-02-22:
>  I haven't made it into the github world yet, the modified source is here:
> https://github.com/Mellvik/elks/blob/master/elkscmd/sys_utils/clock.c

@Mellvik: Your ability to find and fix user land bugs is quite helpful!

If you create an account on GitHub and fork this repository (which copies to your account), then after your local branch changes are commited, if you 'git push' your branch (to your GitHub repository), you will see that back on this repository under Pull Requests, GitHub automatically sees you have a branch push and prompts you to open a PR with those changes.

I've created a PR for your work on this, but try this method on your next bug fix if you like.

Mellvik, 2020-02-23:
Thanks @ghaerr - I'll do that.
This issue should probably then ble closed.
Suggestion: When the date is set in rc.sys, send the date to the console. It's not only a tradition, it's actually useful.

tkchia, 2020-02-23:
Hello @ghaerr, hello @Mellvik,

I have submitted a pull request (https://github.com/jbruchon/elks/pull/374) to slightly clean up the `/bin/clock` code.  I hope to further tidy up the general ELKS code structure in the next few weeks (or months).

Thank you!

closed 2020-02-23 by jbruchon