ghaerr/elks #360: Shared libc code between kernel and user land

jbruchon, 2020-02-21:
I would really like to see at least the libc code shared among all processes in some way. Perhaps further out in development, but one of the driving factors behind me trying to write BusyELKS was how every single executable had its own copy of libc functions, and some programs were being written with write() purely to avoid the weight of printf().

If there were some way of the kernel providing libc and linking executables to the internal libc when loaded, that would be fantastic. I know that the segment limits become a major problem with such a thing. I suggested elsewhere that perhaps the kernel should take some pages from microkernels like Minix and split off some non-critical parts from the core such that they'll get their own segments. If the kernel was split into such components, libc could easily be one of them, and the kernel would simply link itself to the libc that is packed with it once the early boot code hands over control.

tkchia, 2020-02-23:
Hello @jbruchon,

I am curious to know if there are any existing systems that try to do such a thing.  How did they do it, and what were the kind of issues that faced?  I know Linux has a `vdso`(7) mechanism for exporting certain kernel functionality as userland code, but I do not know of any OS that goes further.

Thank you!

tkchia, 2020-02-24:
Hello @jbruchon,

Another issue I am wondering about is how this would interact with different function calling conventions.  I implemented a `regparmcall` calling convention (`-mregparmcall`, `__attribute__ ((regparmcall))`) for `gcc-ia16` last year, where functions will try to pass their arguments in registers `%ax`, `%dx`, and `%cx`, and arrange for the callee to pop any stack arguments.

`regparmcall` support is now complete for the MS-DOS platform.  I am currently considering adding support for a `regparmcall` calling convention to ELKS's `libc` as a way to save code space, and possibly getting the ELKS kernel to also use it internally.

Thank you!

ghaerr, 2020-02-24:
Hello @tkchia,

Your support for `regparmcall` could be very interesting to reduce kernel and application sizes, would setting a global kernel and application -m option allow that to be tried out?

With regards to some kind of minimal 'shared library' support - this could possibly be implemented through the notion of a 'secondary' text segment at ES: and using trampolines and/or stub functions to get in and out of it. If libc has any static data (I presume yes), then things get quite a bit more complicated. If the call chain allows procedure calls back out of the secondary segment, also a big complication. Then there's all the ES protection wrappers to consider as well.

And then there's the additional issue that libc would then end up being loaded in its entirety, and have to be loaded for any command execution, the opposite of the approach that has just been taking by @marcin-laszewski 's busyElks libc contribution where everything was just split up. So the benefit of even minimal shared libraries is least in single command execution, which is used most, and only greater when running multiple applications at the same time, which is used least (except for the shell is always loaded, of course). The shell could be the 'secondary' text segment including the whole C library, but I don't like that idea.

Ultimately, I just don't see the benefit in trying to bring any sort of shared libraries into ELKS without the benefit of memory mapping.

jbruchon, 2020-02-24:
@ghaerr I would love to see libc packaged up with the kernel and the kernel's linker handle linking to the packaged copy of libc That way, the kernel AND applications can share the code, and libc can be tweaked as required to facilitate that. Even if no further "shared library" support is ever added, all of the programs are carrying chunks of libc with them anyway, and the redundancy between binaries is almost certainly quite significant. The obvious downside is packaging all of libc, but if a working system requires most of it anyway, it makes that downside seem moot. Thoughts?

ghaerr, 2020-02-24:
@jbruchon The kernel uses extremely little of libc, essentially just fast memory copies and some string functions, that's it. Adding libc to the kernel would just make the kernel bigger (too large for 64k), slow down the boot, and still have all the issues I brought up in my last post. Having a library that isn't part of a binary is introduces basic "shared" issues, even if the full-blown modern 32-bit stuff isn't implemented.

Nothing wrong with discussing it though, to implement in the much further future. I personally think the focus for ELKS should be closer to getting it working well, and supporting more hardware disks/probing, stuff like that.

tkchia, 2020-02-24:
Hello @ghaerr,

> Your support for `regparmcall` could be very interesting to reduce kernel and application sizes, would setting a global kernel and application `-m` option allow that to be tried out?

Well... there are two problems:
  * The assembly language routines that interface with C code will need to be updated.
  * Some C routines, as written, pretty much assume a `cdecl` calling convention, and must also be updated.  E.g. `elks/libc/system/exec.c` has this for the variadic function `execl`: (!)
    ```
    int
    execl(fname, arg0)
    char * fname, *arg0;
    {
       return execve(fname, &arg0, environ);
    }
    ```
But other than that, yes. :-)

The `ia16-elf-gcc` man page (installed under `elks/cross/share/man/man1/ia16-elf-gcc.1`) has a bit more information about the available function call conventions and how to enable them.

Thank you!

ghaerr, 2020-02-24:
> * The assembly language routines that interface with C code will need to be updated.

That will be a bit of work. I can now see how beneficial it could be to use `gcc` routines instead of our own when possible, so this wouldn't have to be redone when calling conventions change.

> * Some C routines, as written, pretty much assume a `cdecl` calling convention, and must also be updated.  E.g. `elks/libc/system/exec.c` has this for the variadic function `execl`: (!)

Does adding ellipsis to variadic function's declaration arguments and using <varargs.h> to access them automatically select the `cdecl` calling convention with `gcc-ia16`? 

I'd say that all variadic functions in ELKS kernel and applications should be updated to <varargs.h> compatibility, which would also make a transition easier down the road. Is <varargs.h> a `gcc` header, meaning it can be included in kernel sources (as well as user land C)?

ghaerr, 2020-02-24:
Hello @tkchia,

What I meant to ask in the prior post was whether `cdecl` functions can be called from `regparmcall` functions and have the registers and/or stack automatically handled by `gcc`, or whether this must be done manually with stubs or something else. 

And when taking the address of an argument, as in `execl` above, is that a disallowed operation when the function is `regparmcall`?

ghaerr, 2020-02-25:
Hello @tkchia, no need to answer my questions. I read the `gcc` man page as you suggested and got my questions answered! I'll look at more ELKS code and submit PR's for cleanup in the cases of variadic functions.

tkchia, 2020-02-25:
Hello @ghaerr,

Thank you!  Some further notes:

  * `gcc-ia16` actually already has a provision for automatically building `cdecl`, `stdcall`, _and_ `regparmcall` versions of ELKS `libc` at the same time, and installing them in different parts of `elks/cross/`.  This is based on the GCC front-end's multilib facility.

    > For some targets, invoking GCC in different ways produces objects that can not be linked together.  For example, for some targets GCC produces both big and little endian code.  For these targets, you must arrange for multiple versions of `libgcc.a` to be compiled, one for each set of incompatible  options.  When GCC invokes the linker, it arranges to link in the right version of `libgcc.a`, based on the command line options used. [-- `gccint` info file]

    I had actually deliberately disabled the build of non-`cdecl` versions of `libc`, since --- as I said --- some of the `libc` routines have not yet been fixed to work with the `regparmcall` (and/or `stdcall`) calling convention.  The `$(BUILDMULTIS)` macro in `elks/libc/Makefile` controls which particular multilibs get built.

  * Within a `.c` or `.S` file (but not a `.s` file...) it is possible to query the calling convention given by the compiler options, via the macros `__IA16_CALLCVT_CDECL`, `__IA16_CALLCVT_STDCALL`, and `__IA16_CALLCVT_REGPARMCALL`.

> Does adding ellipsis to variadic function's declaration arguments and using `<varargs.h>` to access them automatically select the `cdecl` calling convention with gcc-ia16?

No, actually --- any fixed (non-variadic) arguments will still be passed in registers as far as possible.  The easiest way to declare a function as `cdecl` is to use an `__attribute__ ((cdecl))` specifier.

Thank you!

ghaerr, 2020-02-25:
Hello @tkchia,

Thank you for your explanation, things are more complex than I had imagined. 

I plan on submitting various cleanups on getting the source more portable, and using the <stdarg.h> mechanism. Then, these procedures, e.g. `execl`, `execle`, etc., will work, regardless of the setting of `-mreparmcall` or not, and won't require `cdecl` either. I plan to start with libc/system/exec.c, but within the kernel `debug()`, `debug1()`, all need cleanup as well.

closed 2023-08-03 by ghaerr