ghaerr/elks #366: Application programs compiled with bcc do not run

georgp24, 2020-02-22:
I think an operating system should allow to develop programs with different compilers and run these. I compiled a hello world program with bcc.
elksemu will report: "Not a elks binary." ELKS will load and execute the program but I get garbled output, no "Hello world".
This program does run though with a version of ELKS compiled with dev86.

tkchia, 2020-02-22:
Hello @georgp24,

`bcc` is no exactly being maintained nowadays, from what I understand.  And, there have been some changes to ELKS's binary program format and its system call interface since `bcc` was last updated --- and some further changes are in the works.

Meantime, if you modify the first 4 bytes in your "Hello world" program from `01 03 20 04` to `01 03 30 04`, it should now run properly.  But more complex programs may fail to run.

Thank you!

georgp24, 2020-02-22:
Thank you tkchia, but this did not work. Enclosed please find the patched bcc executable, hello-bcc.txt and an unpatched one called hellow.txt. The txt extension allows to add them to this post.

[hello-bcc.txt](https://github.com/jbruchon/elks/files/4239408/hello-bcc.txt)
[hellow.txt](https://github.com/jbruchon/elks/files/4239409/hellow.txt)

tkchia, 2020-02-22:
Hello @georgp24,

It seems you accidentally inserted a `30` byte, instead of replacing the `20` with `30`.  Here is a working patched executable: [`hello-bcc`](https://github.com/jbruchon/elks/files/4239475/hello-bcc.txt).  Thank you!

ghaerr, 2020-02-22:
> I think an operating system should allow to develop programs with different compilers and run these. I compiled a hello world program with bcc.

@georgp24 's point, that of allowing older or other compilers to be used to develop applications for ELKS, brings to mind another problem that has come up a couple times recently: that of the content and structure of ELKS include files. This needs community discussion, and would need a separate issue, but I bring them up here quickly to ask whether there is such a need.

It seems to me that the ELKS include structure and contents are a bit of a mess. Simple user land applications end up getting entire sections of kernel defines and structures. If `bcc` or older compilers are to be usable, '#pragma once' would have to be eliminated from user land include files, and kernel CONFIG_* defines should also be confined to kernel includes only. I am not sure it is possible to easily untangle kernel, architecure-specific, and libc (user land) include files.

tkchia, 2020-02-22:
Hello @ghaerr,

> Simple user land applications end up getting entire sections of kernel defines and structures. ... I am not sure it is possible to easily untangle kernel, architecure-specific, and libc (user land) include files.

Yes, I have been trying to more cleanly separate out the kernel and userland stuff in `elks/include/arch/*` and `elks/include/linuxmt/*`, and it is not a lot of fun. :-|

I have half a mind to import some of the older, simpler include files used by `bcc` and the `dev86` project, into `elks-libc`, instead of including a bunch of kernel stuff.  Actually, maybe I should do just that.

Thank you!

jbruchon, 2020-02-22:
I strongly prefer abandoning `bcc` entirely at this point. ELKS has always relied on a specific compiler to build. I'd rather use GCC if I had to pick between the two. I tried looking at the code for `bcc` to improve it and it was frankly too annoying and poorly commented for me to be sufficiently motivated to work on it. GCC is far more complex, but it is also far better structured and extremely heavily tested. While I am interested in the long-term prospect of supporting other platforms, I am very much opposed to sticking with `bcc` moving forward, especially since 6809 support was discarded from it quite some time ago.

I also definitely agree that the layout of everything needs to be reworked. If you look back at my commits from many years ago, a lot of the work I ended up doing was the unpleasant and thankless task of structural cleanups and normalization. It was pretty amazing how scattered, fragile, or broken things were back then. The whole make inclusion mess was quite terrible, too. One of my commits made parallel make functional and that made a huge difference in testing and fixing things moving forward. Any efforts toward simplification or hierarchy cleanup is welcome and encouraged.

ghaerr, 2020-02-22:
> Yes, I have been trying to more cleanly separate out the kernel and userland stuff in `elks/include/arch/*` and `elks/include/linuxmt/*`, and it is not a lot of fun. :-|

No doubt, but I would encourage you to keep trying :)

> I have half a mind to import some of the older, simpler include files used by `bcc` and the `dev86` project into `elks-libc`, instead of including a bunch of kernel stuff. Actually, maybe I should do just that.

Moving towards a clean seperation between an ELKS libc and kernel would be a good move, provided that the architecture-specific portions (would that be <sys/*> only?) don't require separate editing when kernel changes.

The discussion of improving `elks/libc/` is ongoing, please note the questions I've asked @marcin-laszewski regarding his major libc enhancement PR. I support moving the entirety of `dev86` libc to ELKS, and its simpler include files, but probably should be done the way @marcin-laszewski is doing, that is, only adding those portions of `dev86` libc that have not already been added (he has not responded on whether his new code came from `dev86` or not).

Are you suggesting a separate `elks-libc/`, from a `elks/libc/` that would then only contain the kernel-required routines? I support that idea.

ghaerr, 2020-02-22:
> I strongly prefer abandoning `bcc` entirely at this point. ELKS has always relied on a specific compiler to build. I'd rather use GCC if I had to pick between the two. 

To clarify my point: I completely agree that `gcc-ia16` is and will be the *only* compiler used to compile the ELKS kernel and all applications within the ELKS project. I was only responding to @georgp24 's comment that he might want to use an external tool (compiler, in this case, or could be an assembler) to write an application that would run on ELKS. This is easily accommodated provided that libc include files are standard, that's all.

tkchia, 2020-02-22:
Hello @ghaerr,

> Are you suggesting a separate `elks-libc/`, from a `elks/libc/` that would then only contain the kernel-required routines? I support that idea.

Well, more specifically, I would like to arrange for things so that
  * all the include files used for compiling the kernel will only come from `elks/elks/include/*` and possibly GCC's `libgcc` (for OS-independent stuff like `<stddef.h>`, `<stdint.h>`, and whatnot); and
  * all the include files used for compiling userland code and the C library will only come from `elks/libc/include/*` and `libgcc`.

There are a few files which can be fruitfully shared by both the kernel and user code.  E.g. `<arch/stat.h>` is possibly one of them, and `<linuxmt/termios.h>` another.  But I think this sharing should be done in a much more explicit manner (rather than the current arrangment of simply including all directories in `-I`...).

Thank you!

ghaerr, 2020-02-22:
Hi @tkchia,

Thank you for your explanation. I am in favor of each of your ideas regarding kernel and userland include file separation.

> Are you suggesting a separate elks-libc/, from a elks/libc/ that would then only contain the kernel-required routines?

From my statement above then, we are not yet talking about any separation of kernel and user land libc code, only include files, correct?

And my statements on adding other functions from `dev86` libc are an entirely separate matter, the consideration for now of which I will drop.

tkchia, 2020-02-23:
Hello @ghaerr,

> From my statement above then, we are not yet talking about any separation of kernel and user land libc code, only include files, correct?

From what I know, the ELKS kernel does not use any code from the userland C library anyway.  So yes, I am only referring to the need to separate include files used by the kernel and user.

I have started a new branch [`tkchia/simplify-libc-include`](https://github.com/tkchia/elks/commits/tkchia/simplify-libc-include) on my repository to work on this separation.  I expect this to take quite a while.  (`libc` now builds, but `elkscmd` does not build properly yet, and the kernel still uses `libc` include files.  Also it seems that the `ktcp` userland daemon will have to be heavily tied to the specific kernel configuration.)

Hello @georgp24,

If I have time I will see if I can do some minor maintenance on `bcc` so that it can work with newer versions of ELKS.

Thank you!

georgp24, 2020-02-23:
Hello @tkchia,

thank you very much for the hint, when I patch the third byte in a bcc program to 0x30, ELKS will load and run it.

I think it is an improvement to compile ELKS with gcc-ia16 but I think `elks/fs/exec.c` should not be modified so that it will no longer load bcc programs which it did for decades. It should load gcc-ia16 programms and bcc programs as well. Apparently all it needs is to get `exec.c` to accept a program with 0x20 in the third byte as well as programs with 0x30 in the third byte.

A further improvement would be if `exec.c` would also handle programs with 0x10 in the third byte, i.e. the tiny model.

tkchia, 2020-02-23:
Hello @georgp24,

A lot more things have changed in the ELKS kernel than just the one byte.  In fact I think I will need to go through the `dev86` code to see what else in the `bcc` toolchain needs updating.

Thank you!

georgp24, 2020-02-23:
Hello @tkchia ,
bcc generates executables in the a.out format which was the first executable format used by Linux. I think ELKS should continue to support this format. [https://en.wikipedia.org/wiki/A.out](url)
ELKS can support a special executable format for gcc-ia16 but that should not replace the a.out format completely.

georgp24, 2020-02-23:
I made a small change to `fs/exec.c` in my repo and now programs compiled with bcc load and run again. Programs compiled with gcc-ia16 run as well as before.
[https://github.com/georgp24/elks/commits/master/elks/fs/exec.c](url)

tkchia, 2020-02-23:
I stand corrected, kind of: https://github.com/jbruchon/elks/pull/373#commitcomment-37443831 .  Thank you!

jbruchon, 2020-02-24:
I'm closing this due to the fix, but if something else needs to be looked at, let me know and I'll reopen.

closed 2020-02-24 by jbruchon