ghaerr/elks #368: OSX build fails in awk after libc change

ghaerr, 2020-02-22:
@marcin-laszewski: Unfortunately, it looks like the `awk` change you made does in fact fail on OSX (like I said, I've had problems with `awk` compatibility between OSX and Linux). 

The `awk` failure is a result of too many open files, I'm not sure how I'll be able to fix this, so I'm now pretty much dead in the water with the ability to build ELKS.

Here's the error:
```
/Library/Developer/CommandLineTools/usr/bin/make -C /Users/greg/net/elks-gh/libc/build-ml/elkslibc/system VPATH=/Users/greg/net/elks-gh/libc/system -f /Users/greg/net/elks-gh/libc/system/Makefile all
awk -f /Users/greg/net/elks-gh/libc/system/syscall.awk /Users/greg/net/elks-gh/elks/arch/i86/kernel/syscall.dat > syscall.mk
awk: syscall/_getpid.s makes too many open files
 input record number 41, file /Users/greg/net/elks-gh/elks/arch/i86/kernel/syscall.dat
 source line number 36
make[3]: *** [syscall.mk] Error 2
make[2]: *** [/Users/greg/net/elks-gh/libc/build-ml/elkslibc/system] Error 2
make[1]: *** [/Users/greg/net/elks-gh/libc/libc.a] Error 2
make: *** [all] Error 2
Gregs-MBP2:elks-gh greg$ 
```

This is in commit 199aa32167474c3d85e26b3f4ae3fbfab1ef4d0b.

jbruchon, 2020-02-22:
Is there a way to just dump awk entirely for this?

ghaerr, 2020-02-22:
With more rewrite... probably not worth that. I think @marcin-laszewski 's splitting up just went a bit too far, trying to split up each of the syscalls separately. I think if he just 'unsplits' that change, we'll be good to go.

Lets see what he has to say about being able do to that and submit a PR quickly. I can back this commit out on my end to continue working for the time being.

ghaerr, 2020-02-22:
@jbruchon, there's other places in the build that use awk, so removing it just for this isn't likely worth it, that's what I was trying to say. I don't want to try any awk usage rewrite myself, as I'm not an awk expert.

ssieb, 2020-02-22:
@ghaerr can you try adding "close(sf)" at line 47 in libc/system/syscall.awk to see if that solves your problem?

ghaerr, 2020-02-22:
Very nice @ssieb! This fixes the problem. I'll submit a PR now. Thanks!!!

closed 2020-02-22 by ghaerr