ghaerr/elks #371: ELKS now hangs at startup?  "Running /etc/rc.d/rc.sys script"

tkchia, 2020-02-23:
For some reason the merge commit c1cddd353ea52ac6e0add0088fb2bbb36fd5b9ac (or thereabouts) is causing ELKS to hang under QEMU:

[image]

This is, of course, making it a bit difficult for me to do further work on ELKS.

Any ideas what might be causing this?  Thank you!

tkchia, 2020-02-23:
Hello all,

It seems the `clock -s` command was acting up.  I hope to figure out what exactly went wrong.  Thank you!

mfld-fr, 2020-02-23:
Wrong design. HW specific code to be in the kernel. Not in user application.

tkchia, 2020-02-23:
It seems that I accidentally linked `clock.c` with an old broken version of `getopt` which caused `clock -s` to go into an infinite loop.  Once I rebuild everything and confirm this, I will close this issue.

Thank you!

tkchia, 2020-02-23:
Yes, it turns out that the broken `getopt` was indeed the issue.

(@mfld-fr: you do have a point --- it might be good for the kernel to support an additional device file like `/dev/rtc0` for operations on the real-time clock.  But it is probably not a critical flaw to allow applications to access hardware directly.  Linux itself does have `iopl` and `ioperm` syscalls to allow just that, after all.)

Thank you!

closed 2020-02-23 by tkchia

jbruchon, 2020-02-23:
In real mode, you can't really stop applications from doing anything they want to do. That being said, an RTC device would make common clock code accessible to more than one program without having to include that code.

ghaerr, 2020-02-23:
> Wrong design. HW specific code to be in the kernel. Not in user application.

Agreed. Although that wasn't actually the problem here. It seems that changing the getopt.c file didn't cause clock to be rebuilt... so a Makefile problem as well.

> Yes, it turns out that the broken `getopt` was indeed the issue.

Strange how the replaced `getopt` didn't force a remake.

>  But it is probably not a critical flaw to allow applications to access hardware directly. Linux itself does have `iopl` and `ioperm` syscalls to allow just that, after all.)

In any case, the idea that user programs should have non-portable `asm` calls embedded in C code and not use at least a library call for interrupts on/off or a kernel call as @mfld-fr suggested will just lead to more trouble. If we're going to be strict on maintaining ABIs elsewhere all hardware ifdefs around `asm` calls should be eliminated. This isn't necessarily about where the function lives (libc or kernel) but instead keeping user land C files using some provided ABI and no ifdefs.

tkchia, 2020-02-23:
Hello @ghaerr,

> If we're going to be strict on maintaining ABIs elsewhere all hardware ifdefs around `asm` calls should be eliminated.

Agreed.  Thank you!

tkchia, 2020-02-23:
Hello @ghaerr,

> It seems that changing the getopt.c file didn't cause clock to be rebuilt... so a Makefile problem as well.

It turns out that I was linking against an `elks-libc` from a different toolchain installation (i.e. not under `elks/cross/`), which caused the problem.

Thank you!

Mellvik, 2020-02-24:
@tkchia,
I still think the issue needs investigation. I had the same problem when debugging, and only one toolchain installation. It was my impression that the problem was libc not being rebuilt automatically when getopt.c changed.

--Mellvik

tkchia, 2020-02-24:
Hello @Mellvik,

Argh!  It seems you are right --- the dependency checking which I wrote for the `libc` makefile is apparently not as strong as it should be.  I am submitting a patch (https://github.com/jbruchon/elks/pull/381) to fix this.

Thank you!