ghaerr/elks #383: boot loader expansion

ssieb, 2020-02-25:
Instead of trying so hard to squeeze the entire boot loader into less than a single sector, why not just use another sector as well?  I've seen mention that the minix filesystem supports that, but so does FAT.  There is a reserved sector count which defaults to 1 for the boot sector, but can be increased to allow the boot loader to take more space.  I've tested this with various dos versions and they all support this (it's part of the FAT spec).  FAT32 has a higher default value due to other structures that are stored in that space, but the same idea applies.
This would allow not having to make the text messages as small as possible and allow smarter kernel loading.  According to wikipedia, microsoft started doing this as well with their bootloader, starting with fat32.  I really do understand the fun of seeing how much you can you can fit into a small space, but it would make things a lot easier to have some more room.

tkchia, 2020-02-25:
Hello @ssieb,

> There is a reserved sector count which defaults to 1 for the boot sector, but can be increased to allow the boot loader to take more space. I've tested this with various dos versions and they all support this

Strange --- a while back (https://github.com/jbruchon/elks/issues/296#issuecomment-582012628) I tried creating a 720 KiB disk image with 2 reserved sectors using `mkdosfs -R 2`, and MS-DOS 6 failed to recognize it, treating it as a 360 KiB disk (with 1 reserved sector).  But perhaps I missed something.

Thank you!

ghaerr, 2020-02-25:
Hello @ssieb, thank you for your comments.

Your idea of setting the FAT reserved sector count from 1 to 2 was the first idea on our list, and I've checked that the ELKS FAT filesystem driver also supports that. We've decided to use standard tools in creating our filesystems, rather than require the host operating system's FAT filesystem driver in loopback mode. @tkchia reported that when creating a FAT volume using `mkdosfs`, that the resulting disk wasn't usable with MSDOS 6. The ELKS project and myself use  the `mformat` tool, but I don't have a copy of MSDOS 6 with which to test on. Thus, we crammed everything into  1 reserved sector. We need more testing with these two tools and MSDOS in order to move forward, can you elaborate a bit more on the testing you've done?

Mellvik, 2020-02-25:
@Ghaerr,
just marginally out of context: Here's a DOS622 boot image.

—Mellvik
https://www.dropbox.com/s/cv4lphg1aq2fdwj/dos6.img?dl=0 <https://www.dropbox.com/s/cv4lphg1aq2fdwj/dos6.img?dl=0>

ssieb, 2020-02-25:
I just added "-R 2" to the mformat command in the Makefile.  I tested the resulting image with DOS3, DOS6, and win95.
I know it had been mentioned before as not working, but I couldn't remember which issue that was in.

ghaerr, 2020-02-25:
@ssieb, thank you for your testing. Let me test as well, since I'm also using `mformat` (v4.0.23). If it works for me, then we can have @tkchia retry his test to determine whether this is a `mkdosfs` bug or not.

tkchia, 2020-02-25:
Hello @ssieb, hello @ghaerr,

Is there really a way to get `mformat` to format more than 1 reserved sector?  My copy of `mformat` only has a `-H` option for setting the number of "hidden sectors" --- used for hard disk partitions to indicate where they begin in the hard disk.

If the reserved sectors are set correctly, then the first FAT (`f0 ff ff` ...) should begin at offset 1,024 (`0x400`) in the resulting disk image, instead of offset 512.

Thank you!

tkchia, 2020-02-25:
I will try again using a newer version of `mtools`.  Thank you!

ghaerr, 2020-02-25:
> Is there really a way to get `mformat` to format more than 1 reserved sector? My copy of `mformat` only has a `-H` option for setting the number of "hidden sectors" --- used for hard disk partitions to indicate where they begin in the hard disk.

Yes, you have to use `mformat` v4.0.23, as I indicated in my first patch. The -R 1 was removed because most distributions seem to include only `mformat` v4.0.18. They added -R afterwards, and v4.0.23 is the latest version. I can post another copy of the source tarball if needed.

> If the reserved sectors are set correctly, then the first FAT (`f0 ff ff` ...) should begin at offset 1,024 (`0x400`) in the resulting disk image, instead of offset 512.

Yes.

ghaerr, 2020-02-26:
> Here's a DOS622 boot image. —Mellvik https://www.dropbox.com/s/cv4lphg1aq2fdwj/dos6.img?dl=0 

@Mellvik, I can't get that image to boot on QEMU as a floppy or hard drive, but thanks!

Mellvik, 2020-02-26:
@ghaerr,
that's interesting - since it boots fine in Virtualbox, ELKS mounts it like a charm and MacOS likewise.

Anyway - some poking around and it turns out QEMU does not read (or understand) the MBR correctly, you need this in order to boot:
-drive format=raw,file=Dos6.img,cyls=203,heads=16,secs=63

FWIW - the image has a full GCC for MSDOS installation and a bunch off other 'fun' stuff.

—Mellvik

tkchia, 2020-02-26:
Helo all,

While waiting for the 100 MiB image (of some unknown format...) to download, I tried running the "MS-DOS 6.22 (1.2M Boot)" disk image on [Jeff Par's IBM PC AT simulation](https://www.pcjs.org/devices/pcx86/machine/5170/vga/4096kb/), with a disk containing 2 reserved sectors in the (virtual) B: drive.  I am still getting the same error:
```
A:\>chkdsk b: /f

Instead of using CHKDSK /F, try using SCANDISK.  SCANDISK can reliably detect
and fix a much wider range of disk problems.  For more information,
type HELP SCANDISK from the command prompt.

Do you still want to run CHKDSK /F (Y/N)?y
Volume Serial Number is 391D-C945
Probably non-DOS disk
Continue (Y/N)?y

   4 lost allocation units found in 4 chains.
Convert lost chains to files (Y/N)?y

      362,496 bytes total disk space
        4,096 bytes in 4 recovered files
      358,400 bytes available on disk

        1,024 bytes in each allocation unit
          354 total allocation units on disk
          350 available allocation units on disk

      655,360 total bytes memory
      635,328 bytes free
```
Here is a copy of the 2-reserved-sector disk images I created, one with `mformat` and the other with `mkdosfs`: [`20200226.zip`](https://github.com/jbruchon/elks/files/4256461/20200226.zip).  I had also copied a simple "Hello world!" text file into each of them.

Thank you!

ghaerr, 2020-02-27:
>  Anyway - some poking around and it turns out QEMU does not read (or understand) the MBR correctly, you need this in order to boot: -drive format=raw,file=Dos6.img,cyls=203,heads=16,secs=63 

@Mellvik: The above option doesn't work on (my version at least of) QEMU:
```
Using QEMU: qemu-system-x86_64
Using disk image: -drive format=raw,file=MSDOS-6.22.img,cyls=203,heads=16,secs=63
char device redirected to /dev/ttys000 (label chardev1)
qemu-system-x86_64: -drive format=raw,file=MSDOS-6.22.img,cyls=203,heads=16,secs=63: 
Block format 'raw' does not support the option 'cyls'
```
Any other ideas?

After I get MSDOS 6.22 to boot, I'll look into @tkchia 's test results above. 

@tkchia, is it just `chkdsk` that is failing, or does a  directory listing (using `dir b:`) also fail with your above test?

ghaerr, 2020-02-27:
> I just added "-R 2" to the mformat command in the Makefile. I tested the resulting image with DOS3, DOS6, and win95.

@ssieb: What exactly did you test, was it that `dir` worked? What else did you try? Can you try running `chkdsk` on it? How about copying and/or creating files on the volume?

ghaerr, 2020-02-27:
> @tkchia, is it just `chkdsk` that is failing, or does a directory listing (using `dir b:`) also fail with your above test?

I looked at your images, both the mformat.img and mkdosfs.img mount and appear properly using `ls -l` in ELKS, and work with the `mdir` mtools command. If `dir` works on @ssieb's testing, then it appears that the compatibility issue is limited to `chkdsk`. More testing is required reading and writing files on such an image as well.

On ELKS FAT boot images, the cluster size, fat table size, and number of root directory entries are all different from the two test disks. It might be interesting to run `chkdsk` on a working ELKS FAT boot image as well, just to see what it reports.

ssieb, 2020-02-27:
I only have scandisk on the win95 dos boot, but it has no complaints about the disk.  dir works and I copied and deleted a file from win95 and dos3.
Maybe I should point out though that I haven't managed to successfully boot elks using either the fat or minix floppy image.  The minix one doesn't even get the kernel, but the fat one loads the kernel and then I assume can't find something, but there's no message.  But changing the reserved space doesn't affect that in any way.  When I get a chance I will investigate that further.

tkchia, 2020-02-27:
Hello @ghaerr,

> @tkchia, is it just `chkdsk` that is failing, or does a directory listing (using `dir b:`) also fail with your above test?

`dir b:` _seems_ to work, but wrongly thinks that the disk has no files (even though it has 1 file).

Incidentally FreeDOS 1.2 has no problems with the disk — `dir` works correctly, and Imre Leber's `chkdsk` thinks the boot sector is "suspicious" but otherwise gives the disk a clean bill of health.

tkchia, 2020-02-27:
Anyway as I said, I did the MS-DOS 6.22 tests with the help of Jeff Par's PCx86 JavaScript simulation (linked above).  Thank you!

tkchia, 2020-02-27:
Hello all,

> @Mellvik: The above option doesn't work on (my version at least of) QEMU:

I got it to run (along with my disk image in drive B:) with
```
qemu-system-i386 -drive file=Dos6.img,cyls=203,heads=16,secs=63 \
    -fdb 20200226-mformat.img
```
The system was apparently set to use some non-US keyboard layout, so I also had to run `keyb us` within the VM before continuing.

Thank you!

ghaerr, 2020-02-27:
Hello @tkchia, 

I'm glad it's working for you! Using your command line mine is still failing with
```
WARNING: Image format was not specified for 'MSDOS-6.22.img' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
qemu-system-x86_64: -drive file=MSDOS-6.22.img,cyls=203,heads=16,secs=63: Block format 'raw' does not support the option 'cyls'
```
My QEMU is version 4.2.0

Ultimately, it seems your findings that some versions of MS/PC/DOS work with 2 reserved sectors, others don't, in a variety of failure modes, suggests that we might have to stick with 1 reserved sector for a while? I mentioned earlier that the cluster, fat table size, and number of root directory entries are different on your test disks than the ELKS boot disk, and we haven't yet testing the actual compatibility of our boot disk. Thank you!

Mellvik, 2020-02-27:
I updated my QEMU to 4.2.0 and I now have the same problem. QEMU wants me to use the -device keyword instead, which takes the CHS parameters, but I haven't found a way to tell -device which file (and format) to use for a disk.

—Mellvik

ssieb, 2020-02-28:
Here's my disk: http://sieb.net/elksfatboot.bin
```
# minfo -i image/fd1440.bin 
device information:
===================
filename="image/fd1440.bin"
sectors per track: 18
heads: 2
cylinders: 80

media byte: f0

mformat command line: mformat -t 80 -h 2 -s 18 -i "image/fd1440.bin" ::

bootsector information
======================
banner:"ELKSFAT1"
sector size: 512 bytes
cluster size: 1 sectors
reserved (boot) sectors: 2
fats: 2
max available root directory slots: 224
small size: 2880 sectors
media descriptor byte: 0xf0
sectors per fat: 9
sectors per track: 18
heads: 2
hidden sectors: 0
big size: 0 sectors
physical drive id: 0x0
reserved=0x0
dos4=0x29
serial number: 00000000
disk label="NO NAME    "
disk type="FAT12   "
```

ssieb, 2020-02-28:
Here's the qemu parameter: `-drive file=image/fd1440.bin,format=raw,index=0,if=floppy`

tkchia, 2020-02-28:
Hello @ssieb,

> Here's my disk: http://sieb.net/elksfatboot.bin

Thanks!  Something weird seems to be happening here --- the kernel does start up in my QEMU VM, but it keeps complaining about "unexpected" IRQs:
```
ttyS0 at 0x3f8 (irq = 4) is a 16550
lp0 at 0x378, using polling driver
Unexpected interrupt: 14
athd: no drives found
doshd: 1 floppy drive and 0 hard drives
PC/AT class machine, Unknown x86 CPU
639K base RAM.
ELKS kernel (54096 text + 7618 data + 57115 bss)
...
ps2: PS/2 pointing device detected
ELKS version 0.3.0
Unexpected interrupt: 6
Unexpected interrupt: 6
Unexpected interrupt: 6
Unexpected interrupt: 6
```
At the moment I am not sure what to make of this.  I guess I will first see if I can reproduce the problem on my own setup.

Incidentally, if I boot MS-DOS 6.22 with the ELKS disk image in drive B:, then DOS thinks that the disk is completely empty:
```
C:\>keyb us

C:\>dir b:

 Volume in drive B has no label
 Volume Serial Number is 0000-0000
 Directory of B:\

File not found
```
Meanwhile FreeDOS 1.2 correctly sees the files on it:
```
C:\>dir b:
 Volume in drive B has no label
 Volume Serial Number is 0000-0000
 Directory of B:\

BIN                  <DIR>  02-28-2020 12:21a
DEV                  <DIR>  02-28-2020 12:21a
ETC                  <DIR>  02-28-2020 12:21a
HOME                 <DIR>  02-28-2020 12:21a
...
         0 file(s)              0 bytes
        11 dir(s)         533,504 bytes free
```
Thank you!

ghaerr, 2020-02-28:
> Here's the qemu parameter: `-drive file=image/fd1440.bin,format=raw,index=0,if=floppy`

What options are you specifying to get MSDOS 6.22 to boot? Substituting 'MSDOS-6.22.img' for 'image/fd1440.bin' above for the 100MB hard drive MSDOS 6.22 image won't boot in QEMU.

ghaerr, 2020-02-28:
@Mellvik, do you have an MSDOS 6.22 floppy boot image? I think my problem is that MBR isn't supported yet for mounting with ELKS. This would make testing a lot easier, thanks!

ghaerr, 2020-02-28:
> Incidentally, if I boot MS-DOS 6.22 with the ELKS disk image in drive B:, then DOS thinks that the disk is completely empty

@tkchia: It would be interesting to see whether this problem occurs with an ELKS disk *without* our boot sector applied. That is, just using the disk created with the build or package manager process without running `setboot` on it. That will help determine whether it is a tools issue or a boot block issue.

tkchia, 2020-02-28:
Hello @ssieb,

I cannot reproduce the problem you describe on my end.  Here is the (working) disk image I created, after adding a `-R 2` option to pass to `mformat`: 
[`20200228.bin.zip`](https://github.com/jbruchon/elks/files/4267916/20200228.bin.zip).  Thank you!

tkchia, 2020-02-28:
Hello @ghaerr,

> That is, just using the disk created with the build or package manager process without running setboot on it. That will help determine whether it is a tools issue or a boot block issue.

Here is what I get if I exclude the `setboot` command: [`20200228-2,3.zip`](https://github.com/jbruchon/elks/files/4268036/20200228-2.3.zip).  The `20200228-2.bin` file is directly from `elks/image/`, while for `20200228-3.bin` I filled in the necessary bytes (OEM name, boot signature, etc.) at the beginning and end of the first sector, to make the boot sector and BPB valid.

I am still getting the same results for the `20200228-3.bin` disk image --- i.e. MS-DOS incorrectly thinks the root directory is empty, while FreeDOS correctly lists its files.  (Both DOS variants fail to process `20200228-2.bin` with its invalid boot sector.)

So it does not look like a tools issue to me.

Thank you!

tkchia, 2020-02-28:
Hello @ghaerr,

> Ultimately, it seems your findings that some versions of MS/PC/DOS work with 2 reserved sectors, others don't, in a variety of failure modes, suggests that we might have to stick with 1 reserved sector for a while?

It does very much look that way.

Well, there is at least one huge reason the FAT bootloader is so squeezed for space: the `/linux` file cannot be loaded as a single binary blob --- unlike MS-DOS kernels --- but the bootloader must split it up into its "setup" component and its "kernel" component.  This is really a holdover from [the boot protocol for Linux/x86-32](https://www.kernel.org/doc/html/latest/x86/boot.html).  (The Minix filesystem bootloader must also deal with this, but it has a bit more breathing room.)

So, some initial ideas:
  * It might be slightly easier for the boot sector to load a stage-2 bootloader, than to load (and split) `/linux`.
  * An alternative --- or perhaps a complementary solution --- is to rearrange the contents of `/linux` so that we can somehow load and run the file as one blob.  This will almost certainly break compatibility with The Boot Protocol&#8482;, but maybe that is not too bad.

Thank you!

Mellvik, 2020-02-28:
Here you go - 
https://www.dropbox.com/s/4tn1myr3qiweo6l/msdos622boot.img?dl=0 <https://www.dropbox.com/s/4tn1myr3qiweo6l/msdos622boot.img?dl=0>

—M

ghaerr, 2020-02-28:
Hello @tkchia and @ssieb,

I used @Mellvik 's MSDOS 6.22 floppy and was finally able to do some testing for myself. (Still no luck on getting @Mellvik 's 100MB hard drive to boot). My results, testing both `dir` and `chkdsk` on each image:

- MSDOS 6.22 recognizes ELKS FAT boot correctly (uses only 1 reserved sector)
- MSDOS 6.22 dir shows blank disk on @ssieb's `elksfatboot.bin` and chkdsk fails
- MSDOS 6.22 dir shows blank disk on @tkchia's `20200226.bin` and chkdsk fails

Thus, my results agree with @tkchia's, and it seems that MSDOS 6.22, at least, does not recognize floppies with 2 reserved sectors, by showing an empty root directory and chkdsk failing. It is possible that the media byte could have something to do with this. The cluster sizes are different between the two test disks with the same failed result.

For reference, here are all three images, first is @ssieb's, 2nd is @tkchia's (renamed mformat.img), and 3rd ELKS FAT boot (the only working image, and the only image with 1 reserved sector):
```
Gregs-Home-iMac:elks-gh greg$ minfo -i elksfatboot.bin 
device information:
===================
filename="elksfatboot.bin"
sectors per track: 18
heads: 2
cylinders: 80

media byte: f0

mformat command line: mformat -t 80 -h 2 -s 18 -i "elksfatboot.bin" ::

bootsector information
======================
banner:"ELKSFAT1"
sector size: 512 bytes
cluster size: 1 sectors
reserved (boot) sectors: 2
fats: 2
max available root directory slots: 224
small size: 2880 sectors
media descriptor byte: 0xf0
sectors per fat: 9
sectors per track: 18
heads: 2
hidden sectors: 0
big size: 0 sectors
physical drive id: 0x0
reserved=0x0
dos4=0x29
serial number: 00000000
disk label="NO NAME    "
disk type="FAT12   "
Gregs-Home-iMac:elks-gh greg$ minfo -i mformat.img 
device information:
===================
filename="mformat.img"
sectors per track: 9
heads: 2
cylinders: 40

media byte: fd

mformat command line: mformat -t 40 -h 2 -s 9 -i "mformat.img" ::

bootsector information
======================
banner:"MTOO4023"
sector size: 512 bytes
cluster size: 2 sectors
reserved (boot) sectors: 2
fats: 2
max available root directory slots: 112
small size: 720 sectors
media descriptor byte: 0xfd
sectors per fat: 2
sectors per track: 9
heads: 2
hidden sectors: 0
big size: 0 sectors
physical drive id: 0x0
reserved=0x0
dos4=0x29
serial number: 391DC945
disk label="NO NAME    "
disk type="FAT12   "
Gregs-Home-iMac:elks-gh greg$ minfo -i image/fd1440-fat.bin 
device information:
===================
filename="image/fd1440-fat.bin"
sectors per track: 18
heads: 2
cylinders: 80

media byte: f0

mformat command line: mformat -t 80 -h 2 -s 18 -i "image/fd1440-fat.bin" ::

bootsector information
======================
banner:"ELKSFAT1"
sector size: 512 bytes
cluster size: 1 sectors
reserved (boot) sectors: 1
fats: 2
max available root directory slots: 224
small size: 2880 sectors
media descriptor byte: 0xf0
sectors per fat: 9
sectors per track: 18
heads: 2
hidden sectors: 0
big size: 0 sectors
physical drive id: 0x0
reserved=0x0
dos4=0x29
serial number: 00000000
disk label="NO NAME    "
disk type="FAT12   "
Gregs-Home-iMac:elks-gh greg$ 
```

ssieb, 2020-03-01:
I checked it again and now I see that DOS6 can't read the disk properly.  That is very strange, since it works with both DOS3 and win95.  Sorry, I thought I had tested it more thoroughly than that.

ssieb, 2020-03-01:
> > Here's the qemu parameter: `-drive file=image/fd1440.bin,format=raw,index=0,if=floppy`
> 
> What options are you specifying to get MSDOS 6.22 to boot? Substituting 'MSDOS-6.22.img' for 'image/fd1440.bin' above for the 100MB hard drive MSDOS 6.22 image won't boot in QEMU.

For a hard drive, you would change the `if=floppy` to `if=ide`.  I was just using a DOS6 boot floppy, not a full install.  I did end up trying a full hard drive install, but that was using virt-manager where it's easier to manage devices.  You can even add and remove floppies on the fly.

tkchia, 2020-03-11:
Hello all,

Since unfortunately it does not look very feasible to create disks with 2 reserved sectors _and_ still maintain compatibility with MS-DOS 6, it seems to me that there is not much point in leaving this issue open at the moment.  Thus I think I will close this issue for now.

Thank you!

closed 2020-03-11 by tkchia