ghaerr/elks #384: Enhancement request: Covox audio driver

pawosm-arm, 2020-02-25:
Although not present in Linux kernel itself, we may consider having such driver in ELKS kernel. Covox Speech Thing device (a.k.a Disney Sound Source, a.k.a FTL Sound Adapter) is typically controlled by userspace programs utilizing parallel port driver. Unfortunately, those programs (for DOS or Linux, e.g. https://github.com/yeokm1/covox-music-player) are typically dedicated to support particular music formats. It would be better to have a proper in-kernel driver exposing old OSS's /dev/dsp interface abstracting away audio generation (e.g. for sound FX in games) from audio format decoding typical to aforementioned music players. This could be a simple driver, e.g. the device itself does not request any interrupts.

A side note about some of the challenges. The Covox Speech Thing was a device from 286 AT era, too demanding for 8088/8086 PC or PC/XT machines. Most of the games utilizing this device are not working on 8088/8086 CPU at all, some of those that can start, suffer from terrible slowdowns during game play (e.g. Dungeon Master, lovely covox music played on startup, then the game itself isn't comfortable to play on PC/XT). Popular MOD players either don't work at all or malform the music after few seconds of correct play.

mfld-fr, 2020-02-25:
Are you dealing with that HW : https://en.wikipedia.org/wiki/Covox_Speech_Thing ?

pawosm-arm, 2020-02-25:
Yeah, that's the thing.

pawosm-arm, 2020-02-26:
Actually, there is something to look upon: ftp://ftp.informatik.hu-berlin.de/pub/Linux/HU-Sound a source code of an OSS driver,as they claim, for both PC speaker and DAC on LPT (yet another name for Covox Speech Thing device). I guess, having a driver for both PC speaker and a Covox thing could be beneficial to ELKS.

mfld-fr, 2020-02-26:
To get a first idea on the feasibility, do you have any measurement of the current ELKS bandwidth when writing from user space using the kernel API (`open`, `write`, `close`) to the parallel port ?

pawosm-arm, 2020-02-26:
I wonder how a scenario for such measurement would look like. Basically, parallel port in pre-PCI PC is just a bunch of wires attached directly to ISA bus and there's no special chip or logic behind this (apart from I/O port address decoding), some wrongly assume it uses 8255 chip (typical parallel port implementation in 8-bit computers), but the truth is, 8255 although actually being present on XT mainboard, it is used there for completely different purposes and there's no such thing as 'parallel port chip' in the PC world. Thus I assume the bandwidth is limited by ISA bus bandwith.

pawosm-arm, 2020-02-26:
This reminded me, decades ago I was using two devices based on KingByte KBIC-971A - 8bit IDE controller, one was IDE HDD enclose, second was CF-IDE adapter. Both were connected to PC via parallel port and the Linux kernel driver worked with both devices nicely. AFAIR KingByte KBIC-971A used very simple protocol and the driver wasn't much complicated. Eventually, as it wasn't resource hungry, I could use it as a mass storage for a small z80-based 8-bit embedded device. Maybe that could be another thing to port to ELKS (just a loose proposal)? However, I guess, although these storage adapters were pass-through devices, having both Covox and IDE adapter in one chain would be slightly dangerous (to the mass storage medium mostly).

ssieb, 2020-02-26:
The bandwidth question is more about whether or not the user-space application can push data to the parallel port fast enough and even enough for the audio to sound good.

pawosm-arm, 2020-03-23:
I've managed to get something much easier to handle: https://www.serdashop.com/OPL3LPT

Basically it's Adlib music device yet not in a form of 8bit ISA card, but as a small device attached to printer port. Which makes sense, as printer port in PC's is just a bunch of wires attached directly to data lines of ISA bus. So instead of using IOBase 0x388, one needs to use port 0x378. Some of the older games (Rick Dangerous 2, Prehistorik) can be easily forced to make use of it by means of binary patching (and they sound OK), and programming this device is no different than original Adlib music card.

Comparing to Covox, it's much easier to handle in ELKS, as it does not require active CPU participation which would kill scheduler of any time-sharing multithreaded system.

pawosm-arm, 2020-07-18:
Not needed anymore. I've bought the real Sound Blaster 2.0/Adlib clone on 8-bit ISA card and now parallel port is occupied by S2P MIDI synth which is very simple to control.

closed 2020-07-18 by pawosm-arm