ghaerr/elks #387: ^C hangs ELKS

Mellvik, 2020-02-25:
Since grep and bc etc. are now working, it's time for some more command testing. I'm collecting those for a summary, but here is a potentially more general issue:
More + ^C hangs ELKS. Terminal driver issue? (more has its own problems, that's for later).

-Mellvik

ghaerr, 2020-02-25:
I'm not sure whether `more` has a signal catching problem, or not. There are certainly tons of problems with the current set of applications within ELKS. When I was testing the FAT filesystem, I created an infinite looping shell script. If the shell script were run and ^C typed quickly thereafter, ^C would work and stop the script. On other times, ^C would do nothing, and the script could not be stopped thereafter, but ELKS was not hung. Thus, there seem to be some timing issues related to ^C handling, or signal catching within `/bin/sh` (ash).

ghaerr, 2020-03-02:
@Mellvik: I can't get this to repeat in QEMU, more works fine. Can you try it in QEMU as well? 

In QEMU, ^C doesn't do anything when more is waiting for another return to be pressed. When another return is pressed after various ^C's, things still work fine.

Mellvik, 2020-03-02:
@ghaerr,

I’m getting the same on physical, QEMU (ELKS build 19f2c635) and Virtualbox: ELKS becomes completely unresponsive.

And I agree, ’nothing’ is the expected  response.

There are other issues with more(1), but I figure it makes sense to let them be till we have decided which version of more() is the survivor.

—Mellvik

Mellvik, 2020-03-04:
Typing ^Z at the login prompt or any time later, hangs ELKS in the same way.

I suspect we're having a kernel signal handling issue.

--Mellvik

ghaerr, 2020-03-04:
> Typing ^Z at the login prompt or any time later, hangs ELKS in the same way.
> 
> I suspect we're having a kernel signal handling issue.

Geez. I've verified that the tty driver sends a SIGTSTP (stop) signal to the process group on ^Z, I'll check the login and shell code to see whether they handle it or not. I'm deep in the sash shell code now improving it for use as a default shell and will test it there.

jbruchon, 2020-03-05:
@ghaerr fixed a major bug in memory allocation at the kernel level. Let me know if that has fixed this issue.

ghaerr, 2020-03-05:
@jbruchon: I've already tested and this bug is still there. I can reproduce the ^Z problem, and can see its related to the tty driver sending stop signals to the process group. I'm working on both shells now, running them hard, especially cleaning up `sash` for use as a possible 'busyelks' style shell. I will be submitting a series of PRs as mentioned in #410, and will include a fix for the ^Z problem.

I can't duplicate the ^C `more` problem yet, but will investigate if that's related to SIGINT, or something else.

Mellvik, 2020-03-20:
Seems like we need to reopen this one.
^C in more() still hangs ELKS. Tested in QEMU and VirtualBox, no difference between 'old' and 'new' more() - and no explicit signal handling in the program.

--Mellvik

ghaerr, 2020-03-20:
@Mellvik: Thanks for the update above, I had tested `more` heavily but do get the `cat | sort` hang above. I hope this is the last ^C system hang to track down!!

ghaerr, 2020-03-21:
I have solved the `cat | sort` "system hang".  The system wasn't hung, instead `sh` ended up waiting indefinitely for the second process exit status.

There were two big problems, and then further issues between running this in `sh` and `sash`.

When executing this in `sh`, the ^C killed both child processes, but it turns out that ELKS doesn't handle process exit properly. That is, it doesn't handle 'zombie' processes for the `wait` system call, instead it just stuck the exiting process return value in the parent task struct. Since both processes were killed at nearly the same time, the second process exit information overwrote the first's in the shell's task struct. Then, the POSIX-compliant-written `sh` (ash) waited on both processes for exit status and hung in `wait` waiting for process exit information that was never coming. The system was running fine, but the shell prompt wouldn't appear since it was waiting.

Solving this first problem required implementing TASK_ZOMBIE states for processes, and rewriting `sys_wait` to handle them. 

Everything worked great, until doing the same thing with `sash`. Typing ^C when run from `sash` resulted in `sash` dying and the login: prompt appearing. Took a while to find, but sash calls `system`, which saves the `signal` state before forking.  `signal` was buggy and didn't return a saved signal-handler address when setting SIG_DFL, it always returned 0. This caused the default action to occur on ^C, which is terminate, so the shell was killed with the children. This left the children in zombie state, but no parent, so additional work is now done in `wait` to reparent zombie orphans to `init`, where they are finally waited on.

`exec` had to be updated to not try to reuse code segments for zombie tasks, which showed up when running `cat` again after ^C.

A PR will be forthcoming, there's still a few strange cases I'm checking while working on regression testing.

Mellvik, 2020-03-22:
Congratulations @ghaerr,
This is and impressive piece of debugging -and an interesting one to follow thanks to the detailed narrative. I'm very much looking forward to 'bang up' the next commit.

That said, I have to ask: Is it my expectations that are off, or are you - and I mean all in this group doing low level debugging, fixes and development - actually rebuilding a very shaky foundation? A seemingly rhetorical question with a less rethorical follow-up: Do you ever consider rip-and-replace for larger parts of the basic system?

Given the mixed Minix/Linux origin of ELKS, it occured to me to add Minix as a metric when testing disk/floppy i/o speed on physical hardware next week. In addiditon to DOS. If this is useful, which version of Minix would make sense?

//Mellvik

ghaerr, 2020-03-22:
Hello @Mellvik,

> ... actually rebuilding a very shaky foundation?

No, its foundation as a real-mode 8086 OS is solid. It needs more users running it, IMO. There's lots that hasn't been used in many years, and seemingly lots of things broken, but that's changing fast. I'm not sure it's worth trying to update every command, but certainly now various uses are exposing fixes needed in the kernel.

That said, ELKS is really about learning and education. It's a relatively small system, but is complex enough to be interesting to those involved. 

> Do you ever consider rip-and-replace for larger parts of the basic system?

I have thought about that sometimes, but have come to the conclusion that for most cases, it would be better to slightly improve what already is, rather than wholesale replacement. (Some commands could use wholesale replacement, but that's limited). Trying to take ELKS and make it much larger is akin to rebuilding Linux all over again... the kernel is already getting close to 8086 segment limits, so care must be taken as to whether new functions are added to the kernel.

> which version of Minix would make sense?

Minix 1, or an early version. ELKS isn't 32-bit and can't compete with large system implementations.

Mellvik, 2020-03-26:
@ghaerr - what's the fast way to syncronize sash.c sources? My staring point for the history work is older than your latest fixes. Files sash.c, sash.h, config.h

--Mellvik

ghaerr, 2020-03-26:
@Mellvik, why don't you attach a patch file here ('git diff') from your tree, and I'll take a look to see how much work there might be to get your stuff merged into later source.

Mellvik, 2020-03-26:
Thanks @ghaerr,

patch file enclosed.

If the verdict is to merge manually, I'll do that tomorrow.
I left a bit of debug code in the source to accommodate the next round - history search and argument substitution (!$ style).

—Mellvik

ghaerr, 2020-03-26:
@Mellvik, for some reason neither your screen shots nor your patch files are attaching. Are you previewing the submission for either on this website before submitting? That should help determine the problem.

Mellvik, 2020-03-26:
[sash.diff.gz](https://github.com/jbruchon/elks/files/4388294/sash.diff.gz)

Mellvik, 2020-03-26:
@ghaerr,
sorry, I keep sending attachments and uploading via the web-if, don't really understand what's happening. Yes, I'm previewing too.

I've uploaded both again, and here are Dropbox links:
https://www.dropbox.com/s/7prdvrd79q3dsqs/ScreenDump-ELKS-mar26.png?dl=0

https://www.dropbox.com/s/l1g8haakkdlojnb/sash.diff.gz?dl=0

—Mellvik

ghaerr, 2020-03-26:
@Mellvik, got both the patch and screenshot this last time.

Getting your sash patch working shouldn't be too difficult. It is failing on only one hunk. Lets get your system booting again first so we can see what the issue is there. Then I'll prepare a diff for you that works, and you can patch sash and submit the PR.

ghaerr, 2020-03-26:
@Mellvik,

Good news on the boot issue. I'll plan on going ahead with the commits.

BTW, if this ^C hangs ELKS issue is fixed, please close this issue.

Here's a diff that compiles with latest sources. 
[sash.diff.gz](https://github.com/jbruchon/elks/files/4388478/sash.diff.gz)

A couple comments on it:
Running 'history' immediately after starting 'sash' produces strange results.
Given the size of this addition, it probably ought to be placed in a separate 'cmd_history.c' file. Put all the local data in there as well. All routines not required to be exposed to `sash` should be static. The main file 'sash.c' should be minimally impacted, and an init routine could be called for the 'HISTORY' processing, etc.
Add command in 'cmdtab[]' in alphabetical order.
Look at the compiler warnings produced by your patch.
For all new extern declarations, use STDC (that is, list the parameter types, not ()'.
Do a 'git diff' and look for end-of-line space and tab characters.

We can further discuss this in your PR,  and thanks!

Mellvik, 2020-03-26:
Thank you @ghaerr,
I really appreciate your feedback. 30 years 'hiatus' and K&R clearly visible.
Will work in your suggestions tomorrow. And fix the empty history list problem ...

While we're at it, internal sash commands will not work with pipes. This may be ok with aliases, but 'history | more' would clearly be useful. Any thoughts?

--Mellvik

ghaerr, 2020-03-26:
> While we're at it, internal sash commands will not work with pipes. This may be ok with aliases, but 'history | more' would clearly be useful. Any thoughts?

My last enhancement to `sash` was to integrate running `ash` silently/seamlessly for pipes, redirection, and shell scripts. This is because it is far too much effort to add pipes and redirection to `sash`, might as well just use `ash`.

That said, history could paginate itself and/or with some mods to when 'needfullshell()' is called, intrepret more of its own command line (and prohibit calling `ash` on it for processing). Probably best to get the basics working well first, then add enhancements later.

Mellvik, 2020-03-26:
@ghaerr,
point taken - I’ll put the local (history) more on my list.

One more thing (!).
Elvis is running - no compilation errors, some curses/ANSI issues - I’ll create at separate PR for that - but here’s the thing:
When running under ash, elvis file I/O is fine.
Running under sash, there is a VFS error - every time when saving a file:
VFAT: put: trying to free free inode
VFS: device 0000, inode 203, mode 0100600

I’ll upload a QEMU screen dump separately (web).
The error message comes inside elvis, after the save command (say ZZ) before getting the sash prompt back, and the content of the message seems to be the same every time (even the inode#), even if a different file is opened (temp file related?).

—Mellvik

Mellvik, 2020-03-26:
[image]

ghaerr, 2020-03-26:
Open up a separate issue for Elvis, and upload your patch file for getting it running there.
Try it on Minix as well, if possible.
I'll try to duplicate the issue and we'll discuss it over there.

Lets close this issue if ^C doesn't hang anymore.

PR #446 (2020-03-11): [signal] Fix userland signals (take 2)
This PR fixes #387.

Getting ^C and ^Z handling working involved fixing one big thing and lots of little things:
First, the entire userland signal numbers were incorrect, due to libc having it own list of signals requiring 32-bit signal masks, and the kernel's version, which uses 16-bit masks. Since the signal numbers themselves were all different, an entirely different signal was being generated on any signal event. Libc now includes the kernel signal.h header to rectify this.

Various important utilities, include getty, login, ash and sash, didn't handl [... cut]
closed 2020-03-26 by Mellvik