ghaerr/elks #391: Create bootable hard disk with menuconfig fails and FD2880 missing

georgp24, 2020-02-26:
I can generate a bootable FD1440.bin image with the Minix file system ok. This is also possible with the FAT file system, however, the disk is called FD1440.bin and not FD1440-fat.bin as I did expect. 
Also I would like to generate FD2880 disks which are listed in the qemu.sh script. But menuconfig does not offer that disk size, just FD1680.

When I try to create a bootable HD with a Minix file system I get the following message when booting:
``` 
Booting from Hard Disk
ELKS.....Linux found
......*****1!
Press any key
```
When I try to create a bootable HD with a FAT file system I get the following message when booting:
``` 
Booting from Hard Disk
ELKS.3!
Press any key
```
Is it possible to create a bootable hard disk using menuconfig?

ghaerr, 2020-02-26:
> I can generate a bootable FD1440.bin image with the Minix file system ok. This is also possible with the FAT file system, however, the disk is called FD1440.bin and not FD1440-fat.bin as I did expect.

It is currently confusing, as there are two mechanisms that create disk images at this point: 1) the menuconfig system, and 2) the package manager system (`cd images; make images`), which is not run by default. I propose adding `make all` from top level to do both.

The menuconfig system generates the same named file, whether generating a MINIX or FAT image. The package manager always creates separately named files so as to not overwrite the menuconfig image. These files are always named 'fd\<size>-\<type>.bin'.

> Also I would like to generate FD2880 disks which are listed in the qemu.sh script. But menuconfig does not offer that disk size, just FD1680.

The package manager was added to allow extensive user testing while making a wide variety of images quickly available. I added 2.880M disks as an option for testing the cylinders/heads function and FAT16 testing, it has never been available in menuconfig. Nobody seems to know exactly what a 1.680M disk is, so it was not added to package manager.

> When I try to create a bootable HD with a Minix file system I get the following message when booting.  Is it possible to create a bootable hard disk using menuconfig?

We already have an open issue #348 that ELKS isn't properly booting an HD image. It seems that ELKS can't currently boot HD images of any format; this needs to be fixed in this release. I would suggest keeping comments regarding HD boot failing in issue #348, not this one.

georgp24, 2020-02-27:
Thank you Greg, I will try this. 
If nobody has seen a 1.68M floppy disk yet, I suggest to replace that in menuconfig with 2.88M.

georgp24, 2020-02-27:
Still found no way to generate a bootable FAT hard disk.

The package manager works ok. However, if I select in qemu.sh:
`IMAGE="-fda image/fd1440-fat.bin"` or `IMAGE="-fda image/fd1440-minix.bin"`
combined with
`DISK2="-fdb image/fd720-fat.bin"`
I cannot mount the second disk. When I enter: `mount /dev/fd1 /mount` I get:
`VFS: dev 03C0 is not minixfs`
if I combine with 
`DISK2="-fdb image/fd1440-minix.bin"` I can mount the disk ok.

ghaerr, 2020-02-28:
> However, if I select in qemu.sh:
> `IMAGE="-fda image/fd1440-fat.bin"` or `IMAGE="-fda image/fd1440-minix.bin"`
> combined with
> `DISK2="-fdb image/fd720-fat.bin"`
> I cannot mount the second disk. When I [...]

The reason the mount is failing for FAT disks is that `mount` assumes a MINIX image unless the -t option is used. To mount a FAT image use:
```
mount -t msdos /dev/fd1 /mnt
```

georgp24, 2020-02-28:
Thank you Greg. Now I can use the package manager and mount a fat disk. 

I still would like to be able to generate a bootable FAT disk of a size of 32 MB.

ghaerr, 2020-03-19:
Hello @georgp24,

Did the recent work resolve this issue with creating hard disk images? I would like to get resolved issues closed.

With regards to creating 2880k disks, I am wondering whether 1680k would work for your needs. The package manager is going away and it is the only 'extension' creating this size. There is also the issue that 2880k disks are over the FAT12 limit, which requires kernel 32-bit inodes to be selected, so it won't work without building a specific kernel.

georgp24, 2020-03-19:
When I look at the Wikpedia page I find no 1,680 K floppy disks, just 2,880 K ones. These had two sectors per cluster.
https://en.wikipedia.org/wiki/Floppy_disk#cite_note-Mueller-53
Also the 2,880 K floppy disks were formated with FAT12:
https://en.wikipedia.org/wiki/File_Allocation_Table#FAT12
So I think we should drop the 1680 K floppy disk format and support the 2880 k size instead.

ghaerr, 2020-03-19:
Thanks @georgp24. My comment about FAT12 being over the limit was assuming one sector per cluster; if cluster size is 2, then 2880k will work for FAT12.

If 1680k floppies aren't a real thing, I agree, we should replace that selection in ELKS with 2880k. We need a larger size than 1440k, as all ELKS applications won't fit on 1440k anymore, should all apps be wanted on a floppy image. I am working on a method to select applications based on image size instead of source directory now, we could use 360k, 720k, 1440k and 2880k.

ghaerr, 2020-03-22:
Hello @georgp24,

PR #485 added 2880k disk image selection for both FAT and Minix, and previous commits should have handled all issues in here, except the following. Please verify all is working for you, thanks.

> I still would like to be able to generate a bootable FAT disk of a size of 32 MB.

This is currently being held up due to the mtools-4.0.23 package `mformat` not allowing a disk to be created larger than 2880k. Should a tool be found, or a portable mechanism not requiring kernel loopback or host native filesystem support, this can be added.

georgp24, 2020-03-22:
I will test it but I am sure you got it right. I still think one should give non-OS-X users a chance to generate a bootable hard disk with my script as long as there is no alternative available.

ghaerr, 2020-03-22:
> generate a bootable hard disk with my script as long as there is no alternative available.

My last comment was referencing building a 'flat' 32Mb FAT disk image, not an MBR image, as discussed in #406 and #199. I rewrote your script for testing in #406, and that should be used as a portable way to move forward when you have time, but it still only handles the Minix, not FAT case.

If your desire is for a building a partitioned FAT HD image, please close this issue as everything is completed, the MBR hard disk building issues are being discussed in the long-open #199. 

Thanks!

ghaerr, 2020-04-04:
Hi @georgp24,

With PR #506, the ability was added to allow you to easily create MBR partitions from any ELKS disk image. When you find time, let me know how it works for you (see image/Makefile::hd32mbr-minix').

All other items brought up in this issue have been solved with other PRs. Please close this after you've checked that it works, or let me know if you need any help.

georgp24, 2020-04-05:
I am currently working on another project but will look at it within the next few days.

georgp24, 2020-04-07:
I tested it and it works well. So one could close this issue.

However, after some testing with two floppy disks configured, two hard disks configured, floppy and hard disk configured, I just configured one hard disk in qemu.sh. No floppy disk image in the image directory and none specified in qemu.sh. 
However, when ELKS boots, it displays `1 floppy drive and 1 hard drive`. It should not detect a floppy drive!

Also, what are the chs parameters good for in menuconfig? If you leave these untouched but increase just the hard disk size (blocks), when ELKS boots it will display more cylinders due to the increased size. When you call menuconfig again, still the default chs parameters are present.

ghaerr, 2020-04-07:
> However, when ELKS boots, it displays 1 floppy drive and 1 hard drive. It should not detect a floppy drive!

I will look into this. I remember seeing it and  during the same testing and can't remember if ELKS used to display that message or not. I think ELKS does some probing using BIOS calls, and the BIOS is saying the floppy is present. I haven't changed the floppy code.

> Also, what are the chs parameters good for in menuconfig? If you leave these untouched but increase just the hard disk size (blocks), when ELKS boots it will display more cylinders due to the increased size. When you call menuconfig again, still the default chs parameters are present.

At this point, it is up to the user to configure the hard drive geometry and size properly. The CHS parameters are used for disk I/O, the MBR settings, and some other items. The disk block size is passed to `mfs`, and it creates a filesystem without knowledge of CHS. As users start to use the hard disk generation enhancements just added, perhaps the way these settings are entered may change.

georgp24, 2020-04-07:
I think it is more "user-friendly" if you would just have to enter the blocks to specify the size of the hard disk and ELKS will figure out the chs parameters it requires.

ghaerr, 2020-04-07:
> if you would just have to enter the blocks to specify the size of the hard disk and ELKS will figure out the chs parameters it requires.

That would be nice, but won't work for older drives, whose geometries need to match the actual drive.

georgp24, 2020-04-07:
Maybe you can leave the chs fields empty in menuconfig and just require the size in blocks. Then ELKS will calculate the chs values. If you enter the chs values, ELKS will use these.

ghaerr, 2020-04-07:
Hello @georgp24 and @tkchia,

> However, when ELKS boots [with hard drive only], it displays 1 floppy drive and 1 hard drive. It should not detect a floppy drive!

I have debugged this and found the reason is that the BIOS floppy driver defaults to using the location of BIOS data segment at 0010:0040 to read the number of floppy drives.
Only when the config option CONFIG_HW_USE_INT13_FOR_FLOPPY=y does the driver call the BIOS directly to query the number of floppy drives.

@tkchia, by default CONFIG_HW_USE_INT13_FOR_FLOPPY is off. Do you think, given the changes that have been made to the boot sector and BIOS driver that this should default to =Y in defconfig and config? This would fix the issue brought up by @georgp24. Is there a reason why ELKS uses a low RAM memory byte to get the number of floppy drives?

georgp24, 2020-04-08:
On a hard disk there is enough space for all the Nano-X demos. My favorite is NXeyes by the way. Is there a way to generate a hard disk with these applications on it?

ghaerr, 2020-04-08:
> My favorite is NXeyes by the way. Is there a way to generate a hard disk with these applications on it?

The version of Nano-X used in ELKS is v0.86, from circa 1999, as that is the last version that fit in 64k. It is way prior to the NX11 library required by NXeyes and the X11 demos, so none of those applications are in ELKS.

PR #528 fixes the the '1 floppy found' issue brought up, so I'm going to close this issue.

@georgp24, can you test whether FAT flash disks can be booted, as requested in the open issue #320? Various enhancements done over the last month should have enabled that capability.

closed 2020-04-08 by ghaerr