ghaerr/elks #392: Compilation issues with 'ash'

jbruchon, 2020-02-27:
I created a Makefile with the top lines added for the ELKS framework removed and tried to build `ash` by itself under `elkscmd/ash`. Apparently, `ash` is not using the system standard headers and there is a good chunk of `libc` stuff that is duplicated, especially within `mystring.c`. There are also several archaic or redundant code chunks, e.g. the `ck*alloc()` functions in `memalloc.c`. The code also still uses K&R C function declarations. If we're going to offer `ash` as a selectable shell, a lot of these things need to be cleaned up and modernized. (Where is modern `ash` source code kept online, and could we just snatch and rework that instead? Or perhaps we could use `dash` from Debian?)

ghaerr, 2020-02-27:
> I created a Makefile with the top lines added for the ELKS framework removed and tried to build `ash` by itself under `elkscmd/ash`. Apparently, `ash` is not using the system standard headers and [...]

Sadly, I'm not surprised. This is just one taste of how archaic most of our utilities are. I would venture that few have looked closely at most of what is in elkscmds for 20 years.

> If we're going to offer `ash` as a selectable shell, a lot of these things need to be cleaned up and modernized. (Where is modern `ash` source code kept online, and could we just snatch and rework that instead? Or perhaps we could use `dash` from Debian?)

Ash (/bin/sh) is the largest executable command distributed on ELKS. There are considerable reasons that it should remain small, as it is basically always loaded. I would guess that almost all modern shell code (and certainly many new versions of commands that I talked about modernizing via `coreutils`), are larger than the 64k code and data space limitation in ELKS.

I presented my earlier thoughts on `coreutils` as well as the problems with ELKS' command-set in a few earlier posts with little response. I am willing to dive into this, but am looking for some sort of community agreement on the approach to take in these issues. It's not necessarily the way to go to just start individually on each application, as it will take a very long time and there may be better alternatives for the command-set as a whole. Similar discussions are occurring in #388.

Mellvik, 2020-02-27:
FWIW - shells and sizes —

UNIX V7/86 (directly from Unix V7 sources) 
/bin/sh - filesize 36256, size: 17904-2416-620
/bin/csh filesize  53280, size: 45072-8176-4020
Venix/86 (1984) (almost pure V7):
/bin/sh: Filesize 18200
/bin/csh: Filesize 48050 

—Mellvik

georgp24, 2020-02-27:
Busybox contains ash.

jbruchon, 2020-02-27:
@georgp24 It does indeed, though it only loosely tracks ash upstream improvements and BusyBox has some BusyBox-specific peculiarities that would have to be massaged out to use their copy.

georgp24, 2020-02-28:
The latest version of ash will probably have increased in size compared to the busybox version. So it could make sense to make use of that.

closed 2020-04-08 by ghaerr