ghaerr/elks #399: Build-time regression: error: unrecognized command line option '-maout-chmem=0xc000'

pawosm-arm, 2020-02-29:
This commit: 3f1187c92432cf7633aa9b425e6c6d8abb5c0c03 causes following link-time error:
```
ia16-elf-gcc -o fbc -fno-inline -melks-libc -mcmodel=small -mno-segment-relocation-stuff -mtune=i8086 -Wall -Os -maout-chmem=0xc000 bc.o scan.o util.o main.o number.o storage.o load.o execute.o  global.o 
ia16-elf-gcc: error: unrecognized command line option '-maout-chmem=0xc000'
make[2]: *** [Makefile:57: fbc] Error 1
```

jbruchon, 2020-02-29:
You need to build the latest ia16-gcc.

tkchia, 2020-02-29:
Hello @pawosm-asm,

Most probably you had previously installed a slightly older version of the toolchain,  You need to download and build a newer version of the `gcc-ia16` development toolchain, which should supports the `-maout-chmem=` option.  You can do that by running `tools/build.sh`.

Thank you!

pawosm-arm, 2020-02-29:
All it needs is to delete the `cross` directory. Doing it now, it will take some time to build.

pawosm-arm, 2020-02-29:
Indeed, it builds now. Closing it.

closed 2020-02-29 by pawosm-arm