ghaerr/elks #401: Enhancement: PS/2 mouse driver should also support serial mouse

pawosm-arm, 2020-02-29:
PS/2 interface is an extremely rare thing in the majority of real 8088/8086 CPU machines. After spending some cash on various ISA 8-bit mouse adapters that the sellers claimed to be PS/2 being in facts Bus Mouse controllers, I doubt such thing ever existed for 8086 machines outside of the original IBM PS/2 Model 25 and Model 30, which was never a majority of pre-286 PC market. Contrary to this, Microsoft mouse on Serial Port (as detected and operated by cutemouse FreeDOS driver, sources: http://cutemouse.sourceforge.net) was more common.

jbruchon, 2020-02-29:
There is a high chance that Nano-X related stuff won't be a focus for a while  especially with bugs like "control-c locks up the system" right now. The report is appreciated, though! We'll definitely need to implement serial and bus mouse drivers in the future if we end up with a level of stability where we can effectively support Nano-X.

pawosm-arm, 2020-02-29:
I'd add Amstrad's two button mouse (for 8086-based machines) to that list too, although it may require a bit of reverse engineering work as the only drivers for this mouse I've got are the Amstrad's MOUSE.COM and Windows 2.0 driver (surprisingly, works with Windows 3.0 too), both binary only.

ghaerr, 2020-02-29:
Hello @pawosm-arm,

There isn't really a PS/2 driver in ELKS that actually decodes PS/2 (or serial mice). Setting CONFIG_HW_PS2_MOUSE just calls BIOS INT 11h to see if the BIOS thinks a pointing device is installed.

Regarding usage of Microsoft mouse on the serial port and Nano-X, Nano-X is actually quite stable and works well on ELKS with Microsoft mice, as it does its own interpretation on data received on /dev/ttyS0 by default. An ELKS kernel driver is not required. See `elkscmd/nano-X/drivers/mou_ser-new.c` for details, and let me know if you still have problems. You might try `stty -raw 1200 < /dev/ttyS0; cat /dev/ttyS0` to determine the mouse data is coming in before running nano-X, for instance.

pawosm-arm, 2020-02-29:
Hi @ghaerr, thanks for your input.

> You might try stty -raw 1200 < /dev/ttyS0; cat /dev/ttyS0 to determine the mouse data is coming in before running nano-X, for instance.

There's no `-raw` option in ELKS's `stty`, but I guess it's not needed. Continuing, `cat /dev/ttyS0` indeed shows some trash characters as I'm moving the Microsoft serial mouse connected to COM1: or as I am pressing its (two) buttons. Also, cutemouse driver in FreeDOS operates it correctly (e.g. I can make use of it in Norton Commander or SimCity).
Unfortunately, it does not move 'the player' in the landmine game. I suspect it is due to general serial port issues I'm observing in ELKS that I don't observe in FreeDOS.

pawosm-arm, 2020-03-01:
@ghaerr I have some suspicion regarding those serial port issues that only appear on ELKS. Could it be caused by `Calibrating delay loop... failed` message on boot as I set `CONFIG_BOGOMIPS=0` in menuconfig? If all the timings in ELKS kernel are based on this value, then maintaining baud rate in serial port driver seems impossible. Unfortunately, I don't know how to estimate this value for any given PC/XT machine.

ghaerr, 2020-03-01:
@pawosm-arm: It looks like the problem is that elkscmd/nano-X/Makefile has no mouse setup by default! Not sure how that happened. In the first few lines of that file, there are a few DRIVERS += lines. There are three drivers that may work for you, mou_ser.o, mou_ser_qemu.o, and mou_ser-new.o. I suggest setting the last one, commenting out the mou_null.o one and recompiling. Look inside the C file in drivers/mou_ser*.c for more details as to the default device name.

Let me know what works for you and I'll prepare a PR to fix this permanently. It is important that ELKS users can play landmine :)

Thanks and have fun!

ghaerr, 2020-03-01:
I should mention that elkscmd/nano-X/drivers/mou_ser-new.c has support for Microsoft, Logitech, and PS/2 mice, by setting MOUSE_TYPE in the source file. The serial port used is also settable within. Once you get this working, I'll look at combining the incompatible Microsoft Mouse support in QEMU and the hardware serial into a single driver, so things will be easier in the future. To test with QEMU, use mou_ser_qemu.o and add -serial msmouse to the QEMU command line.

pawosm-arm, 2020-03-01:
`mou_ser-new.c` causes immediate crash with artifact drawn below mouse pointer.
`mou_ser.c` (with the mouse type changed from `"pc"` to `"ms"`) also causes a crash (with slightly smaller artifact drawn), yet some time later, with overall poor user experience as the mouse arrow moves bit too slow.
Also, I've noticed, reading /dev/ttyS0 with mouse attached stuck after a while of fast repetitive mouse motion.

pawosm-arm, 2020-03-01:
> It is important that ELKS users can play landmine :)

Would be good to port FreeGEM to ELKS/nano-X at some point - it works nicely on low-end 8088/8086 PCs, and old GEM could run even on Atari ST.

ghaerr, 2020-03-01:
> `mou_ser-new.c` causes immediate crash with artifact drawn below mouse pointer.

Geez.

> `mou_ser.c` (with the mouse type changed from `"pc"` to `"ms"`) also causes a crash (with slightly smaller artifact drawn), yet some time later, with overall poor user experience as the mouse arrow moves bit too slow.

Ok, so 'mou_ser.c' runs for a little while, then crashes, correct?

I'm not sure if the crash is the result of kernel hang in serial driver, or in nano-X...

> Also, I've noticed, reading /dev/ttyS0 with mouse attached stuck after a while of fast repetitive mouse motion.

It sounds like that's a possible bug in our serial driver not handling input fast enough, which may also be the reason for the slow handling above.

pawosm-arm, 2020-03-01:
> Ok, so 'mou_ser.c' runs for a little while, then crashes, correct?

Yes, the length of that 'while' depends on the amount of the mouse motion.

ghaerr, 2020-03-01:
BTW, you can tell whether it's nano-X or ELKS hanging by pressing the ESC key, which will exit Nano-X.

pawosm-arm, 2020-03-01:
The system is completely died and unresponsive after the crash, pressing ESc has no effect, it must be kernel that hung.

ghaerr, 2020-03-11:
Hi Paul,

Are you still wanting a mouse driver? I think a better option would be to leave this to a userland mouse emulator, not a kernel driver. If so, then this issue can be closed.

pawosm-arm, 2020-03-11:
Yeah, this is kind of a thing better handled in userspace. I need to open serial port issue though.

closed 2020-03-11 by pawosm-arm