ghaerr/elks #402: Can't get busyELKS options to build proper image

ghaerr, 2020-02-29:
Hello @marcin-laszewski and @mfld-fr,

When running `make config` and selecting busyELKS in two different scenarios, both produce failures in the resulting image. I am starting from scratch by deleting .config.

When all options are set to default, except answering Y to this question:
```
busyelks (CONFIG_APP_BUSYELKS) [N/y/?] (NEW) 
```
an fd1440.bin image is produced, but the symlinked set of busyELKS commands are not included in it. That is, in elkscmd/busyELKS/:

 ```
Gregs-MBP2:busyelks greg$ ls -l busyelks*
 -rw-r--r--   1 greg  staff  28164 Feb 27 20:39 busyelks.tar.xz
 -rwxr-xr-x   1 greg  staff  46453 Feb 27 20:36 busyelks
 -rw-r--r--   1 greg  staff      0 Feb 27 20:36 busyelks.map
 Gregs-MBP2:busyelks greg$ tar Jtf busyelks.tar.xz 
 ./
 ./bin/
 ./bin/busyelks
 Gregs-MBP2:busyelks greg$ 
 ```

On the other hand, when I select busyELKS as above in config, and also answer 'b' to each of the other busyELKS application categories as follows:
```
busyelks (CONFIG_APP_BUSYELKS) [N/y/?] (NEW) y
dbgutils (CONFIG_APP_DBG_UTILS) [Y/n/?] 
diskutils (CONFIG_APP_DISK_UTILS) [Y/b/n/?] b
elvis (CONFIG_APP_ELVIS) [Y/n/?] 
fileutils (CONFIG_APP_FILE_UTILS) [Y/b/n/?] b
levee (CONFIG_APP_LEVEE) [Y/n/?] 
m4 (CONFIG_APP_M4) [Y/n/?] 
minix1 (CONFIG_APP_MINIX1) [Y/b/n/?] b
minix2 (CONFIG_APP_MINIX2) [Y/n/?] 
minix3 (CONFIG_APP_MINIX3) [Y/b/n/?] b
miscutils (CONFIG_APP_MISC_UTILS) [Y/b/n/?] b
```
then the fd1440.bin image produced does not contain commands required to login, as /bin/login, /bin/cat, etc are missing.

Who is testing this? I'm not getting either option to work.

I repeat below my questions brought up in PR #389, which were unanswered.

I'm getting confused as to how the busyelks option is supposed to work. What exactly is selecting 'b' rather than 'y' in the subsequent config questions supposed to do?

It is supposed to include the busyELKS (symlinked) versions of each of the applets in each category, right? And NOT include the standalone versions of the applets, since they're the same filename and location (e.g. /bin/ls).

What is supposed to happen when 'y' is answered to 'busyELKS' *and* each application category? With the last change in elkscmd/Makefile #389, this seems to build both the original command versions and the busyELKS commands. It seems to only be doing the former. If it succeeds in building the latter, then there will be duplicate commands being installed, as mentioned in PR #389.

marcin-laszewski, 2020-02-29:
I have tested it and it seems to be.
Look at attached typescripts: *.txt - typescript, *.t - timing.
You can use 'scriptreplay', eg.:
$ scriptreplay -t a.t a.txt

marcin-laszewski, 2020-02-29:
Upss.. gitub does no accept *.xz file. Here are gziped files:

[a.tar.gz](https://github.com/jbruchon/elks/files/4270915/a.tar.gz)
[b.tar.gz](https://github.com/jbruchon/elks/files/4270916/b.tar.gz)

ghaerr, 2020-02-29:
I am on OSX, no scriptreplay.

Are you saying this all works on Linux? Perhaps the scripts in elkscmd/busyelks/Makefile aren't compatible.

Can you clarify what the 'b' options are supposed to do above? I am still confused as to what the two config options (Y for busyELKS, Y for everything else, and Y for busyELKS, B for everything else) are supposed to do. In all cases, busyelks.tar.gz is not being created properly. In the first case, elkscmd/Makefile builds duplicate commands. Right?

marcin-laszewski, 2020-02-29:
All command from config/menuconfig are gruped into moduls: Fileutils Minix1, Minix2, ...
Y for BusyELKS - compile & install /bin/busyelks
Y for everything - compile & install regular command.
B for everything - commands from this module which are supported by busyelks are included into busyelks. Regular command are not installed, but symlinks to busyelks are created.

ghaerr, 2020-02-29:
I looked through your a.txt and b.txt, and can see that the links are in fact installed from one of them in the resulting image (very hard to follow with the embedded cursor addressing). Thus I believe the first problem is that the elkscmd/busyelks/Makefile script is failing on OSX, I will look further into that.

ghaerr, 2020-02-29:
> All command from config/menuconfig are gruped into moduls: Fileutils Minix1, Minix2, ...
> Y for BusyELKS - compile & install /bin/busyelks
> Y for everything - compile & install regular command.
> B for [...]

Thank you, this what I expected. It appears that /bin/busyelks is being created when either Y or B selected, just the command symlinks are not when B is selected.

ghaerr, 2020-02-29:
@marcin-laszewski,

I have debugged the problem: OSX `ln` doesn't support the -r option (relative path on symlink). I don't want to change your code without having written it, can you remove -r and test or revise, and submit a PR? Thanks!

pawosm-arm, 2020-03-02:
> then the fd1440.bin image produced does not contain commands required to login, as /bin/login, /bin/cat, etc are missing. Who is testing this? I'm not getting either option to work.

I've faced it too: yesterday I've build MINIX HD image containing busyELKS for a first time, also I've selected `B` for every piece of userland software that offered such option. The problem became apparent right after boot: init script started to complain about missing `cat` command.

ghaerr, 2020-03-02:
> I've faced it too: yesterday I've built an MINIX HD image containing busyELKS for a first time, also I've selected `B` for every piece of userland software that offered such option. The problem became apparent right after boot: init script started to complain about missing `cat` command.

Exactly. Are you on Linux? OSX needs ln's '-r' removed for sure, but I think there may be more problems. If you're on Linux and this doesn't work, then it's still buggy as I suspected.

pawosm-arm, 2020-03-02:
Ah, mid-air collision.  I've added comment about doing it on Gentoo Linux the same time you asked your question :)

marcin-laszewski, 2020-03-02:
Could you record your activity using "script" program?

............
$ script --timing=x.t x.txt
$ make clean
$ make config
$ make
..........

ghaerr, 2020-03-02:
I can’t on OSX, but perhaps Paul can. 

I think the first bug is the ‘-r’ option on ln. it would be easier to debug if you used separate commands in shell or make rather than using &&. Then command echoing could be turned on, and the script would stop with errors.

pawosm-arm, 2020-03-02:
It's here:

[x.t.gz](https://github.com/jbruchon/elks/files/4277460/x.t.gz)
[x.txt](https://github.com/jbruchon/elks/files/4277461/x.txt)

ghaerr, 2020-03-08:
Hello @marcin-laszewski,

Were you ever able to figure out what the problem is that prevents 'B' options from producing symlinks to /bin/busyelks on OSX and Gentoo Linux?

ghaerr, 2020-03-18:
Fixed by #466.

closed 2020-03-18 by ghaerr