ghaerr/elks #404: umount does not leave MINIX-fs in a 'checked' state

pawosm-arm, 2020-02-29:
**Description**
I've mounted MINIX-fs in `/mnt` then un-mounted it, then mounted it again. Turned out, MINIX-fs became 'unchecked' after `umount`.
- What is expected ?
As a 'checked' MINIX filesystem was mounted, it should remain 'checked' after `umount`.
- What is observed ?
`umount` does not leave MINIX-fs in a 'checked' state.

**Configuration**
- Version (commit or release):
939241982f3c80ab93178fba8fc35cb23163726a
- Emulator or HW platform:
Amstrad PC 2086

**How to reproduce ?**
- Does the problem always happen ?
Yes
- List the steps to reproduce the problem.
With rootfs mounted from floppy, I'm mounting MINIX-fs installed in XT-IDE CF card adapter device, directly on `/dev/bda` (i.e. there's no partition table).
1. login as `root`
2. `mount /dev/bda /mnt`
3. `ls /mnt` - should list files and directories on CF card
4. `umount /dev/bda` .or. `umount /mnt`
5. `ls /mnt` - should list no files or directories
6. `mount /dev/bda /mnt` - it prints:
```
MINIX-fs: mounting unchecked file system, running fsck is recommended.
```

The filesystem is mounted regardless, e.g. I can see the files and directories doing `ls /mnt`.

After I'd un-mounted it again, I tried to run `fsck /dev/bda` on it. Unfortunately, it refuses to operate as such:
```
sys_brk failed: len 41360 > endseg 33168
fsck: Unable to allocate buffer for zone count
```

This system has 640kB of ram, no EMS or high memory of any kind. `CONFIG_MEM_PAGES=10` is set in the `.config`.

While booting, ELKS prints:
```
640K base RAM.
525 K of memory for user processes.
```

**Raw data**
- Screenshot(s):
- Log file:

**Additional information**
Add any other context information about the problem here.

ghaerr, 2020-02-29:
@pawosm-arm, thank you for your bug report.

> I've mounted MINIX-fs in `/mnt` then un-mounted it, then mounted it again. Turned out, MINIX-fs became 'unchecked' after `umount`.

It looks like the MINIX superblock VALID_FS flag is being improperly reset in `elks/fs/minix/inode.c` function `minix_write_super`. I don't yet understand why it isn't setting the flag instead, nor why remounting a previously written MINIX filesystem doesn't show this message an a reboot, for instance.

> After I'd un-mounted it again, I tried to run `fsck /dev/bda` on it. Unfortunately, it refuses to operate as such:
> 
> ```
> sys_brk failed: len 41360 > endseg 33168
> fsck: Unable to allocate buffer for zone count
> ```

This is yet another case of trying to use commands that haven't been run for awhile... the problem is that the default memory allocation to `fsck` is just 16k. It's Makefile will have to be updated with a new option we recently added `-maout-chmem=0xfffc` to allocate as much memory as possible, which hopefully will be enough to check the filesystem.

> This system has 640kB of ram, no EMS or high memory of any kind. `CONFIG_MEM_PAGES=10` is set in the `.config`.

ELKS processes are strictly limited to 64k code and 64k data, so unfortunately your total RAM won't help in this case.

ghaerr, 2020-03-01:
@mfld-fr: I would like your help on this one.

I've debugged the umount problem down to where a single line fix, commenting out the `unmap_buffer(bh)` call at the bottom of the successful minix_read_super function in elks/fs/minix/super.c fixes this issue.

It seems that unmap_buffer is allowing the buffer to be released (I don't see that code), and thus, in the following code (see my debug statements inserted) in minix_put_super, the saved value of the struct minix_super_block pointer `ms` is not pointing to the buffer contents (which should be the super block contents) anymore:
```
void minix_put_super(register struct super_block *sb)
{
    register char *pi;

    lock_super(sb);
    if (!(sb->s_flags & MS_RDONLY)) {
if (sb->u.minix_sb.s_ms != sb->u.minix_sb.s_sbh->b_data) printk("BAD\n");
printk("put state1 %x\n", sb->u.minix_sb.s_ms->s_state);
        sb->u.minix_sb.s_ms->s_state = sb->u.minix_sb.s_mount_state;
printk("put state2 %x\n", sb->u.minix_sb.s_ms->s_state);
        mark_buffer_dirty(sb->u.minix_sb.s_sbh, 1);
    }
```
Thus, the rogue saved pointer is pointing into garbage and the saved MINIX VALID_FS flag isn't being reset when the filesystem is unmounted. In the above example, "BAD" is printed, and the previous value (which should be 0 or 1) is 364 - garbage.

So we have a buffer bug here - not sure if this is related to any of your previous commits on buffer enhancements, or whether this unmap_buffer call has been buggy the whole time. Can you take a quick look and advise me as to what you think?

I also don't quite understand how unmap_buffer ends up releasing the buffer, I don't see a call chain finally going to `brelse`. Thanks.

ghaerr, 2020-03-01:
After having studied the MINIX filesystem and buffer code extensively, it is clear that this issue has uncovered a longstanding bug in the MINIX filesystem. Sections of read_super, remount, and umount will need to be rewritten, as the current code erroneously tries to keep an L1 buffer with the superblock contents and local pointers into it around the entire length of time the filesystem is mounted, but releases the buffer anyways. There's no need to keep an L1 buffer permanently locked  just to write the checked flag on unmount, so the superblock can be reread and written on remount and unmount.

On another matter, few may realize another memory-hungry part of the MINIX filesystem code: In a section marked FIXME, whenever a MINIX image is mounted, the system reserves and reads into up to 4 imap (inode bitmap) buffers and 8 zone (data bitmap) buffers, permanently, depending on the size of the filesystem. For floppies, this is only two buffers, but for large HD images, much larger.

I'll fix the first problem in a PR soon, but just wanted to bring up that ELKS uses lots more memory than may be presumed for things we may not be aware of.

Can anyone clarify how the buffer system works, if `bread` is called, but not `map_buffer`, is that buffer contents preserved, or may it be thrown out if more buffers needed (if s_dirty is 0)? Is there a limit on the amount of L2 buffers used? And finally, since much of the  kernel uses `bread` followed by `map_buffer`, wouldn't it make sense to have a `bread_map` call that reads and maps a buffer, so to avoid a possible far memcpy from L2 space to L1 (kernel data) space?

ghaerr, 2020-03-01:
@pawosm-arm: I've fixed this problem in PR #408, thank you!

@mfld-fr, I've figured out the buffer system, sorry for all my questions.

mfld-fr, 2020-03-02:
@ghaerr : Sorry for the late response. I agree, the current code looks to mix the very different multi-tasking things that the buffer locking (= while working on the buffer data) and the reference counting (= need to keep the buffer in memory) are. Plus the fact that synchronization is performed in the same object (the buffer head) for both L1 and L2, despite they are two different caches, one relying on the other.

I saw it while cleaning the  code in #380, so I avoided to modify the behavior, just cleanup, until having more standard and comprehensive locking and reference counting primitives (started in #361).

I am currently trying to implement a generic cache that would be multi-task safe and that could replace the different and unsafe implementations repeated for block L1 cache, block L2 cache and inode cache, as said in https://github.com/jbruchon/elks/pull/393#issuecomment-592409128.

ghaerr, 2020-03-02:
> I am currently trying to implement a generic cache that would be multi-task safe and that could replace the different and unsafe implementations repeated for block L1 cache, block L2 cache and inode cache, as said in [#393 (comment)](https://github.com/jbruchon/elks/pull/393#issuecomment-592409128).

Now that I understand the buffer cache better, I'll comment on your redesign as you present it. I wasn't aware of an inode cache, I thought inodes were just allocated and freed out of static data with iget/iput. Are you talking about a better integration of caching with your dynamic memory allocation improvements?

mfld-fr, 2020-03-02:
Yep, exactly, the goal is to cache using dynamic memory, to retain as much data as possible, and thus improving the performance with slow devices, like FDD. I am wondering if such caching would be able to handle the 'big' BusyELKS executable...

ghaerr, 2020-03-06:
@pawosm-arm: My recent commits should have fixed this problem. Please close this issue should you find it working, and thanks!

pawosm-arm, 2020-03-07:
It works now. Thanks!

closed 2020-03-07 by pawosm-arm