ghaerr/elks #405: floppy driver disabled

ssieb, 2020-03-01:
I spent a long time trying to figure out why my floppy images wouldn't boot.  I was also getting "unexpected interrupt 6" repeatedly.  I finally discovered that these are connected because the floppy driver is completely disabled.  IRQ 6 is the floppy one.
https://github.com/jbruchon/elks/blob/939241982f3c80ab93178fba8fc35cb23163726a/elks/arch/i86/drivers/block/floppy.c#L80-L88
So my big question is, how are you all booting from a floppy (or image)?

tkchia, 2020-03-01:
Hello @ssieb,

May I know how exactly you are building ELKS?  Did you follow the `README.md` procedure, do a `make menuconfig` --- or `make defconfig` --- before doing a full `make`?

(Yes, it will help greatly if you give further details on _how_ exactly you were building the system.  Especially so, if you were doing anything different from the documented procedure.  This will help us help you resolve your problems more quickly, and save a lot of time.)

Anyway, on my end, the default ELKS build options includes the following:

[image]

I have no idea why "BIOS block device support" is somehow not enabled on your end --- since you did not say anything about how you were building the system --- but on my end it is very much enabled.

Perhaps what you can do now is to run a `make defconfig` first to restore default settings, then run a `make menuconfig` to modify any settings that you think need to be modified.

Thank you!

tkchia, 2020-03-01:
Hello @ssieb,

Regarding the kernel internals: the kernel has (at least) two sets of drivers for accessing floppy disks:
  * the "raw" driver, which you found, where the kernel directly deals with the floppy drives' I/O ports;
  * the BIOS driver, where the kernel calls out to routines in the computer's ROM BIOS to do the floppy I/O --- this is in `elks/elks/arch/i86/drivers/block/doshd.c` (and perhaps elsewhere too).

Thank you!

ssieb, 2020-03-01:
I did follow the README, but I enabled the direct floppy driver without knowing that it would break things.  It also turns out that just unchecking it doesn't fix the problem.  There are some persistent settings that don't get reset.  I did a full clean and make and now it works.
I would leave this open to see about fixing the floppy driver unless you don't think that's useful.

Mellvik, 2020-03-01:
@ssieb, @ghaerr - 
Your exchange reminds me of a question I have wanted to ask the key developers for a while - about BIOS vs independence:  Is there a long term idea about where we're going with this? Getting rid of code that doesn't work or is unused is appealing.
Maybe it makes sense to keep using the BIOS for legacy x86 hw, saving kernel space to boot?
Close to the top of my list of things to test in ELKS is a headless (serial console, no CGA/VGA/...) & flash (2,8M) floppy config. Enough storage for a really workable system and a bigstep towards a legacy-free SoC system down the line.

--Mellvik

tkchia, 2020-03-01:
Hello @ssieb,

Thank you for your information!  Perhaps naming the particular build option as "Normal floppy disk support" was a bit too confusing --- even if I do know what it is supposed to mean --- and it should be reworded.

Hello @Mellvik,

> Is there a long term idea about where we're going with this? Getting rid of code that doesn't work or is unused is appealing.

Indeed, this is something that needs to be considered...

Thank you!

ghaerr, 2020-03-01:
> > Is there a long term idea about where we're going with this? Getting rid of code that doesn't work or is unused is appealing.
> 
> Indeed, this is something that needs to be considered...

I for one got quite confused trying to understand which floppy driver actually worked when first getting involved here. Does `drivers/block/floppy.c` actually work? If not, it would be nice to remove it and its config option.

ssieb, 2020-03-01:
See the code segment I linked. When I tried enabling it, there were a bunch of compile errors. I didn't look too closely at them at the time.

ghaerr, 2020-04-06:
Hello @ssieb,

As mentioned in #504, the floppy driver you selected in config (as 'Normal') that brought about this issue won't even compile, and inspection shows it is extremely out of date and would need lots of work to even compile, let alone work. As a result, PR #499 changed the name of this driver to 'Experimental HW floppy disk support' from 'Normal' so users won't select it in the future.

Personally, I think the config option ought to be entirely removed, along with the other 'Experimental HW hard disk support', since neither work at all. 

So, I am closing this issue for you, since there is little chance this driver will ever see the light of day at this point.  If you believe something else should be done, please reopen this issue.

Thank you!

closed 2020-04-06 by ghaerr