ghaerr/elks #425: Huge number of compilation warnings when building

jbruchon, 2020-03-06:
I rebuilt ELKS from scratch today, and I'd really like to see the huge number of compilation warnings during building resolved. There are hundreds, possibly even thousands, of warnings present; fixing them could result in bug fixes, but more importantly, any new warnings are easily lost in the noise of the existing warnings.

ghaerr, 2020-03-06:
We've fixed major bugs that had compiler warnings and went unnoticed, as well as missed new warnings and introduced new bugs. I agree completely, but the problem is just too large to tackle at once. Both the kernel and applications have tons of warnings.

The warnings need to be broken down into classes, which will help plans to get rid of them. A large class of warnings is that of calling libc routines without extern declarations, because many libc headers don't include them. @marcin-laszewski's work in splitting libc routines has helped a lot, since he's been adding undefined routines into the headers (thanks Marcin!). Unfortunately, there are lots of elkscmd/ warnings around 1) Pre-K&R C issues and 2) C source without previous declarations; both of these will be very slow to fix, as they require hand-editing of each of many of our 107 applications. The kernel also suffers a bit from the latter problem as well.

If you can come up with some large categories of warnings, perhaps we can work to get a plan together to attack en-masse, rather than one-at-a-time.

mfld-fr, 2020-03-09:
> any new warnings are easily lost in the noise of the existing warnings

And so you added more warning in 6cb81e26e0658064058d532a67d915f0ed104fda to make more noise... what is the logic here ?

jbruchon, 2020-03-09:
I'm working on the warnings myself.