ghaerr/elks #428: Weird FAT filesystem corruption causing login failures?

tkchia, 2020-03-07:
Hello all,

It seems that if I build a FAT filesystem image for ELKS and run it, I am able to log in the first time:
```
FAT: me=f0,csz=1,#f=2,floc=1,fsz=9,rloc=19,#d=224,dloc=33,#s=2880,ts=0
FAT: 1440k, fat12 format
VFS: Mounted root (msdos filesystem).
Running /etc/rc.d/rc.sysinit script
ELKS built from commit cf18a39

ELKS 0.3.0

login: root
#
# sync
# reboot
```

Thank you!

ghaerr, 2020-03-07:
Hello @tkchia,

Wow. Just when I thought we had all the major ELKS bugs tamed, you find this one,
and it's a real wild pony ride. It's not a FAT bug, but instead a major memory corruption
problem that just happens to show up on FAT. It also shows up on MINIX, but isn't related to a filesystem driver issue. It is most strange, I've got corruption bugs reproducing by removing `/bin/init` from Packages, thus the system mounts root (MINIX or FAT) then runs /bin/sh standalone. At that point, both FAT and MINIX seems to run normally, and `reboot` works great. However, spurious calls to sys_open show that garbage filename and mode are being passed when running `ls`, even on MINIX. (This was part of the reason FAT failed in your test, sys_open was open for writing somehow and caused the disk to sync with bad blocks).

Have there been changes made to sys_open recently that might cause this? It appears that the corruption problems may be happening as early as the first call to `open("/dev/tty1", O_RDWR)` in any program, including `init` and the system code that runs `/bin/sh` when init is missing.

ghaerr, 2020-03-07:
Hello @tkchia,

I have located the problem, finally. My earlier post was incorrect, when displaying names passed to sys_open, I hadn't realized those addresses are in user space, so that they display incorrectly.
This was happening on FAT and MINIX.

The bug is that, on FAT, since /dev/tty1 is opened read/write, when the filesystem is unmounted by `reboot` (which apparently hasn't happened much), that inode, marked dirty, is flushed to disk. Since /dev inodes are faked on FAT, garbage is written to the disk sector derived from the inode number. In this case, the /dev/tty1 number is (hilariously enough) '42', which happened to overwrite the directory for /bin/sh.

A PR will be forthcoming, along with major cleanup of /bin/init and some other kernel init startup used to track this down!

Thank you!

jbruchon, 2020-03-07:
@ghaerr This post is great for so many reasons. I feel like it could have been so nicely developed into a passage within The Hitchhiker's Guide to the Galaxy. Did you happen to be drinking a cup of tea while finding this? :-) Very nice find!

ghaerr, 2020-03-07:
@jbruchon Indeed LOL. I'm definitely thinking that /dev/tty1 = 42 is a sign of the end of FAT bugs and a window to beyond...

tkchia, 2020-03-08:
@ghaerr, @jbruchon: thanks for the fix!  (I am a bit disappointed that the Answer does not involve Tibetan monks, though.)

closed 2020-03-08 by tkchia

jbruchon, 2020-03-08:
I didn't actually do anything other than joke around, but I'm happy to take credit for being silly!