ghaerr/elks #436: Check code cleanups for issues

jbruchon, 2020-03-09:
I've just committed a big pile of code cleanups, mostly automated but some done by hand. I committed several times during the process to aid in git bisection if necessary. Please feel free to check for issues that may have been introduced by this big cleanup. I have verified that compilation still succeeds. The specific things I went after were:

* Spacing around arithmetic/logic operators and if/for/while/switch statements
* Converted a few functions from K&R to ANSI
* Some formatting/whitespace cleanups

I'll close this issue after either a few people report that I didn't bust anything or a few days elapse without issues.

mfld-fr, 2020-03-09:
What is the added value of this reformatting except more burden to keep our development branches synchronized, and regression risk all over the source tree ?

There are nice and well tested reformatters (like `clang-format` used by Linux) that are safer than your hand-made command lines, and that are available as code checkers on the Github market place for automation in the PR workflow.

I won't spend time to test non-regression because of these commits. It is already hard to insulate the effective and valuable changes in the code because of much code reformatting in the latest PRs. If there are too many conflicts to push to upstream, I will merely not push.

jbruchon, 2020-03-09:
Then send a well-tested reformat in a PR.

ghaerr, 2020-03-09:
Hello @jbruchon,

I am sorry but I have to *completely* agee with @mfld-fr regarding the consequences a 10,000+ line commit like this has with this project and to its active development team. I will give the list of reasons here:

- The very large commit with large consequences was made without using the normal process of pull request and discussion, and the other programmer's opinions on its usefulness were not asked.
- For all programmers that keep large amounts of source code in various other repositories, this commit makes the comparison of changes almost untenable, and definitely decreases maintainability in ELKS. For instance I've been the maintainer of the Microwindows project for 20 years, and I stuck a copy of v0.86 in ELKS in 1999. When I was debugging Nano-X to work on ELKS last year, I would frequently use `diff -urN` to compare subtrees, to get it working by seeing changes that were made in ELKS versus the working 0.86 version. This commit changes that v0.86 copy, and should not. There is all sorts of code, including libc, elkscmds/*, ash, that are negatively affected by this commit. Any source that came from an early package will not be able to be compared with modern versions in the cases of a programmer considering a user request to upgrade a package, as has been discussed more than a few times recently.
- This commit was made during one of the recently busiest multi-programmer efforts to ELKS, stopping them in their tracks. Myself, I've been in the middle of a tedious visual and diff comparison (#410) of each of three versions of each source version of each command, between file_utils, sash, and busyelks. This commit has made that hard job much harder, to the point it won't be worth it, and I will be closing that issue.
- This commit touches code actively maintained code by another programmer. IMO, others should let active portions be changed only by that maintainer, and they may have other work in the pipeline that is affected negatively.
- There is no added value in a large portion of this massive commit. Almost all the changes are made to whitespace that can't even be seen, yet affect maintainability greatly forever. Also, the main issue with the ELKS source code wasn't even addressed, which is the fixed leading 4-space indentation instead of tabs used affecting readability. Even though I've long thought about it, I would never make that change, for the reasons above.
- The massive touch of so much source should have used a professional or even fairly good reformatter, like `indent`, to fix the many remaining problems with readability even after this commit. However, because of the reasons above, using a reformatter to change 10,000+ lines in the middle of a release and during extensive programmer activity should still be firmly rejected.
- The added value of the non-whitespace changes is far less than the increased work required by all programmers working on this project. I have spent many hours working hard on ELKS lately, and this is a major step backwards, at least at this time. I have recently tracked down the '^Z/^C' problem, and discovered that signals are pretty screwed up in ELKS. I had created another cleanup of usable debug statements all over the kernel to track this down. I don't even want to pull upstream to start the work required to send this fix over.
- This commit changes very little of the problem of K&R -> ANSI C prototype conversions, which should be approached using the 'class' mechanism discussed in another thread, to actually help with the massive compiler warnings problems ELKS has. And sometimes when looking at warnings, diffing with earlier versions is a good idea for correctness, which this commit hinders.

Conclusion: this commit should not have happened without using the PR process, and definitely should be discussed to see what others think about its 'usefulness'.

jbruchon, 2020-03-09:
Alright, I'll revert it all.