ghaerr/elks #437: 'shutdown' and 'reboot' should un-mount rootfs upon termination

pawosm-arm, 2020-03-09:
As in the subject. The `shutdown` command does not seem to do anything (the system is still alive after `System halted` message) while rebooting the system with the `reboot` command results in un-checked rootfs filesystem warning message upon the subsequent system boot.

ghaerr, 2020-03-09:
@pawosm-arm, thank you for your continual testing of commands. It seems many still either have or show more bugs!

The `reboot` command does an umount of root, so apparently that's not working. I just debugged and fixed this issue on FAT, brought up by @tkchia, so I assume you're talking about Minix filesystem being left unchecked? If so, that would indicate a bug in umount, which I'll look into.

pawosm-arm, 2020-03-09:
Hi @ghaerr, it's indeed in MINIX, yet umount itself works properly for MINIX (my tests with mounting/umounting HD filesystem with ELKS booted from floppy confirm that), it's just `reboot` (and `shutdown`) seem not to do this... (e.g. they may attempt to do so and give up as the filesystem remains 'busy' at that point...).

closed 2020-03-10 by ghaerr

pawosm-arm, 2020-03-11:
The fix worked, no more un-checked rootfs filesystem warning message upon the subsequent system boot.
Thanks @ghaerr!

pawosm-arm, 2020-03-11:
@ghaerr I'm observing something opposite now: a crash in nano-X forced me to reboot the machine without doing proper reboot (with umount), yet subsequent boot didn't print any warning message that the MINIX filesystem was un-checked. Is it something to worry about?

ghaerr, 2020-03-11:
@pawosm-arm I tested that ELKS halting without 'umount /' (remounting read-only) should leave filesystem in unchecked state, but I'll test again... The proper behavior is that filesystem should be in unchecked state unless either `reboot`, `shutdown` or `umount /` performed, yes.

ghaerr, 2020-03-11:
@pawosm-arm : I just checked, and you're right, there is still an issue here.

The problem is: the filesystem super block is mounted and then the unchecked flag set in the superblock, and the buffer marked dirty, but the I/O for that *delayed* (normal). However, this means that if the system is halted *without* a sync, then the dirty flag isn't written to disk. Everything else works fine.

The fix requires that any filesystem booted from, whether ever written to or not, requires an immediate write of the superblock, which always sets the unchecked flag. It would be nice to avoid that on floppies, but since other I/O could have been queued, and written but not the superblock yet, I suppose the only answer is to commit the superblock unchecked flag immediately (as it does now when setting clean).

Do you understand? What do you think?

pawosm-arm, 2020-03-11:
IMO to keep filesystem 'checked', mounting it in r/w mode should always be paired by umount, no matter what kind of device is in use. Also I kinda like Fuzix behaviour: it starts fsck on any un-checked filesystem mounted at boot time. It takes time and many times requires interaction with the user (whenever a problem with fs is found), but it gives good impression on robustness.

ghaerr, 2020-03-11:
> IMO to keep filesystem 'checked', mounting it in r/w mode should always be paired by umount, no matter what kind of device is in use. 

Ok, I'll fix this remaining problem in a PR. Thanks for your checking and reports!!

ghaerr, 2020-03-12:
Hello @pawosm-arm,

After making this last fix to force-write the minix superblock immediately after mounting RW,
I realize that this has forced users into having to run `shutdown` on *every* invocation
of ELKS, or their filesystem will be left in the 'unchecked' state, and then they have to run
`mfsck` on it to fix it. Running `reboot` won't help, unless they quickly turn the system
off or terminate the emulator within the three-second period before the system reboots
and mounts root RW again.

In the old way, users could boot ELKS for testing with no write to the filesystem, unless
they did something or ran `sync`, as the unchecked state to the superblock was delayed.

I wanted to bring this up to see whether this is still desired behavior, as likely most 
users and developers will notice, since they might not be shutting their system down
in the 'proper' fashion. Also, I haven't yet considered the issues involved with the kernel
asking for a floppy disk switch. 

Does anyone else have comments on this new behavior?

pawosm-arm, 2020-03-12:
`sync` setting filesystem checked? Could do... although this does not solve the issue of pending write accesses that never affected filesystem due to sudden power down. The user may never find out that loss of (never written) data might have happen.

ghaerr, 2020-03-12:
> `sync` setting filesystem checked? Could do... although this does not solve the issue of pending write accesses that never affected filesystem due to sudden power down. The user may never find out that loss of (never written) data might have happen.

Agreed.

I think I'll just leave the fix in, as it is proper behaviour. It is certainly desired behaviour on HD systems, so only casual floppy users will notice the (now correct) behaviour change.

pawosm-arm, 2020-03-12:
> Running reboot won't help, unless they quickly turn the system off

Hey, that's exactly what I was doing before `shutdown` became able to halt the system for real.

TBH, the `shutdown` role on XT should be further extended! If anyone uses MFM drive (typical to XT machines), shutdown should not only umount rootfs, but also it should `park` the drive. Otherwise, the user should invoke `reboot`, then boot DOS (e.g. from floppy) and execute `park` command from DOS to make sure the machine can be powered down safely.

Mellvik, 2020-03-12:
Hi @pawosm-arm,
Admittedly, your  comment made me smile. It's quite unbelievable that some of those MFM drives still run. My 1987-89 compaqs suddenly seem modern. 

If there is a command, ioctl or whatever that can be sent to the drive to effect parking the heads, this should be easy to add, probably  in a 
If (cpu == 8088 or 8086) then send park command;
Do you agree? Or have a better idea? I don't think MFM drives ever made it beyond the 8086.

-- Mellvik

pawosm-arm, 2020-03-13:
> It's quite unbelievable that some of those MFM drives still run. My 1987-89 compaqs suddenly seem modern.

This 1988's Amstrad PC 2086 (XT clone) came with 3.5'' MFM drive and WD1002A-27X  MFM controller on 8-bit ISA card. As it arrived I've replaced MFM with xt-ide controller, an 8-bit ISA card and CF-IDE adapter, so I can attach Compact Flash cards to it. I don't even know if the original MFM drive works. The original 5.25'' MFM drive from my other 1986's XT machine has died long time ago and this one is also booted from xt-ide. I doubt MFM controller would be of any use in any of the AT machines, their BIOSes support ESDI and IDE drives.

Mellvik, 2020-03-13:
Doesn't that mean that the park head issue is really moot?
That the remaining population of ancient (still living) drives are IDE drives that need no special treatment? [When testing ELKS on physical HW I use original Compaq (Conner) 20 and 40 MB drives. ELKS handles those flawlessly.]

@ghaerr, I'll get back to you with shell test results shortly.

--Mellvik

PR #441 (2020-03-10): [minix] Fix Minix umount super block unchecked flag, add new debug mechanism
This fixes #437. The Minix superblock is now force written on remounts, rather than previously trying to keep track of its status (which didn't work after `sync`). The `reboot` command does a sync, then mounts the root filesystem read-only, then sleeps for three seconds and reboots.

The `sleep` system call is not implemented (I recommend removing it entirely), and libc `sleep` is broken, so `sleep` is now (properly) written in reboot.c, for the time being. I will have a PR for libc `sleep`, although making any change [... cut]