ghaerr/elks #439: More MacOS build (awk?) issues...

Mellvik, 2020-03-09:
While on the road last week, I figured it was time to try to build ELKS on MacOS again, this time on an MBpro 2017 (catalina).
No way - I must be missing something since some have this running.
BTW - I just tried on my desktop, and get exactly the same ...

make menuconfig:
```
/Users/helge/src/elks/Make.defs:10: ELKS is not configured yet
/Library/Developer/CommandLineTools/usr/bin/make -C config all
/Library/Developer/CommandLineTools/usr/bin/make -C lxdialog all
Package ncurses was not found in the pkg-config search path.
Perhaps you should add the directory containing `ncurses.pc'
to the PKG_CONFIG_PATH environment variable
No package 'ncurses' found
make[2]: Nothing to be done for `all'.
config/Menuconfig config.in
Using defaults found in elks/arch/i86/defconfig
Preparing scripts: functions, parsingawk: MCmenu11 makes too many open files
 source line number 22
Awk died with error code 2. Giving up.
.../MCmenu1: line 27: syntax error: unexpected end of file
........../MCmenu9: line 8: syntax error: unexpected end of file
done.
```
BTW - fixing the ncurses.pc PATH didn't make any difference...

The reference to awk and too many open files does ring a bell.

--Mellvik

ghaerr, 2020-03-09:
The big OSX issue that remains is building (and getting a working) config/menuconfig.

I have solved the config issue, you need to install 'ncurses' using homebrew `brew install ncurses`.

After that you can edit a little tidbit I've left in 'config/lxdialog/Makefile', uncomment out the following line, if your version of brew doesn't install the ncurses pkg-config:
```
#LDLIBS = /usr/local/opt/ncurses/lib/libncurses.a
```
And make sure you run `. env.sh` first to set some variables for make, then run `build.sh` to build the latest `gcc-ia16` (that takes a while).

You are then ready to run `make config` and use the text based configurator, which works. `make menuconfig` still fails.

Good luck!

ghaerr, 2020-03-10:
> The reference to awk and too many open files does ring a bell.

I hadn't noticed that in the errors above. It is possible that menuconfig will run on OSX if the awk open files bug is fixed, but I don't know awk well enough to locate the problem. I do get the text-graphic screen displayed, and it dies after a keystroke and spitting out the error message.

Mellvik, 2020-03-10:
Here's the fix (apologise for not moving into git yet):

```
helge@iMac27 elks % diff ~/Menuconfig config/Menuconfig
720c720
< }'
---
> close(ifile) close(menu) }'
```
After this change, Menuconfig works fine on Catalina (btw I didn't change the config/lxdialog/Makefile ...

mfld-fr, 2020-03-10:
@Mellvik : could you please keep the issue open until the fix is eventually merged into `master` ? Otherwise it is hard to track the actually solved issues with fixes 'in the middle of the air'.

ghaerr, 2020-03-10:
> After this change, Menuconfig works fine on Catalina (btw I didn't change the config/lxdialog/Makefile ...

That's great news @Mellvik! This was the last incompatibility I know of on OSX. I'll stick it in a PR. I've also fixed your ^C/^Z signal bug and another problem, PR's forthcoming...

PR #447 (2020-03-11): [config] Fix menuconfig on OSX
This fixes #439 

Thanks @Mellvik for both finding and fixing this!
closed 2020-03-11 by ghaerr