ghaerr/elks #440: 64K DMA boundary on PC XT

mfld-fr, 2020-03-09:
@tkchia : I think I got at least the explanation for the "*do not cross the 64K boundary*".

I was playing with `setup` to move the kernel data segment in memory, when I got the error:
```
hd: error: AX=0x09
BIOSHD:I/O error
dev 380, sector 6
```

In IBM-PC manual, error 09H is "Attempt to DMA across a 64k boundary".

Then I found this page about the DMA controller: https://en.wikipedia.org/wiki/Intel_8237#IBM_PC_use

Do you understand the same thing as me ?

mfld-fr, 2020-03-09:
More info: in my test, the kernel data segment starts at D63:0h, and `system.map` gives:
```
 .bss           0x0000000000001baa     0x359e fs/fs.a(buffer.o)
 .bss           0x0000000000005148        0x0 fs/fs.a(super.o)
```
So yes, the array of L1 buffers is crossing the 64K boundary with the linear range F1DAh to 12778h.

I added a trace just before the error occurs, and it fails when filling a 1K buffer at D63:27C7h, so less than 1K before the 64K boundary.

If I force the kernel data segment to starts at 1000:0h, no more error.

ghaerr, 2020-03-09:
> If I force the kernel data segment to starts at 1000:0h, no more error.

Out of curiosity, could the L1 buffer declarations be taken out of C code and instead be placed by startup code, say, above the kernel stack at a 64k boundary (and KSP reset below it) , so that instead of memory being wasted, that memory goes into the future heap/stack area?

mfld-fr, 2020-03-10:
@ghaerr : sorry, I did not understand your question ?

tkchia, 2020-03-10:
Hello @mfld-fr,

> Do you understand the same thing as me ?

Yes, this is what I was referring to.  Thank you!

mfld-fr, 2020-03-10:
@tkchia : OK, so it means the current memory layout is unsafe, and works by chance because the data segment is relocated as low as possible (actually starts at 80:0h), so it would not cross the first 64K boundary as long as its size is less or equal to F800h (= 62K).

mfld-fr, 2020-03-10:
Issue was already caught in #107 (`BUFSEG` conditionnal).

Other references:
- https://github.com/jbruchon/elks/pull/422#discussion_r389224706
- https://github.com/jbruchon/elks/pull/429#issue-385144653
- https://github.com/jbruchon/elks/commit/9fab4d52ff285735935838e252b0ee5617c2d86d
- https://github.com/jbruchon/elks/commit/7f0c568aaac553b9b9ae7394b4e72287d7465e41

tkchia, 2020-03-10:
@mfld-fr: incidentally:

  * The `DMA_OVR` patch does not look right to me.  It may do multi-sector reads to `BUFSEG:0` (= `0x60:0`), but if the kernel data segment starts at `0x80:0`, then there is really only space for 1 sector at `BUFSEG:0`.
  * `BUFSEG:0` currently overlaps with the setup variables at `SETUP_DATA:0`.  Is this a good idea?

Thank you!

mfld-fr, 2020-03-10:
@tkchia : what do you think about moving the data segment to the highest 64K in `setup` ? This would secure the DMA anywhere in that segment, especially if the DMA buffers would be dynamically allocated. There would not be any memory wasting as the data segment would be extended anyway to 64K to allow the largest heap (see #393).

ghaerr, 2020-03-10:
> what do you think about moving the data segment to the highest 64K in `setup` ? This would secure the DMA anywhere in that segment, especially if the DMA buffers would be dynamically allocated. There would be no memory wasting as the data segment would be extended anyway to 64K to allow the largest heap.

Yes I like this idea, it is what I was trying to suggest but using startup code instead of setup:

> could the L1 buffer declarations be taken out of C code and instead be placed by startup code, say, above the kernel stack at a 64k boundary (and KSP reset below it) , so that instead of memory being wasted, that memory goes into the future heap/stack area?

I now realize @mfld-fr's idea is even cleaner, starting the entire data segment, including the L1 buffers, at a 64K boundary and placed by `setup`. No need for kernel stack change.

mfld-fr, 2020-03-10:
@ghaerr : sorry, but I still not understand. What do you call the 'kernel stack' ?

ghaerr, 2020-03-10:
@mfld-fr: Perhaps I'm incorrect, but doesn't the kernel have its own stack area, which is switched to on a system call and timer interrupt? In UNIX, the userland stack pointer is saved system calls and interrupts and %SP is set to a kernel stack area, within which the kernel procedure calls return addresses are saved, then switched back to the userland %SP when user programs run. I was assuming that this 'kernel stack pointer' into the 'kernel stack area' is also how ELKS works... is that correct?

I was also assuming that the kernel stack at the highest kernel data segment address growing downwards, of course.

mfld-fr, 2020-03-10:
@ghaerr : I see... there is no 'kernel stack' at the end of the data segment, as in the user processes. Each task has its own stack (the 'task stack') in the task array, plus an additional stack only for the interrupts (the 'interrupt stack').

ghaerr, 2020-03-10:
@mfld-fr: Thanks for the clarification. So the kernel executes code on the user process's stack at all times, and is uninterruptible by other processes, except for the timer interrupt, which executes on its own stack. Thank you.

mfld-fr, 2020-03-10:
@ghaerr : no, the kernel switches from the user process stack to the 'task stack' upon a `syscall` (INT 80h).

ghaerr, 2020-03-10:
@mfld-fr: Thanks, I finally get it. This answers why various spots in the kernel code mention that the stack usage is too large. I had previously thought the kernel stack was shared and possibly a lot larger.

tkchia, 2020-03-10:
Hello @mfld-fr, hello @ghaerr,

> I now realize @mfld-fr's idea is even cleaner, starting the entire data segment, including the L1 buffers, at a 64K boundary and placed by `setup`.

There is a bit of a problem though with newer machines (including my own).  If there is an [Extended BIOS Data Area](https://wiki.osdev.org/Memory_Map_(x86)) at the top of conventional memory, then the top of _available_ conventional memory may not be nicely aligned to 64 KiB.

It is probably worthwhile to loosen the restrictions on the data segment placement a bit.  E.g. as far as I know only the parts starting from the BSS actually participate in DMA, so only these parts really need to be within DMA boundaries.

Thank you!

mfld-fr, 2020-03-10:
@tkchia : yes, the EBDA...

mfld-fr, 2020-03-10:
@tkchia : OK, so what about inserting again the BUFSEG (with enlarged size from 512 to 1K) between REL_INITSEG and REL_SYSSEG, and enabling again the DMA_OVR patch ?

ghaerr, 2020-03-11:
Hello @tkchia and @mfld-fr,

> There is a bit of a problem though with newer machines (including my own). If there is an Extended BIOS Data Area at the top of conventional memory, then the top of available conventional memory may not be nicely aligned to 64 KiB.

Instead of dealing with the problems of moving the kernel data segment up to higher memory, it seems that the DMA 64k alignment within L1 buffers could be solved differently, in a way that loses at most 1023 bytes of total memory, and nothing in the kernel data segment, by slightly adjusting the location of the kernel data segment in low memory:

> It is probably worthwhile to loosen the restrictions on the data segment placement a bit. E.g. as far as I know only the parts starting from the BSS actually participate in DMA, so only these parts really need to be within DMA boundaries.

Since the 64k DMA boundary issue is only a problem with the L1 data buffers (=1k size each), and the kernel never DMA reads into more than one buffer at a time, then the real issue is that the L1 data buffers must be aligned on a 1k boundary, which automatically solves the 64k DMA overlap problem, correct?

If the L1 data buffers declarations (there are 8-12 of them) were taken out of 'elks/fs/buffers.c' and placed into 'elks/arc/i86/boot/crt0.S', and ensured that they were the first BSS items to the linker, then `setup` could adjust slightly upwards the location of the kernel data segment as a whole to be aligned such that the beginning of the BSS area (= L1 buffers) was always on a 1024 byte physical memory boundary. This would always be <= 1023 bytes, and not affect kernel data segment size, only total memory. 

At the point that the dynamic heap manager is ready, the L1 buffers could stay put, or, preferably be allocated with a 1k-block alignment flag, anywhere from the BSS area. If a flag were used, then the need to keep the BSS starting at a 1k boundary could be relaxed, and that memory used for other allocations.

@mfld-fr: Is this what your PR #445 is doing, or is that to insert the BUFSEG back in, as you mention above?

mfld-fr, 2020-03-11:
@ghaerr : the problem is not related to any 1K alignment.

tkchia, 2020-03-11:
Hello @mfld-fr,

> @ghaerr : the problem is not related to any 1K alignment.

The way I see it, it _is_ kind of related.  If each of the `BLOCK_SIZE`-sized buffers in
```
static char bufmem[NR_MAPBUFS][BLOCK_SIZE];
```
are aligned at `BLOCK_SIZE` (i.e. 1 KiB) boundaries, then each individual `bufmem[`_pi_`]` will always be entirely on one side of any 64 KiB DMA boundary.  This arrangement will work if you do not do reads across several `bufmem[]`'s.

Thank you!

mfld-fr, 2020-03-11:
@tkchia : aligning buffer to 1K and reading / writing only 1K at a time is a way to solve the problem, not THE problem.

mfld-fr, 2020-03-11:
Aligning L1 buffers would also cause much heap fragmentation in the kernel data segment. Better solution is to modify the segment allocator with a ' do not cross 64K boundary' flag, and perform DMA in that allocated segment, leaving the kernel heap for small objects.

ghaerr, 2020-03-11:
Hello @mfld-fr,

What is the whole problem? I don't quite get it yet. Are you talking about the ability to DMA (in the future, or in setup now) more than a 1k disk block into memory? Or are you talking about the whole problem being compatibility with the future dynamic memory manager?

> Better solution is to modify the segment allocator with a 'do not cross the 64K boundary' flag, and perform DMA in that allocated segment, leaving the kernel heap for small & unaligned objects

Are you talking about in the future, with the dynamic heap allocator, or now with buffers other than L1 here? I'm sorry if my questions appear stupid, I'm just trying to understand. Moving L1 buffers out of kernel heap would be a good idea, but if left in the kernel data segment they would not be dynamically allocated without fragmentation.

> Aligning L1 buffers would also cause much heap fragmentation in the kernel data segment. 

I assume here you mean with the dynamic heap manager in the future. Aligning them now to, say, BSS and a 1k boundary does solve the current problem.

After thinking more about all this, if you are talking about the whole problem being not now, but for a dynamic memory manager in the future, then I agree with you. The L1 buffers still need to reside in kernel data segment and will still cause fragmentation, unless far pointers are used for buffer data, which seems to me a better solution getting L1 buffers out of kernel data entirely.

mfld-fr, 2020-03-11:
@ghaerr : yes, I am thinking since the beginning about the future, even if pushing small steps toward that envisioned future : dynamic allocation in local memory (= kernel data segment) for small objects like descriptors (SB, inode, dirent) and global memory for large objects like buffers.

ghaerr, 2020-03-11:
@mfld-fr: Ok, I understand now. In your previous RFC you were talking about allocating L1 buffers in the kernel heap, have you backed off that idea, since the fragmentation problem of large objects as well as limited space in heap?

If so, and you think far pointers affect too much code, perhaps keeping L1 as static BSS allocation and using 1k alignment of kernel data segment as I discussed now would be good idea to solve current problem, while waiting for dynamic memory manager and global buffers? Why not?

BTW, I like the idea of moving large object creation out of the local heap, but this still needs consideration because tty, pipe, task, and other allocations are large. That's why I was trying to solve our current problems now, and also understand the future issues.

mfld-fr, 2020-03-11:
@ghaerr : L1 buffers cannot escape from the kernel data segment without highly impacting the existing file systems.

Idea is more to perform I/O in the L2 buffers, so in global memory, before 'mapping' them into the kernel space (= L1 buffers in data segment). The current design does exactly the opposite of what is performed in modern OSes, with I/O in physical memory, and R/W in mapped memory.

ghaerr, 2020-03-11:
@mfld-fr : I agree with you regarding more I/O in L2. It seems certainly a bottleneck possibly waiting for one of 8 possibly locked buffers. Thus I can see need for segment allocator mod now to place L2 buffers on 1k aligned boundary.

IMO, it still makes sense to fix L1 buffer alignment now, either with method I suggested, or another better one.

I still think it worthwhile to review assumption that adding far pointers highly impacts existing filesystems... I will take a closer look at that. Now that we have better compiler, and that far memcpy's are already used to transfer data in and out of user space, keeping all buffer heads near but just buffer contents far may not be that impactful.

ghaerr, 2020-03-11:
> keeping all buffer heads near but just buffer contents far may not be that impactful.

@mfld-fr : Quickly I am starting to see your thoughts. I suppose you're saying that the impact is that all filesystem code would have to use far pointers, thus greatly increase code space used? I agree.

Perhaps best option is near what you suggested: move I/O to use L2 buffers, if no L1 buffer available. Then copy for map_buffer.

commit a9eecd4e (2020-03-11): [kernel] Swap back code and data segment

Allow dynamic extent of data segment.
Restore DMA bouncing buffer below 64K boundary.

Close #440
closed 2020-03-11 by ghaerr