ghaerr/elks #444: TZ/localtime support

Mellvik, 2020-03-10:
Discussed in #377: Change clock.c to support/set localtime.

- Vintage PCs: RTC time = localtime: For this setting we're already OK
- Emulators: Some use UTC (QEMU), some don't (VirtualBox).
- ELKS has 'structural' TZ support, but (afaik) no tzset() [and the date command (sh_utils & busyelks) has no TZsupport].  
- clock.c has a -u option to indicate the RTC is UTC, but gettimeofday() always returns tz=0 

It seems to me we're hosed on this issue until timezones are implemented in the kernel. Am I missing something?

-Mellvik

ssieb, 2020-03-10:
The kernel doesn't do anything with timezones.  It's up to the libraries and environment variables to determine the timezone.

ghaerr, 2020-03-11:
Hello @Mellvik,

> It seems to me we're hosed on this issue until timezones are implemented in the kernel. Am I missing something?

I looked through all of this a week or so ago and I think you're correct on all counts, except, yes, the solution would be dealt with in libc, not the kernel.

> * Vintage PCs: RTC time = localtime: For this setting we're already OK

I didn't realize that vintage PC RTC was local time. Now I see how the clock code could have worked originally.

> * Emulators: Some use UTC (QEMU), some don't (VirtualBox).

I was playing around with the idea of changing the setting from UTC to local time on QEMU, but realized that caused bigger problems, and thought it better to stick with UTC.

> * ELKS has 'structural' TZ support, but (afaik) no tzset() [and the date command (sh_utils & busyelks) has no TZsupport].
> * clock.c has a -u option to indicate the RTC is UTC, but gettimeofday() always returns tz=0

I also tried recompiling clock several ways, and came to the conclusion it can't be made to work without TZ support.

I'm thinking of a potentially simple solution: put export TZ=0 in /root/.profile, so that `/bin/sh` (and soon `sash`) export TZ to the environment. Then make a quick change (I haven't looked yet, perhaps you can) to the libc `time` and `ctime` functions (used in 'ls' and 'date') to offset the time by TZ hours if found in the environment. In this way, things stay simple, but users can edit /root/.profile to set their local timezone. Leave `clock` the way it is, since it executes at init time before any other programs.

Mellvik, 2020-03-11:
Thanks @ghaerr,

I like this approach, and will make an attempt to fix it, possibly even turning it into a PR (that would be a first). It's worth keeping in mind that for MSDOS and Windows to have correct time, RTC time has to be localtime. I don't know if this applies to more recent versions of Windows, but that's less relevant.

While we're at it, there are two slightly different time.h files and this may be a good opportunity to merge or delete one (libc and linuxmt).

Finally, 
`qemu -rtc base=localtime`
will take care of the issue short term.

--Mellvik

ghaerr, 2020-03-11:
Hello @Mellvik,

The latest commits add TZ into /root/.profile, for your work.

> Finally, `qemu -rtc base=localtime` will take care of the issue short term.

As mentioned, I tried that and don't recommend it, it brings up other time complications. Leave it set to UTC when you're working on your fix will help.

Keep me posted on other libc issues like 'time.h', and I'll get them fixed all at once. (I'm running into quite a few of them myself).

tkchia, 2020-03-11:
Hello @ghaerr, hello @Mellvik,

> I'm thinking of a potentially simple solution: put export `TZ=0` in /root/.profile, so that `/bin/sh` (and soon sash) export `TZ` to the environment.

Incidentally, POSIX [says](https://pubs.opengroup.org/onlinepubs/9699919799/basedefs/V1_chap08.html) that the `TZ` variable should either start with a colon, or it should start with a standard time zone name, followed by the time offset.

We can probably parse `TZ=0` as a sort of non-standard extension, but to be anywhere near POSIX-ly correct, we probably want to also parse some simple things like `TZ=GMT0` and `TZ=JST-9`.

Thank you!

Mellvik, 2020-03-11:
@tkchia,

I will take care of this when I merge time.h-versions, one of which defines the literal TZ names.

—Mellvik

ghaerr, 2020-03-12:
Hello @Mellvik,

> I will take care of this when I merge time.h-versions, one of which defines the literal TZ names.

I'm not sure the kernel and libc time.h includes should be merged... it might be best to leave libc time.h unmodified, and just read the environment for a TZ= string, decode that and store it statically in a new routine _tzset(struct timezone *) called after `gettimeofday` in ctime.c and localtime.c. Ideally this would be handled within gettimeofday, but it's a system call.

To meet the POSIX requirements that @tkchia brought up, a simple scan of TZ for either a numeric character or '-' and calling `atoi` on it should suffice to set the struct timezone 'tz_minuteswest' field. It looks like ctime and localtime already handle any non-zero offset, so that's all that's needed.

Mellvik, 2020-03-12:
@ghaerr,
I agree. I'm proposing to merge the two userland time.h versions, in libc/ and linuxmt/ - a cleanup more than a technical thing I suppose.

-Mellvik

ghaerr, 2020-03-12:
Hi @Mellvik,

> I'm proposing to merge the two userland time.h versions, in libc/ and linuxmt/ - a cleanup more than a technical thing I suppose.

No, the file libc/include/time.h is userland, but elks/include/linuxmt/time.h is a kernel file with required kernel definitions and declarations, which should not be merged. Are you talking about a third time.h somewhere?

Mellvik, 2020-03-12:
Sorry @ghaerr,
my mistake. I wrongly assumed include/sys/time.h to be the kernel time.h .
Possibly confused by the fact that some userland commands include linuxmt/time.h.

I'll leave the header files alone :-) .

—Mellvik

ghaerr, 2020-03-18:
Hi @Mellvik,

PR #475 fixes this and a number of other issues, please test by modifying TZ= in .profile and close if everything works. `clock` will always use UTC, then time modified by TZ string to local time.

Mellvik, 2020-03-19:
Thanks @ghaerr,

this looks really good.
I suggest updating date() to show the current TZ.

-Mellvik

closed 2020-03-19 by Mellvik