ghaerr/elks #452: /dev/tty broken

Mellvik, 2020-03-11:
While working on fixes to more(), I noticed that /dev/tty is broken.
```
ls -l /dev > /dev/tty
Cannot open /dev/tty: Error 19
```
Reading from /dev/tty gives the same error.

BTW - /dev/tty should have access rw-rw-rw-. Currently it is r-xr-xr-x.

--Mellvik

ghaerr, 2020-03-11:
@Mellvik,

There is no /dev/tty, you just tried to create it with `ls ... > /dev/tty'.

We could create one, but the FAT fake /dev entries are out of space, there's only room for 16. So we would need to pick another one to delete. Take a look and let me know whether you think this is worth it. Adding it just for Minix filesystems isn't a great compatibility idea.

In the meantime, '/dev/tty1' is usually your controlling tty. (I know that's not what you're requesting here though).

Mellvik, 2020-03-11:
Thanks @ghaerr,

there is indeed a /dev/tty - character device major 5 minor 0 …

And yes, I think it's worth it - not the least for portability.

I'm using /dev/tty1 for now.

—Mellvik

ghaerr, 2020-03-11:
> there is indeed a /dev/tty - character device major 5 minor 0 …

Why don't you use the ELKS command 'mknod' to create /dev/tty and see whether it works?

If so, I'll add it to the Minix build ASAP. We would need to delete one from the FAT build, and I'm not certain which fake /dev entry there to delete yet.

Mellvik, 2020-03-11:
@ghaerr,

it's already there, so it's in the build somewhere. I didn't create it (then it would have been a regular file …)
It just doesn't work.

—M

ghaerr, 2020-03-11:
> it's already there, so it's in the build somewhere. I didn't create it (then it would have been a regular file …) It just doesn't work.

You're right, it's there, I missed it. Also noticed a note:
```
# Alternate TTY devices. These are not yet supported.

    $(MKDEV) /dev/tty   c 5 0
```

I can't see that /dev/tty has mode  r-xr-xr-x. Are you running FAT filesystem? It sets 'x' on all files, since there's no execute attribute. I show /dev/tty on Minix filesystem as 'crw-r--r--'.

Mellvik, 2020-03-11:
@ghaerr,

you're right, it's 644, not 555. Anyway - it's still wrong, should be 666, a non-issue for now since it doesn't work.

OK; so I gather /dev/tty is on someone's todo list then … :-)

—Mellvik

ghaerr, 2020-03-12:
> there is indeed a `/dev/tty` - character device major 5 minor 0 …

Yes, it is in /dev, but I just checked into this and there's no driver code at all for character major 5, the special file shouldn't even have been created.

Volunteers?

Mellvik, 2020-03-12:
Thanks @ghaerr,
I'll have a look at it. I'm surprised that isatty() seems to work

Med vennlig hilsen/best regards,
Helge Skrivervik
myMAYDAY.com - 9201 7146

Mellvik, 2020-03-12:
[I deleted an incomplete response on this issue]
@ghaerr,
I'll have a look at it. Let's keep it open for now.

--Mellvik

PR #478 (2020-03-19): [kernel] Implement /dev/tty
This implements `/dev/tty` in the kernel, as well as for FAT in /dev fake devices.

Requested by #452.

@Mellvik, if this works for you, please close #452.
closed 2020-03-19 by ghaerr