ghaerr/elks #453: csh-like command line editing/history [shell enhancement)

Mellvik, 2020-03-11:
csh-like simple command line history and editing in one of the ELKS shells would be extremely useful:
!!, !<number>, editing with ^ (substitute), :p etc.

[I'm not suggesting full bash arrow key editing, assuming it's too heavy on limited resources.]

--Mellvik

ghaerr, 2020-03-11:
Funny how both of us are running into the same desires...

Realistically, after looking at too much elkscmd/ and both `ash` and `sash` code, I don't think upgrading these programs in whole from a more current version will buy much (there aren't any I can find that continue to be small). 

That leaves us with improving them with desired smaller features (which we're doing).

That conclusion leads me to suggest that you should write this addition yourself, and close this request. Unfortunately, I'm way too busy fixing other problems to do this myself at the moment, and suffer retyping commands instead!

Mellvik, 2020-03-11:
Makes sense, @ghaerr. I'll take close look at V7 csh and figure out which way (porting csh or adding to one of the current shells) makes most sense. Csh weighs in at about 45kB filesize on Venix.

--Mellvik

closed 2020-03-11 by Mellvik

ghaerr, 2020-03-11:
@Mellvik,

Also, I meant to tell you that I have been working on `sash` and completed changes just for people like you with old floppy systems. Compile up the latest version and run `sash` after login in. Things will look identical but you'll be in the smaller shell (^D to exit, 'help' for help). I'm interested in your thoughts running `ls` and other commands, there should be NO disk access. To run the disk versions of commands, use `/bin/ls` for comparison. It's a bit tricky, but basically `sash` runs `/bin/sh` when it has to, for pipes and redirection. I'd like to know what you think its usefulness would be as a base shell for systems like yours are. (And it is very simple, look at 'sash.c', you could probably add command history, '!!' and such far easier than other shells).