ghaerr/elks #464: enhancement: `fsck` should be started at boot on un-checked filesystem

pawosm-arm, 2020-03-12:
Many other OSes do that, even as small as Fuzix.

pawosm-arm, 2020-03-12:
Currently, to do fsck on un-checked rootfs one needs to boot from floppy. It does not help anyway, as `Unable to allocate buffer for zone count` still haunts fsck. The only option is to connect the medium to full blown Linux host, and run fsck from there.

ghaerr, 2020-03-12:
The new 'proper' behavior introduced in #437 always keeps Minix filesystem superblocks updated as to their actual checked state, so users may want to reset that state with `fsck` if ELKS is terminated any other way than using `shutdown`. (`reboot` won't work, and will introduce unchecked status). Reference #437 for whether this behavior should stay, even though correct.

>  It does not help anyway, as Unable to allocate buffer for zone count still haunts fsck.

That is likely fixed by running 'chmem' on `fsck'. Try setting the dynamic data to 65520 and see whether that allows `fsck` to run. If so, we can set that at compile using @tkchia's compiler option.

> Currently, to do fsck on un-checked rootfs one needs to boot from floppy.

IMO, that's way too much work, more work needs to be done to allow `fsck` to fix filesystems while running. Not sure whether that's worth it at this time.

 > The only option is to connect the medium to full blown Linux host, and run fsck from there.

No - ELKS has its own `mfsck` that I ported. Run `mfsck -a` on the volume and that'll reset it.

pawosm-arm, 2020-03-12:
Can't find `mfsck` command. Is it built automatically with other stuff? I'm enabling everything for a HD image.

Indeed, `chmem =65520 /bin/fsck` enables `fsck` to proceed, however, as I finished running `fsck`, the system could not start after subsequent `reboot` due to `3!`.

ghaerr, 2020-03-12:
‘mfsck’ is a host tool, built with ELKS, and works well. 

Looks like the ELKS ‘fsck’ is buggy and corrupts the filesystem. Ugh.

pawosm-arm, 2020-03-12:
> ‘mfsck’ is a host tool, built with ELKS, and works well.

As I understand, it replaces /sbin/fsck.minix from util-linux?

ghaerr, 2020-03-12:
> As I understand, it replaces /sbin/fsck.minix from util-linux?

Yes. When `mfs` was being enhanced and debugged, I found a relatively recent (2006) version of Linux `fsck.minix` to test with, so that the Minix filesystem generation could be tested with code I didn't write. Looking at ELKS `fsck`, it is from the same source, but a 1994 vintage. It is likely that we could replace it with this recent version and use it for ELKS too.

If you want to try that, the `mfsck` source is at elks/tools/mfsck/mfsck.c, and fsck is at elkscmd/disk_utils/fsck.c.

ghaerr, 2020-09-01:
Hello @pawosm-arm,

fsck finally works on ELKS, fixed in #733. You're welcome to test this by including fsck and various options in the /etc/rc.d/rc.sys script, which should solve this issue for you. I don't think fsck will run automatically by default, as the time taken for this on floppy boots may not be desired for all users.

pawosm-arm, 2020-09-02:
It would require clever scripting to make it run from rc.sys. I can imagine following scenario:
- check if rootfs is mount read only
- if yes:
  - check if rootfs is marked clear (namely, does not need fsck as it was previously unmounted cleanly)
  - if no:
     - run fsck
  - re-mount rootfs read/write

This brings some questions:
- Last I've checked, the `CONFIG_ROOT_READONLY=y` option does not seem to do anything. Is it yet another dead option in `.config`?
- Is there an easy way to query ELKS if rootfs (or any other pointed filesystem) is marked clear?
- Can `mount` command be used to remount rootfs?

ghaerr, 2020-09-02:
>  I can imagine following scenario

Yes, and no easy way to perform those checks (unless fsck were modified to do all the work), along with the need to mount root readonly at boot and remount rw every time. For floppy users, this would considerably slow boot times.

> the CONFIG_ROOT_READONLY=y option does not seem to do anything. Is it yet another dead option in .config?

I'll have to check. When I implemented the /bootopts "ro" option, which mounts root read only now, I found it doesn't work - VFS mounts root readonly and the system promptly seems to stop running. Its on my long list of things to investigate.

> Is there an easy way to query ELKS if rootfs (or any other pointed filesystem) is marked clear?

No. It could be possible to add that as an option to `fsck` and have it return an exit value that could be tested in shell script, or just run fsck all the time at boot, and do everything required, since it would have to run again anyways.

> Can mount command be used to remount rootfs?

Not currently. The flags for ELKS mount system call are nonstandard and would have to be modified to take a MNT_UPDATE flag along with a separate flags parameter.

The mere act of booting ELKS will leave the filesystem in an unchecked state, unless the reboot or shutdown command is used before shutdown, which I'm guessing is not frequent. In addition, there is no "auto-sync" capability yet, so system buffers are never flushed until a manual sync. All of that, and given that fsck has never been tested as to its actual ability to repair filesystems well, lead me to believe that ELKS isn't quite ready for this feature. When/if people start using ELKS on hard drives, all of this is probably warranted though.

ghaerr, 2020-09-02:
> Indeed, chmem =65520 /bin/fsck enables fsck to proceed, however, as I finished running fsck, the system could not start after subsequent reboot due to 3!.

You might try re-testing fsck. It was buggy on all inodes numbers that were a multiple of 32, and reported them as having mode 000000. It probably deleted them which caused the system to not boot.

I have tested running fsck on a mounted filesystem (not in repair mode), and it seems to report all is well. There's more work to be done getting this whole issue moved forward, but we should probably test fsck's repair facilities first to make sure its even worth it.

pawosm-arm, 2020-09-02:
> For floppy users, this would considerably slow boot times.

No one would ever consider it for floppy-booted systems.... ELKS configure can select HD vs. floppy image profiles.

> You might try re-testing fsck. It was buggy on all inodes numbers that were a multiple of 32, and reported them as having mode 000000. It probably deleted them which caused the system to not boot.

I've booted ELKS floppy and turned off the machine, so any subsequent attempt to mount the floppy in HDD-booted ELKS resulted in `fsck is recommended` message. So I ran `fsck /dev/fd0`. It did complete (listing all of the files and directories on floppy), but still subsequent mount attempts resulted in `fsck is recommended` message, as if `fsck /dev/fd0` did nothing.

> There's more work to be done getting this whole issue moved forward, but we should probably test fsck's repair facilities first to make sure its even worth it.

Yes, indeed.

pawosm-arm, 2020-09-02:
As a side note, fortunately, the floppy is still bootable after that.

ghaerr, 2020-09-02:
> So I ran fsck /dev/fd0. It did complete (listing all of the files and directories on floppy), but still subsequent mount attempts resulted in fsck is recommended message, as if fsck /dev/fd0 did nothing.

I'm pretty sure you have to run fsck -r (or -a for automatic) to have it actually write anything, including the valid superblock flag, even if the image is found to be OK. The filesystem valid flag is kept separately in the superblock and is always set to invalid on any mount, regardless of other filesystem state.

> No one would ever consider it for floppy-booted systems.... ELKS configure can select HD vs. floppy image profiles.

Agreed. Except I'm not sure how you select separate image profiles, other than application groups. Currently, there isn't any change made to /etc/rc.d/rc.sys between floppy and HD systems, for instance.

If you and/or @Mellvik can confirm that fsck can in fact be made to repair and/or set the valid flag on a filesystem, it would probably be worth looking into getting ELKS to boot with root read-only, and then having fsck run on hd images in rc.sys, then remounting the filesystem read-write once the filesystem is repaired or verified.

pawosm-arm, 2020-09-03:
> I'm pretty sure you have to run fsck -r (or -a for automatic) to have it actually write anything, including the valid superblock flag, even if the image is found to be OK.

Indeed, `fsck -r /dev/fd0` has set the clear flag and subsequent mount succeeded without a warning. I wonder if it could be a default...

ghaerr, 2020-09-03:
> fsck -r /dev/fd0 has set the clear flag and subsequent mount succeeded without a warning. I wonder if it could be a default...

Although it happens to work for you likely just after boot, we definitely can't just run fsck -r on a mounted filesystem... it would render the kernel buffers and disk image out of sync if any changes were made, including superblock changes. It would have to be run on the device with the boot volume mounted read-only as you previously mentioned, then possibly remounted read-write afterwards.

And then there's the issue of not running on floppy boots but running on HD boots, and somehow getting the name of the root volume, etc.

More testing needs to be done on wrecked filesystems, do ensure that fsck -r repairs well enough to trust it with a half-written filesystem on boot doing less damage than allowing the filesystem to be mounted unclean. An unrequested, untested fsck could delete lots of otherwise possibly important files on a production system.

Mellvik, 2020-09-03:
@ghaerr, @pawosm-arm

the first test of fsck -r is a pass. 
The test is as follows:

fire up qemu with two floppies.
# mkfs /dev/fd1 2880
# mount /dev/fd1 /mnt
# cp /bin/* /mnt
# ls /mnt
ctrl-alt-del or equivalent
after reboot, fsck -rsl /dev/fd1
- a zillion messages most of which needs a ‘y’ to continue (suggestion: add a ‘-y’ option to answer test to all interactive questions).
FILSYSTEM WAS MODIFIED
New fsck and the file system is clean.

The next test is a fail - still the same qemu with 2 floppies:
# mount /dev/fd1 /mnt
fd: probing disc in /dev/fd1
fd: /dev/fd1 probably has 18 sectors, 2 heads, and 80 cylinders
# mkdir /mnt/bin
# mkdir /mnt/etc
# cp /bin/* /mnt/bin
# sync
# cp /etc/* /mnt/etc
/etc/rc.d: Is a directory 
Failed to copy /etc/rc.d -> /mnt/etc/rc.d
# ls /mnt/etc
group    hosts    inittab  issue    motd     passwd   profile  rc.d   
ctrl-alt-del

After reboot a fsck -rsl
lists all files and directory names twice, then complains after listing the etc directory, asks questions about zones:
…
Mode 0 0 0 0
uid 0 size 0 gid 0 links 1
Z0 Z0 Z0 Z0 Z0 Z0 Z0 Z0 Z0 
Mode 0 0 0 0
uid 0 size 0 gid 0 links 1
Z0 Z0 Z0 Z0 Z0 Z0 Z0 Z0 Z0 
Zone 240: marked in use, no file uses it.Unmark (y/n)? y
Zone 241: marked in use, no file uses it.Unmark (y/n)? y
Zone 242: marked in use, no file uses it.Unmark (y/n)? y
Zone 243: marked in use, no file uses it.Unmark (y/n)? y
Zone 244: marked in use, no file uses it.Unmark (y/n)? y
Zone 245: marked in use, no file uses it.Unmark (y/n)? y
Zone 489: marked in use, no file uses it.Unmark (y/n)? y

…
- the finishes up - and all files and directories are gone. 

Tests are easily repeatable.

No cigar, but we’re getting there.

—Mellvik

ghaerr, 2020-09-03:
Hi @Mellvik,

Thanks for running these tests!

> add a ‘-y’ option to answer test to all interactive questions

fsck -a will do that. See the documentation for fsck options in https://github.com/jbruchon/elks/pull/734#issue-478259241.

For the second (failed) test, the entire inode table and likely the /mnt/etc directory are unsynced, but the /mnt/bin directory was synced and seemingly shouldn't have been removed.

> lists all files and directory names twice

This should have been fixed in my latest commit, where I found that -l and -v together interacted badly. We now can use -lv as well as -vv, for more in-depth information.

We are running a very old version of fsck which has been modified to run on systems without large buffers (non 32-bit). It may not be easy to port a more modern version, and that's certainly lower priority on my schedule. Thus, fsck's primary usefulness for ELKS for now is to ensure integrity, but may not be the best program to repair a filesystem. Of course, a modern Linux version of fsck.minix could be used to repair an important volume, that could also be tested to see whether any improvements have been made.

So - running fsck -a/-r could be very dangerous at boot.

Mellvik, 2020-09-03:
> 3. sep. 2020 kl. 18:05 skrev Helge Skrivervik <helge@mymayday.com>:

> (suggestion: add a ‘-y’ option to answer test to all interactive questions).
Correction - my spelling helper kicked in again:

-y option to automatically answer yes to interactive questions.

--M

ghaerr, 2020-09-03:
@pawosm-arm,

I've added the ability to boot read-only in #735.

Also, I saw that ELKS already allows `umount /` to unmount a read-write filesystem and remount it read-only.

I'm still working on the final part, which is remounting a read-only filesystem to read-write. I think that can probably be done with a new `mount` option, something like `mount -o remount,rw /`. While we're at it, probably add `mount -o ro ...` as well to allow read-only filesystem mounts.

Mellvik, 2020-09-03:
It may not be appropriate,
But full applause from this end.

M

ghaerr, 2020-09-03:
@pawosm-arm,

PR #736 adds the ability for you to change the read/write status of the root filesystem so that fsck can be run. You might play around a bit and see whether a shell script can be devised that solves this issue, as should this be included in system startup, it would run in the rc.sys script. The return status of fsck could be used to leave the filesystem in a read-only state, since we're seeing it might be too dangerous to run fsck -a automatically.

Something like the following should work for you:
```
root=/dev/fd0
mount -o ro $root /
fsck $root
mount -o remount,rw $root /
```

ghaerr, 2020-09-04:
A couple more comments about the new `mount` remount option:

The device name is ignored by the kernel on remounts, only the mounted directory is used. This means that the root device name isn't actually required, vis-a-vis to remount root without knowing device name the following can be used:
```
umount /                              Umounts root with auto-remount root read-only
...
mount -o remount,rw any /             Remounts root read-write
```
I haven't yet figured a way to determine the root device name or number for running fsck in between, however.

pawosm-arm, 2020-09-04:
Some observations from today's master:

- `CONFIG_ROOT_READONLY=y` in `.config` still doesn't seem to have any effect on mounted rootfs...
- `umount /` which umounts rootfs with auto-remounting it read-only is very useful and I've found it a good workaround for the problem above
- `shudtown` start to behave weirdly... after 'halting' the system, `login:` prompt is printed, after logging-in it turns out, the rootfs is mounted read-only, so the machine can be powered down safely
- `chmem -h 65520 /bin/fsck` was needed again! I guess it's because this time I started `fsck -r` on `/dev/bda`, not on `/dev/fd0`.
- `touch something` when started on read-only filesystem did not warn that non-existing file `something` wasn't created!

Other than that, `fsck -r` succesfully ran on read-only mounted unclean rootfs marking it clean hence subsequent `remount,rw` completed without a warning.

ghaerr, 2020-09-04:
Thanks @pawosm-arm for your testing, interesting set of items you've found.

> * `CONFIG_ROOT_READONLY=y` in `.config` still doesn't seem to have any effect on mounted rootfs...

Another dead option - will fix, now that ro mount works.

> * `shudtown` started to behave weirdly... after 'halting' the system, `login:` prompt is printed, after logging-in it turns out, the rootfs is mounted read-only, so the machine can be powered down safely.

Looking into this, I'm not sure how shutdown could have ever worked... the bug is that /bin/init is protected from signals, including SIGKILL, which is sent to all processes in a shutdown. So all processes except init are killed, and /bin/init rescans /etc/inittab and restarts the getty's, thinking its another login cycle. Will fix.

> * `chmem -h 65520 /bin/fsck` was needed again! I guess it's because this time I started `fsck -r` on `/dev/bda`, not on `/dev/fd0`.

Missed this before. Yes, fsck uses a recursive algorithm for checking so it needs more stack space.

> * `touch something` when started on read-only filesystem did not warn that non-existing file `something` wasn't created!

Yep, looks like touch doesn't print anything on a create error. Will fix.

> Other than that, `fsck -r` succesfully ran on read-only mounted unclean rootfs marking it clean hence subsequent `remount,rw` completed without a warning.

Good to know the fsck part of your testing actually worked :)

ghaerr, 2020-09-04:
@pawosm-arm,

PR #738 fixes all the above problems, except CONFIG_ROOT_READONLY. Long story short, fsck was totally broken on large filesystems, I have been testing on the ELKS 32MB hd32-minix.bin and it now all works. `shutdown` now also works properly, I have no idea how it could have used to work, since /bin/init was never killed. 

The first mentioned issue, CONFIG_ROOT_READONLY was not fixed, since there is no such option. The CONFIG_FS_RO option, currently in config, compiles the kernel to only include read-only filesystem support, which is not quite the same thing as you're asking about, but will work to mount all filesystems read-only. Otherwise, use /bootopts ro option.

Thanks for the bug report. Trying to duplicate the fsck chmem issue required getting a large filesystem and that ended up taking quite a while to figure out. All should be working now!

ghaerr, 2020-09-05:
Hello @pawosm-arm,

PR #739 implements everything requested in this issue. Filesystem checking is implemented in the rc.sys shell script, which should allow for maximal flexibility for varying user requirements.

Please test and if satisfied, this issue can be closed.

pawosm-arm, 2020-09-05:
Hey @ghaerr, `fsck` needs `-a` option in `rc.sys`. When added, it does what I expected, the objective of this ticket is fulfilled.
the `ROOTDEV` trick seems very neat, although I'm wondering about one more variable, the `ROOTFSTYPE` which could be checked in `rc.sys`, so the call to `check_filesystem` function would not have to be commented out.

Also `System halting` message seems confusing, the previous `System halted` seemed more definitive. When I see `System halting` printed on screen, I start to wonder, how long it would take for the system to be actually halted...

The `touch` command finally prints what it should.

> The first mentioned issue, CONFIG_ROOT_READONLY was not fixed, since there is no such option. The CONFIG_FS_RO option, currently in config, compiles the kernel to only include read-only filesystem support.

Somehow I'm seeing both of those options within just generated `.config`:

``
# CONFIG_ROOT_READONLY is not set
# CONFIG_FS_RO is not set
```

...and there are still two subsequent options for the above listed in `menuconfig`.

ghaerr, 2020-09-05:
> `fsck` needs `-a` option in `rc.sys`. When added, it does what I expected, the objective of this ticket is fulfilled.

Agree, but currently think it too dangerous to default run fsck -a, given @Mellvik's finding that many files could be lost.
I don't want ELKS be responsible for automatically deleting HD contents on a user hard drive during development.

> the `ROOTDEV` trick seems very neat, although I'm wondering about one more variable, the `ROOTFSTYPE` which could be checked in `rc.sys`, so the call to `check_filesystem` function would not have to be commented out.

The problem is that we all of this occurs way before a filesystem is mounted, so the fs type isn't known. I'm already concerned about having adding the above code to the kernel, although small, as we're short on space. Rather than having the kernel figure it out, the rc.sys script is easily editable for system customization. If /bootopts is used, any environment variable can be passed to rc.sys as documented, so system customization can be done without editing rc.sys and regenerating a root filesystem.

That said, if we can figure out a way to determine ROOTFSTYPE, it could be added. We could keep going in this direction - it would also be neat to automatically mount a second drive filesystem found at boot, which would require somehow knowing its type as well. The kernel should not be involved in figuring all this out though.

> 
> Also `System halting` message seems confusing, the previous `System halted` seemed more definitive. When I see `System halting` printed on screen, I start to wonder, how long it would take for the system to be actually halted...

I'll change it back - but I also thought it strange that the system says "System halted" while it clearly is still running (killing processes, which, if networking was running emit more messages). There is no easy way to print the halted message after the process killing, due to the way the kernel operates.

> > The first mentioned issue, CONFIG_ROOT_READONLY was not fixed, since there is no such option.
> ...and there are still two subsequent options for the above listed in `menuconfig`.

Thanks, I got the options confused (again). I'll fix it.

pawosm-arm, 2020-09-05:
> The problem is that we all of this occurs way before a filesystem is mounted, so the fs type isn't known.

The only reason the call to `check_filesystem` function is commented out is that `fsck` does not support FAT. IMO it would be easier to uncomment it and make `fsck` print that it won't check filesystem it doesn't currently understand.

ghaerr, 2020-09-05:
>  IMO it would be easier to uncomment it and make fsck print that it won't check filesystem it doesn't currently understand.

It will already do that, and print Bad superblock, or some other possibly confusing message to some users. My concern is that we're making big assumptions about what is being booted without looking at the config file. ELKS supports lots of different boot and filesystem mechanisms at this point (consider ROMFS). I doubt we can reach a one-size-fits-all rc.sys without user customization. Better would be a customized rc.sys for the configuration selected, but we're not there yet.

> The only reason the call to check_filesystem function is commented out is that fsck does not support FAT.

Not quite - when we get feeling good about running fsck -a, things could possibly get ugly should that be run on a FAT filesystem.

Since you have a working MINIX HD system, it might be worth playing around a bit - try writing the filesystem with copies like @Mellvik has suggested might  break the filesystem with unsynced power-offs, and automatically run fsck -a on reboot. I'm interested to know whether we can realistically achieve what you're actually looking for. All of this can be tidied up once we've had more complete testing.

pawosm-arm, 2020-09-05:
@ghaerr with #741 the objectives of this tickets are fulfilled and I'd close this one, while new ticket ought to be opened for other `fsck` issues we encountered.

closed 2020-09-05 by pawosm-arm

ghaerr, 2020-09-05:
Thanks @pawosm-arm, let me know what further issues you encounter on testing fsck on another ticket. We are making progress!