ghaerr/elks #465: enhancement `miniterm -s` should understand baud rates faster than 57600

pawosm-arm, 2020-03-12:
It looks like `miniterm -s` cannot parse baud rates that cannot fit into 16-bit unsigned integer. Larger values are clamped to 16-bit values. Namely, 115200 is clamped to 49664 while 230400 is clamped to 33792. I suspect it's `atoi()` vs `atol()` case (that matters on 16-bit architecture), or something similar.

ghaerr, 2020-03-12:
Thanks for pointing out the actual problem. @Mellvik mentioned this in #454 and said he's going to fix it.

ghaerr, 2020-03-18:
Hello @pawosm-arm and @Mellvik,

PR #473 fixes this issue, please test and close this PR if satisfied.

pawosm-arm, 2020-03-18:
Yeah, it reads speeds correctly now.

One thing I noticed, when I close the connection by typing `~.` the shell session is also closed. Was this intended?

ghaerr, 2020-03-18:
> One thing I noticed, when I close the connection by typing ~. the shell session is also closed. Was this intended?

I don't know, I've never actually used `miniterm`. Are you saying that the ELKS local shell exits back to login? It is normal that closing a remote connection's OS should kill the remote shell, if the connection is using more than a three-wire serial connection to the remote.

pawosm-arm, 2020-03-19:
> Are you saying that the ELKS local shell exits back to login?

Yes, it's the local ELKS shell that exits, which is more odd than annoying.

ghaerr, 2020-03-19:
> Yes, it's the local ELKS shell that exits back to login, which is more odd than annoying.

Strange. I wonder if it has to do with the ongoing bugs in serial line signal handling. If the shell somehow became part of the process group of the serial port, and a SIGHUP improperly sent, the shell would get killed. Does this occur only after making a connection? Please send the exact command line and/or commands you're using, so I can duplicate it.

ghaerr, 2020-03-19:
@pawosm-arm, I duplicated this last miniterm problem and fixed it in PR #476. Hopefully this is the end of `miniterm` problems!

pawosm-arm, 2020-03-19:
Verified, now it's working good.

closed 2020-03-19 by pawosm-arm