ghaerr/elks #469: [kernel] Improve the segment allocator

mfld-fr, 2020-03-14:
Preview of upcoming changes related to segment allocator in commit 420adfc2:
- segment descriptors in local heap
- segment allocator code simplified (no more peek / poke)
- `seglist` code removal

ghaerr, 2020-03-15:
Hello @mfld-fr,

Thank you for offering a preview. I have studied the code quite a bit and have the following observations and thoughts:

__Summary__
While it is exciting to bring local heap management into ELKS, using local heap to hold descriptors for global memory is not the place for it. It is mentioned in the code that this is done for speed, and also for "when 286 protected mode returns someday". I will explain later in more detail why the former can be made even quicker with a design change, and the return of 286 mode is not a good reason, all the code was thrown out for that last year. The local heap will become needlessly fragmented by separating the header from the storage. The header should be kept contiguously with the allocated data for both local and global allocations. The preview code requires at least four major sections of kernel global allocation to be rewritten, but there is no need for that.

The rewritten portions of mm_alloc, etc getting rid of `pokew` are a good thing, along with defining structures for normal C access. However, likely because thoughts of far pointers being slower, the designed has changed from the elegant (but non-portably coded) earlier implementation of mm_alloc, to the overly complicated proposed version, which requires extra dereferences. Let me explain shortly.

Jumping to my conclusion, I can see the need for a local heap manager, *and* a global heap manager in the kernel. As I think deeper into it, I realize that these could and should be accessible identically to the kernel programmer. With both local and global heap allocators available, solutions become simpler - the mm_alloc rewrite absolutely should use the global heap manager, which will also manage the L2 buffers. Neither of these uses should require space in the local heap (except for buffer headers, something different).

__Details__
The programmatic API for both heap managers should operate as easily as possible for kernel programmers. That is, the main allocation routine should return an immediately usable pointer, *not* a structure pointer that needs another dereference. To wit:
```
/* C local heap*/
char *p = malloc(256);     /* get pointer to memory*/
free(p);       /* free uses allocated ptr and subtracts fixed amount to get private hdr*/
```
The old global kernel allocator mm_alloc operated using the above approach, where an immediately usable value was returned, which also very nicely happened to be the 8086 segment value. In this preview, sections of 'elks/fs/exec.c' and others are being rewritten to dereference off a local-heap structure to get the required segment value needed by the kernel routine. Why? It's slower, not faster. But that's not really an issue. The preview method requires more steps to check for error, and should just return what the routines need: a 16-bit segment value:
```
/* old code*/
seg_t seg = mm_alloc(block);
if (!seg) goto error;
currentp->t_regs.ds = seg;

// new code (not quicker and requires local heap and redirection, both unneeded
// new typedef segment_t required
segment_t *seg = seg_alloc(block);
if (!seg) goto error;
 currentp->t_regs.ds = seg->base;
```

The seg_alloc replacement global memory allocator should return a usable 16-bit 8086 segment value 'seg', and the private header for that memory (using the preview rewritten C structure) should access that header as 'seg-1' internally. 'segment_t' should remain private to the allocator, and not be known global to the kernel.

The preview code requires rewriting kernel code that works and does not need to be rewritten (example 'exec.c'), provided that the allocated memory is global. There is no need for global heap information in the local heap - it can easily be kept in the previous 16 bytes, and exec.c doesn't need to be rewritten, and the local heap doesn't get littered with very small allocations that are only to be used to manage global memory anyways.

__More Details__
Now that the local heap manager is basically working, if a second, global heap manager were also immediately implemented, it could be made almost identical, using the same C structure for management. Both heap allocators would return a 16-bit value, and hide the private heap structure from the kernel programmer. The global allocator would return an actual segment (half a far pointer) seg:0, while the local allocator would return an offset into the local heap DS:offset.

Now - if the doubly-linked list portion of the private headers used the exact same structure, that is, far pointers, and the reference count were moved up to just beyond the two list pointers, more benefits would be realized:
```
// common private heap header structure for global and local heap allocations
// stored 16 bytes prior to returned pointer
struct heap_hdr {
        // common portion for all lists
 	list_s    node;              // doubly-linked far pointers
        word_t ref_count;      // reference count for atomic operations
        // common portion for memory allocators
 	word_t   flags;
 	word_t   size;             // # segments for global heap, # bytes for local heap
        word_t   extra;           // offset 14-15
};
```

Only one set of linked list routines need be included, instead of having two, the private headers used could be identical, the atomic routines used on all lists or heaps would be identical (provided they were passed a far pointer to heap_hdr). Both allocators would return 16-bit values, either a segment, or an offset into k [... cut ...] ts, and matches the structure of a standard local allocator, not requiring additional dereferencing. If far pointers are used for list management, a common header could be used, which would allow shared routines for list linking and atomic operations with no overhead outside the allocator routines themselves, and no requirements for far pointers in the general kernel.  Both allocators would return 16-bit values that are easily usable without using far pointers: global SEG:0, and local DS:OFFSET.

mfld-fr, 2020-03-15:
I thought the same when implementing the 'scattered list allocator' some times ago, but I realized it cannot work in protected mode or, more important, with banked memory above the 1MB barrier.

Merely because you cannot traverse such list when the memory is not mapped in the processor address space, whatever the underlying memory management system (segmenting, paging or switching).

There are other reasons that are obvious in my staging code but not  apparent in such piece, when trying to implementing the memory reclaim algorithm. One could read chapter 17 of the 'understanding the linux kernel', 3rd edition, to get an idea why locality is needed.

I agree there are unnecessary impact in some parts (the copy to / from user space) because the existing code was using the 'mm' structure of the task, and not the 'regs' to get the process active segments (so already mapped). Will rework them to not impact out of global memory management.

ghaerr, 2020-03-15:
>  it cannot work in protected mode or, more important, with banked memory above the 1MB barrier. Merely because you cannot traverse such list when the memory is not mapped in the processor address space, whatever the underlying memory mechanism (segmenting, paging or switching).

My analysis and response were entirely based on the assumption that ELKS would be running in  addressable real mode, I can see why a more complicated scheme is needed for non-mapped address space traversal, but many more changes would be required for EMS, etc.

I wasn't aware that ELKS should be considering such a future at this time - are we trying to prepare ELKS to run on 286 and 386 processors? There are other operating systems for that. 

I think a more important priority for ELKS is to get it to run well given small addressable memory constraints, including better buffer management, dyanamic allocation of kernel data segment resources, etc. For that, it doesn't need to mix with a global memory manager rewrite, it only needs a local heap allocator. Why change global memory allocation routines at same time, I thought your current project was concerned with getting local heap working?

I suggest put off changes for protected and > 1MB address access until later or as separate project, not mixing with local heap implementation. In this way, less kernel routines are affected and heap allocator effectivity can be measured.

> There are other reasons that are obvious in my staging code but not apparent in such piece, when trying to implement the memory reclaim algorithm. One could read chapter 17 of the 'understanding the linux kernel', 3rd edition, to get an idea why locality is needed.

Can you explain a bit further, I don't have that book. I'm interested in how the reclaim algorithm requires this extra complexity.

mfld-fr, 2020-03-15:
Book is available here : http://gauss.ececs.uc.edu/Courses/e4022/code/memory/understanding.pdf

mfld-fr, 2020-04-04:
Closed by #472

closed 2020-04-04 by mfld-fr