ghaerr/elks #470: Latest build not bootable

Mellvik, 2020-03-15:
Unless I'm doing something stupid, the latest build is not bootable.

```
...
fd: probing disc in /dev/fd0
fd: floppy drive autoprobe failed!
hd: error: AX=0x09
BIOSHD: I/O error
dev: 300, sector: 2
...
```
Tried with several physical floppies. 
Yesterday's build worked fine.

--Mellvik

ghaerr, 2020-03-15:
Hello @Mellvik,

It looks like ELKS is trying to boot from dev 300, which is HD not floppy. @mfld-fr's modification to setup.S, where the BIOS boot device is passed from the boot loader, may have broken that,  is the only commit made in the last two days.

Mellvik, 2020-03-15:
Thanks @ghaerr,

does that analysis hold up with the fact that the image is bootable in QEMU?

Anyway, I'm looking forward to a fix - this is getting interesting. (re. test report #471).

—Mellvik

ghaerr, 2020-03-15:
@Mellvik,

I am mistaken. Yesterday's commit #467 did not touch setup.S. Instead, I was thinking of #445, from four days ago, which corrects a DMA segment boundary issue, and did modify 'setup.S'. You might try backing either of these out and seeing whether your HW floppy boot works or not. @mfld-fr can probably help you with instructions for how exactly to revert either of these on your local repo, I don't know git well enough, I'm sorry.

tkchia, 2020-03-15:
Hello @Mellvik, hello @ghaerr,

For some reason I could not reproduce the problem, after a `make clean && make` command.  Perhaps --- once again --- something wonky is going on with the dependency checking, which is causing some files not to be recompiled properly?

Thank you!

Mellvik, 2020-03-15:
Hi @tkchia,
Are you saying you could reproduce the problem before the make clean?

Anyway - I did a `make clean; make` to be sure, and the problem was still there ...
An image from 4 days back boots fine.

--Mellvik

mfld-fr, 2020-03-15:
`git clean -xdf`

Mellvik, 2020-03-16:
Thanks, @mfld,

I did all that. Even rebuilt tools and pulled a new copy of the entire tree. Still no boot (and to make sure, this is a different machine from yesterday):

mfld-fr, 2020-03-16:
Then, `git bisect` please

Mellvik, 2020-03-16:
OK:

usage: git bisect [help|start|bad|good|new|old|terms|skip|next|reset|visualize|view|replay|log|run]

Please advise.

M

mfld-fr, 2020-03-16:
https://git-scm.com/docs/git-bisect

And of course, don't forget to `./clean.sh` before building after each bisect step.

ghaerr, 2020-03-16:
Can you take a screenshot of the entire boot sequence displayed?

There are three recent commits that change setup.S, which are suspect because the boot error shows device 300 rather than 380:
```
BIOSHD: I/O error
dev: 300, sector: 2
```
These are a08561e642e60aec3ff1ac6ae6d3c8bf4dbf40db (DMA SEG),
eb5cedde084a92e05481d2b2003d9c4cd3825e16 (ELKS garbage displayed),
a9eecd4efb70e733015fba55e9185d9026593a15 (swap back code and data).

There is one commit that sets the tracks count going into boot:
3be8084e54040a6897ec8738958b2bb09859a30d (update set boot numtracks).

IMO one of these could be the the reason for not booting.

Mellvik, 2020-03-16:
Seems like the pic was filtered in the previous message.
[image]

Mellvik, 2020-03-16:
> These are a08561e <https://github.com/jbruchon/elks/commit/a08561e642e60aec3ff1ac6ae6d3c8bf4dbf40db> (DMA SEG),
> eb5cedd <https://github.com/jbruchon/elks/commit/eb5cedde084a92e05481d2b2003d9c4cd3825e16> (ELKS garbage displayed),
> a9eecd4 <https://github.com/jbruchon/elks/commit/a9eecd4efb70e733015fba55e9185d9026593a15> (swap back code and data).
> 3be8084 <https://github.com/jbruchon/elks/commit/3be8084e54040a6897ec8738958b2bb09859a30d> (update set boot numtracks).

I'm totally blank on this.

1) How do I find the commit# of the copy that works (5 days old)?

2) How do I revert to a specific commit? I tried gir reset <commit id>, but the result didn't compile.

—Mellvik

mfld-fr, 2020-03-16:
To reset your HEAD / index / working tree to a previous commit : `git reset --hard <commit>`

ghaerr, 2020-03-16:
Thanks @mfld-fr. @Mellvik, if you reset the git HEAD to just before each of the above 4 commits, I think we should find this fairly quickly.

ghaerr, 2020-03-16:
> 1) How do I find the commit# of the copy that works (5 days old)?

Select "<code>" in the top left, then after that "2131 commits below it". That will show all the commits. On the right of each, there's a "copy clipboard" button that will copy the commit number for the git reset command. Select the commit BEFORE the one you suspect is bad.

Mellvik, 2020-03-17:
> Select "" in the top left, then after that "2131 commits below it". That will show all the commits. On the right of each, there's a "copy clipboard" button that will copy the commit number for the git reset command. Select the commit BEFORE the one you suspect is bad.

OK; I have (seemingly) bisected myself back to what appears to be the culprit:

```
helge@iMac27 elks % git bisect bad
Bisecting: 0 revisions left to test after this (roughly 0 steps)
[a9eecd4efb70e733015fba55e9185d9026593a15] [kernel] Swap back code and data segment
helge@iMac27 elks % git status
HEAD detached at a9eecd4e
You are currently bisecting, started from branch 'master'.
  (use "git bisect reset" to get back to the original branch)

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	elks/include/linuxmt/heap.h
	elks/include/linuxmt/list.h
	elks/lib/heap.c
	elks/lib/list.c
	elkscmd/sys_utils/mouse.c

nothing added to commit but untracked files present (use "git add" to track)
```

I have no idea why, but this commit does not show in the list I get via the web interface. 

Hopefully this will give you an indication as to where the problem is.

—Mellvik

ghaerr, 2020-03-17:
Hello @Mellvik and @mfld-fr,

@Mellvik, thanks for doing the bisect.

@mfld-fr, I had been thinking that this bug had to do with the BIOS drive number not being brought over properly, since @Mellvik initially reported the error as 'BIOSHD: I/O error dev 300'. But looking at the photograph, it is actually using dev 380, which is correct. So could this failure to boot just be as simple that @Mellvik's BIOS can't handle I/O at DMASEG and requires BUFSEG?

Wasn't DMASEG originally added a while back and then subsequently removed? Perhaps this is the reason.

tkchia, 2020-03-17:
Hello @ghaerr,

> So could this failure to boot just be as simple that @Mellvik's BIOS can't handle I/O at DMASEG and requires BUFSEG?

But this does not make much sense.  The old setup was essentially loading sectors at `0x80:`_some-offset_ (because the data segment was set at `0x80:0`), while the new setup loads sectors at `0x80:0` (= `DMASEG:0`).

Hello @Mellvik,

How about this...  I wrote a small patch to add further information to the `hd:` error message:

```
diff --git a/elks/arch/i86/drivers/block/doshd.c b/elks/arch/i86/drivers/block/doshd.c
index e5d122f..eca0ce6 100644
--- a/elks/arch/i86/drivers/block/doshd.c
+++ b/elks/arch/i86/drivers/block/doshd.c
@@ -656,7 +656,7 @@ static void do_bioshd_request(void)
     sector_t start, count, tmp;
     int drive, errs;
     unsigned int cylinder, head, sector, this_pass;
-    unsigned short int minor;
+    unsigned short int minor, in_ax, out_ax;
 
 #if 0
     int part;
@@ -753,16 +753,18 @@ static void do_bioshd_request(void)
 		debug5("cylinder=%d head=%d sector=%d drive=0x%x CMD=%d\n",
 		    cylinder, head, sector, drive, req->rq_cmd);
 		debug1("blocks %d\n", this_pass);
-		minor = 0;
+		in_ax = BD_AX;
+		out_ax = 0;
 		if (call_bios(&bdt)) {
-		    minor = BD_AX;
+		    out_ax = BD_AX;
 		    reset_bioshd(drive); /* controller should be reset upon error detection */
 		}
 		dma_avail = 1;
 		wake_up(&dma_wait);
-	    } while (minor && --errs);	/* On error, retry up to MAX_ERRS times */
-	    if (minor) {
-		printk("hd: error: AX=0x%02X\n", minor >> 8);
+	    } while (out_ax && --errs);	/* On error, retry up to MAX_ERRS times */
+	    if (out_ax) {
+		printk("hd: error: out AX=0x%04X in AX=0x%04X "
+		       "ES:BX=0x%04X:0x%04X\n", out_ax, in_ax, BD_ES, BD_BX);
 		end_request(0);
 		goto next_block;
 	    }
```

Perhaps you can try to rebuild the system after applying this patch, and see what the system says?

Thank you!

Mellvik, 2020-03-17:
@tkchia,
here it is - and FWIW - this is a different machine, now it's a 286.

[image]

--Mellvik

tkchia, 2020-03-17:
Hello @Mellvik,

This is very strange.  Do you get the same error when you run the ELKS build under a virtual machine (e.g. QEMU)?  Also, perhaps you can send us a copy of the disk image that was built?

It is strange that the same error shows up on so many different machines.

Thank you!

Mellvik, 2020-03-17:
@tkchia,

yes, very strange, and no, the image boots fine on QEMU.

FWIW:
The QEMU boot never says
`fd: probing dies in /dev/fd0`
but rather right on to
`fd: found valid ELKS disk parameters on /dev/fd0 boot sector`

So it seems to me that there is something in the probing that fails.

I'll upload the disk image. 

—Mellvik

Mellvik, 2020-03-17:
Image ...
[fd1440.bin.gz](https://github.com/jbruchon/elks/files/4344532/fd1440.bin.gz)

ghaerr, 2020-03-18:
Hello @tkchia and @mfld-fr,

I would like to know your thoughts on what best to do regarding source tree given this boot failure:  revert the commit (#445), continue to commit outstanding PRs regardless, or nothing. I don't understand #445 well enough to know whether it stands on its own and could be reverted and introduced later for testing, or is required for subsequent changes to 'setup.S'. 

If #445 could be temporarily removed, and allowing other outstanding PRs to be committed, that would seem to be a good choice, to allow other development to continue. However, #445 can't be directly reverted now, 'git revert -m 1' shows the following needing attention:
```
Gregs-MBP2:elks greg$ git revert -m 1 728eb890cc293a9505128ca7ddf1eb23e80b367b
error: could not revert 728eb890... Merge pull request #445 from mfld-fr/kern_mem
hint: after resolving the conflicts, mark the corrected paths
hint: with 'git add <paths>' or 'git rm <paths>'
hint: and commit the result with 'git commit'
Gregs-MBP2:elks greg$ git diff
diff --cc elks/arch/i86/boot/setup.S
index 8e1789af,8f7cd9ac..00000000
--- a/elks/arch/i86/boot/setup.S
+++ b/elks/arch/i86/boot/setup.S
@@@ -248,15 -248,8 +248,18 @@@ good_sig
  //;-------------------------------------------------------
  //;-- from here the real loader starts
  //;-------------------------------------------------------
- 
  start_os:
++<<<<<<< HEAD
 +
 +      push    %cs
 +      pop     %ds
 +      mov $hello_mess,%si
 +      call prtstr
 +
 +      mov $INITSEG,%ax
++=======
+       mov     $INITSEG,%ax       //;datasegment setup.S
++>>>>>>> parent of 728eb890... Merge pull request #445 from mfld-fr/kern_mem
        mov     %ax,%ds
  
  //; The root device is obviously /dev/rom0
diff --cc elks/arch/i86/kernel/system.c
index acf06d0a,4418cff0..00000000
--- a/elks/arch/i86/kernel/system.c
+++ b/elks/arch/i86/kernel/system.c
@@@ -31,31 -30,28 +31,39 @@@ void setup_arch(seg_t *start, seg_t *en
   *    Fill in the MM numbers - really ought to be in mm not kernel ?
   */
  
 -/*
 - *    This computes the 640K - _endbss
 - */
 +      /* Extend kernel data segment to maximum of 64K */
 +      /* to make room for local heap */
  
 -#ifndef CONFIG_ARCH_SIBO
 +      /* *start = kernel_ds + (((unsigned int) (_endbss+15)) >> 4); */
 +      *start = kernel_ds + 0x1000;
  
 +#ifndef CONFIG_ARCH_SIBO
      *end = (seg_t)((setupw(0x2a) << 6) - RAM_REDUCE);
 +#else
 +    *end = (basmem)<<6;
 +#endif
  
 -    /* XXX plac: free root ram disk */
 +      /* Now insert local heap at end of kernel data segment */
  
++<<<<<<< HEAD
 +      heap_add (_endbss, 1 + ~ (unsigned) _endbss);
++=======
+ #ifdef REL_SYS
+     *start = kernel_cs + (((unsigned int) (_endtext+15)) >> 4);
+ #else
+     *start = kernel_ds + (((unsigned int) (_endbss+15)) >> 4);
+ #endif
++>>>>>>> parent of 728eb890... Merge pull request #445 from mfld-fr/kern_mem
  
 -#else
 +      /* TEMP: just to demonstrate the heap */
  
 -    *end = (basmem)<<6;
 -    *start = kernel_ds + (unsigned int) 0x1000;
 +      void * h1 = heap_alloc (100);
 +      void * h2 = heap_alloc (100);
 +      heap_free (h1);
 +      h1 = heap_alloc (50);
 +      heap_free (h1);
  
 -#endif
 +      /* Misc */
  
      ROOT_DEV = setupw(0x1fc);
```

tkchia, 2020-03-18:
Hello @ghaerr, @mfld-fr, @Mellvik,

> I would like to know your thoughts on what best to do regarding source tree given this boot failure: revert the commit (#445), continue to commit outstanding PRs regardless, or nothing.

Well, maybe we can just do nothing and keep the PRs open for the time being --- I think I have pinpointed the problem behind the boot failures, and we should be able to get it fixed pretty soon.

Again, [PCjs's super-accurate simulator](https://www.pcjs.org/devices/pcx86/machine/5170/vga/2048kb/debugger/) came to the rescue here.  Here is what happened:
  * First, the bootloader code moved the [Diskette Drive Parameter Table](https://www.delorie.com/djgpp/doc/rbinter/it/64/12.html) to the top 64 KiB of conventional memory, so that it could modify the "sectors per track" value.  So far so good.
  * Next however, the (new) `setup.S` _also_ moved itself to the same place --- the top 64 KiB of conventional memory.  This ended up clobbering the DDPT, by replacing it with `setup.S` code.  In particular, the "bytes per sector" parameter, which should be `0x02` (512 bytes), became `0xb2' (!).

Thank you!

ghaerr, 2020-03-18:
Very nice debugging work @tkchia, I am impressed.

This leaves the question of why it works on QEMU. Does QEMU ignore a non-ROM DDPT, or is it that the memory locations occupied by DDPT and subsequently by setup.S are different on QEMU for other reasons?

tkchia, 2020-03-18:
> This leaves the question of why it works on QEMU. Does QEMU ignore a non-ROM DDPT, or is it that the memory locations occupied by DDPT and subsequently by setup.S are different on QEMU for other reasons?

Thanks @ghaerr.  I am not very sure what was happening on QEMU, but it seems that the SeaBIOS firmware which QEMU uses in its emulation actually ignored the DDPT, or at least parts of it.

Thank you!

Mellvik, 2020-03-18:
FWIW - it's not only QEMU - it works on VirtualBox too.

—Mellvik

ghaerr, 2020-03-18:
> it seems that the SeaBIOS firmware which QEMU uses in its emulation actually ignored the DDPT, or at least parts of it.

I suppose that since most emulators don't actually use CHS to read a floppy or HD image, then they can ignore all or portions of a DDPT as well. This solves the mystery of why the wrong track number that had been in use on some floppy disks prior to my `setboot` patch in #450 always worked on QEMU. QEMU is likely ignoring all CHS for actual disk read/write requests.

If this is correct, it would be worth finding an Linux/OSX emulator that does emulate CHS, for additional testing outside QEMU. ('PCjs' requires an image upload, correct?)

tkchia, 2020-03-18:
Hello @ghaerr,

> If this is correct, it would be worth finding an Linux/OSX emulator that does emulate CHS, for additional testing outside QEMU. ('PCjs' requires an image upload, correct?)

Indeed we should! :-)

Yes, if you are using PCjs directly from the `pcjs.org` web site, you will need to upload the disk image to it.  I have not tried setting up PCjs on a local system, but it should be possible.

Thank you!

mfld-fr, 2020-03-18:
Nice finding @tkchia ! I totally forgot that BIOS table when relocating `setup`, and it was not caught by my non-regression testing under QEMU.

Mellvik, 2020-03-18:
Hi all,
I'm happy to confirm that the latest commit boots fine on the 286 compaq.

@ghaerr, looking forward to your making od() (and time) available. Any thoughts on adding a -X to it, giving a xxd-like hex display instead of a separate hexdump/xxd program (which admittedly would be more in accordance with the unix philosophy).

Sash csh-like history is making progress.

--Mellvik

ghaerr, 2020-03-18:
> Hi all, I'm happy to confirm that the latest commit boots fine on the 286 compaq.

Great news, thanks for testing @Mellvik and thank you @tkchia!

> looking forward to your making od() (and time) available. Any thoughts on adding a -X to it, giving a xxd-like hex display 

Both `od` and `time` are finished, I'll get a followup PR just after committing #473 later today, which mostly fixes other problems you've addressed.

`od` is the complete standard octal dumper, so you can use `od -x 4`, etc like normal. I initially thought about porting a hexdump, but busybox didn't have one. Let me know what you think. I've also got the timezone stuff done, which now allows local time by setting TZ= in your .profile.

closed 2020-03-18 by ghaerr