ghaerr/elks #471: Test report - physical PC

Mellvik, 2020-03-15:
I'm not sure what's the most efficient way to communicate a test report like this, since I'm touching a number of different issues. Opinions solicited. This is a mix of bug reports, observations, enhancement requests & opinions.
HW: 386/20, 1.4M boot floppy, 1.2M (5,25in) fd1, CF hard disk - one DOS/FAT partition

- [sash] Extremely speedy, a pleasure to use @ghae, looking forward to put some command history into it one of these days.
- [sash] `du *` doesn't work (no wildcard expansion)
- [sash] `cp dir/file .` doesn't work ('destination file is a directory')
- [I/O] floppy rd/wr is VERY slow: copy a 150k file locally on the 5,25 in drive is an order of magnitude slower than DOS
- [command wishlist] I'm REALLY missing the `time` command - and `od()` or some other hexdump program. [Both are on my list.]
- [FAT fs] removing files works fine but produces error messages:
```
VFS: iput: trying to free free inode
VFS: device: 0000 inode: 0 mode:0000000
```
- [FAT fs] When floppy write fails because the device is full, the error messages are nonsensical, at least to a regular user (the destination file is saved, should probably be deleted since it's incomplete):
```
# cp sys.com xx
hd: bad partition start=7454 sect=0 n_sects=0
BIOSHD: I/O error
dev 3C0, sector 7454 [comment: possibly dev 3C8, the screen on this machine has serious issues]
sys.com: I/O error
```
- [comment] I find it confusing that floppy-related messages start with HD, not fd ...
- [boot] During boot, the kernel reports `Unknown x86 CPU`. The CPU is genuine Intel 386.
- [floppy] probe is better (faster), and acceptable for now ...

More testing on 286 forthcoming.

--Mellvik

ghaerr, 2020-03-15:
Hi @Mellvik, thanks for your testing, I really appreciate it!

 
> * [sash] Extremely speedy, a pleasure to use @ghae, looking forward to put some command history into it one of these days.

Great! Can you also test it as the standalone shell, that is by editing '/etc/passwd' and specifying /bin/sash for the root login shell? This then completely replaces /bin/sh, unless you use pipes or redirection at the command prompt. I'm interested to hear your response on how fast the system is, and whether it's too slow when running commands requiring pipes or redirection (output to a file, which requires running /bin/sh on top of sash).

> * [sash] `du *` doesn't work (no wildcard expansion)

Sash supports wildcards, so I'll look into this. You must be saying that `du` doesn't support command arguments, I think.

> * [sash] `cp dir/file .` doesn't work ('destination file is a directory')

The 'normal' cp has the same limitation, right? If so, then its just another enhancement request for both `cp`'s.

> * [I/O] floppy rd/wr is VERY slow: copy a 150k file locally on the 5,25 in drive is an order of magnitude slower than DOS

Is this on FAT? Can you test on Minix as well? I'd like more information on the exact
command being used for copying, etc, to see whether this is a file i/o issue, or directory
creation, etc. If Minix is faster, then this shows its a FAT issue. Otherwise, the issue is in the
BIOS disk driver.

> * [command wishlist] I'm REALLY missing the `time` command - and `od()` or some other hexdump program. [Both are on my list.]

I just ported `od` yesterday from busybox, so we'll have it shortly. I want to commit #466 first as I need to change Makefile to add it to miscutils.

> * [FAT fs] removing files works fine but produces error messages:
> ```
> VFS: iput: trying to free free inode
> VFS: device: 0000 inode: 0 mode:0000000
> ```

I've never seen this - this needs to be tested on a known-good floppy, fresh from an MSDOS format. If the disk was used by earlier versions of ELKS FAT, it needs to have  `chkdsk` run on it.

Does this happen every time? I have tested `rm` and seems to work fine.

> * [FAT fs] When floppy write fails because the device is full, the error messages are nonsensical, at least to a regular user (the destination file is saved, should probably be deleted since it's [...]

This is likely a FAT bug, haven't got around to testing FS full issues. Thanks.
> 
> * [comment] I find it confusing that floppy-related messages start with HD, not fd ...

This is because the messages are coming from the BIOS device driver, which is a
combined HD/FD driver that uses BIOS to handle read/write sectors on both.

> * [floppy] probe is better (faster), and acceptable for now ...

That's good to know, but nothing has changed for at least several weeks.

ghaerr, 2020-03-15:
I've duplicated both `du` and `cp` bugs on sash and will fix, thanks.

Mellvik, 2020-03-15:
Thanks for fast feedback (and good news) @ghaerr. More sash testing shortly. In the meanwhile a few clarifications.

> 15. mar. 2020 kl. 16:56 skrev Gregory Haerr <notifications@github.com>:
> 
> [FAT fs] removing files works fine but produces error messages:
> VFS: iput: trying to free free inode
> VFS: device: 0000 inode: 0 mode:0000000
> I've never seen this - this needs to be tested on a known-good floppy, fresh from an MSDOS format. If the disk was used by earlier versions of ELKS FAT, it needs to have chkdsk run on it.
> 
The error is consistent across devices & media, HD partition, 1.2M floppy, 360k floppy, neither of the floppies have been touched by ELKS before. Happens every time. I'll run CHKDSK and test again when I get a bootable fd-image.
> [comment] I find it confusing that floppy-related messages start with HD, not fd ...
> This is because the messages are coming from the BIOS device driver, which is a
> combined HD/FD driver that uses BIOS to handle read/write sectors on both.
> 
I realize that, but I still wanted to flag the issue. Would it be reasonable to have the driver adjust the message (first letter actually) based on (BIOS) device #?
> [floppy] probe is better (faster), and acceptable for now ...
> That's good to know, but nothing has changed for at least several weeks.
> 
It has been several weeks since I used physical machines, and the multiple (5) head traversals are gone. Applaud.

BTW - a new more(1) is coming as soon as I pass my own git-exam.

--Mellvik

ghaerr, 2020-03-16:
Hello @Mellvik,

> [FAT fs] removing files works fine but produces error messages: 

> VFS: iput: trying to free free inode 
> VFS: device: 0000 inode: 0 mode:0000000 

> The error is consistent across devices & media, HD partition, 1.2M floppy, 360k floppy, neither of the floppies have been touched by ELKS before. Happens every time. I'll run CHKDSK and test again when I get a bootable fd-image.

Does this only happen on the second floppy on MSDOS disks of your own, or can you
make this happen booting from `image/fd1440.bin` directly from the build? If it's
just 'rm' on mounted second floppies, please send an image of the floppy over.

Make sure you're running build from 'make clean', but also 'rm .config; make defconfig'
in order to be sure we're looking at the same build. You might also do this for your #470 
boot failure, as .config is not touched by 'make clean'. Also what is the 'ELKS commit' being shown on boot?

Exactly what command are you running directly after boot?
Can you get this to fail in QEMU?

Mellvik, 2020-03-17:
@Ghaerr,

> 16. mar. 2020 kl. 03:22 skrev Gregory Haerr <notifications@github.com>:
> 
> 
> Hello @Mellvik <https://github.com/Mellvik>,
> 
> [FAT fs] removing files works fine but produces error messages:
> 
> VFS: iput: trying to free free inode
> VFS: device: 0000 inode: 0 mode:0000000
> 
> The error is consistent across devices & media, HD partition, 1.2M floppy, 360k floppy, neither of the floppies have been touched by ELKS before. Happens every time. I'll run CHKDSK and test again when I get a bootable fd-image.
> 
> Does this only happen on the second floppy on MSDOS disks of your own, or can you
> make this happen booting from image/fd1440.bin directly from the build? If it's
> just 'rm' on mounted second floppies, please send an image of the floppy over.
> 
I'm not sure I understand the question correctly: I have not tried to make a root FAT disk.
The error is completely repeatable over floppies, floppy drives, hard disks, machines whatever. rm() on any FAT volume. Will test on FAT root when we get the boot problem fixed.
Most recent testing is the commit before a9eecd4e (from my bisecting the boot problem).
> Make sure you're running build from 'make clean', but also 'rm .config; make defconfig'
> in order to be sure we're looking at the same build. You might also do this for your #470 [...]
Which commands don't seem to matter - usually cd, pwd, sash, mount, cd, cp - something like that.
> Can you get this to fail in QEMU?
> 
> 
No, it's not repeatable in QEMU … I guess you need the mount debug messages. I'll add that later.
> 
> 
On slow file copy on floppies: 
This is not a floppy issue, and preliminary tests indicate it's not a FAT issue. File copy is extremely slow, period. VFAT local HD copy is marginally faster than floppy, minix local floppy copy is about the same speed. Again, maybe we should add the time() command to get realtime measurements on this in order to get a metric.

—Mellvik

ghaerr, 2020-03-17:
Hello @Mellvik,

> VFS: iput: trying to free free inode VFS: device: 0000 inode: 0 mode:0000000 

I found and fixed this bug in `rm`, thanks! This and a number of other small command issues have been fixed and a PR submitted shortly, along with other fixes already in the queue.

> No, it's not repeatable in QEMU … I guess you need the mount debug messages. I'll add that later.

I think your 'git bisect' shows the problem area now, thanks.

> On slow file copy on floppies: This is not a floppy issue, and preliminary tests indicate it's not a FAT issue. File copy is extremely slow, period. VFAT local HD copy is marginally faster than floppy, minix local floppy copy is about the same speed. 

This will take a bit more work to debug and fix, as it's unclear whether its the floppy driver or perhaps a timing issue that results in sectors just being missed as they're rotating in the drive, resulting in worse case write speed.

> maybe we should add the time() command to get realtime measurements on this in order to get a metric

Can you find a relatively simple source version?

Mellvik, 2020-03-17:
@Ghaerr,

re. slow copy: FWIW - I don't think it can be a sector timing issue since the symptom is the same on 5.25in, 3.5in floppies and HD … maybe we can experiment with larger reads? I can test with dd bs=<whatever> (unless OS buffering screws it up anyway)…

Time.c - sure, right back home: (4.2BSD)
```
/* time command */

#include <stdio.h>
#include <signal.h>
#include <sys/types.h>
#include <sys/times.h>

extern int errno;
extern char *sys_errlist[];

main(argc, argv)
char **argv;
{
	struct tms buffer, obuffer;
	int status;
	register p;
	time_t before, after;

	if(argc<=1)
		exit(0);
	time(&before);
	p = fork();
	if(p == -1) {
		fprintf(stderr, "Try again.\n");
		exit(1);
	}
	if(p == 0) {
		execvp(argv[1], &argv[1]);
		fprintf(stderr, "%s: %s\n", argv[1], sys_errlist[errno]);
		exit(1);
	}
	signal(SIGINT, SIG_IGN);
	signal(SIGQUIT, SIG_IGN);
	times(&obuffer);
	while(wait(&status) != p)
		times(&obuffer);
	time(&after);
	if((status&0377) != 0)
		fprintf(stderr,"Command terminated abnormally.\n");
	times(&buffer);
	fprintf(stderr,"\n");
	printt("real", (after-before) * 60);
	printt("user", buffer.tms_cutime - obuffer.tms_cutime);
	printt("sys ", buffer.tms_cstime - obuffer.tms_cstime);
	exit(status>>8);
}

char quant[] = { 6, 10, 10, 6, 10, 6, 10, 10, 10 };
char *pad  = "000      ";
char *sep  = "\0\0.\0:\0:\0\0";
char *nsep = "\0\0.\0 \0 \0\0";

printt(s, a)
char *s;
long a;
{
	char digit[9];
	register i;
	char c;
	int nonzero;

	for(i=0; i<9; i++) {
		digit[i] = a % quant[i];
		a /= quant[i];
	}
	fprintf(stderr,s);
	nonzero = 0;
	while(--i>0) {
		c = digit[i]!=0 ? digit[i]+'0':
		    nonzero ? '0':
		    pad[i];
		fprintf(stderr,"%c",c);
		nonzero |= digit[i];
		c = nonzero?sep[i]:nsep[i];
		fprintf(stderr,"%c",c);
	}
	fprintf(stderr,"\n");
}
```
—Mellvik

> 17. mar. 2020 kl. 15:26 skrev Gregory Haerr <notifications@github.com>:
> 
> 
> Hello @Mellvik <https://github.com/Mellvik>,
> 
> VFS: iput: trying to free free inode VFS: device: 0000 inode: 0 mode:0000000
> 
> I found and fixed this bug in rm, thanks! This and a number of other small command issues have been fixed and a PR submitted shortly, along with other fixes already in the queue.
> 
> No, it's not repeatable in QEMU … I guess you need the mount debug messages. I'll add that later.
> 
> I think your 'git bisect' shows the problem area now, thanks.
> 
> On slow file copy on floppies: This is not a floppy issue, and preliminary tests indicate it's not a FAT issue. File copy is extremely slow, period. VFAT local HD copy is marginally faster than floppy, minix local floppy copy is about the same speed.
> 
> This will take a bit more work to debug and fix, as it's unclear whether its the floppy driver or perhaps a timing issue that results in sectors just being missed as they're rotating in the drive, resulting in worse case write speed.
> 
> maybe we should add the time() command to get realtime measurements on this in order to get a metric
> 
> Can you find a relatively simple source version?
> 
> —
> 
>

Mellvik, 2020-03-17:
@ghaerr,
I have a different time.c, from minix - which compiles, but like the BSD version, times(3) is missing from libc. Any thoughts on that?

-Mellvik

ghaerr, 2020-03-17:
@Mellvik: no, I just wrote `times` and almost have `time` ready. It'll be included in the new PR just after we get your boot working.

Mellvik, 2020-03-17:
@ghaerr,

great, looking forward to it.

—M

ghaerr, 2020-03-17:
Also, while I'm in the time code... did you finish the work on TZ processing, or would you like me to do it while in there?

Mellvik, 2020-03-17:
No, I did not, just go ahead!!! Great.

-M

ghaerr, 2020-03-19:
@Mellvik,

 > * [FAT fs] When floppy write fails because the device is full, the error messages are nonsensical, at least to a regular user (the destination file is saved, should probably be deleted since it's incomplete):
 > 
 > ```
 > # cp sys.com xx
 > hd: bad partition start=7454 sect=0 n_sects=0
 > BIOSHD: I/O error
 > dev 3C0, sector 7454 [comment: possibly dev 3C8, the screen on this machine has serious issues]
 > sys.com: I/O error
 > ```

This is the last problem in this open issue. I can't get it to repeat. I'd like you to try this first on the ELKS FAT boot floppy, then try it on an MSDOS native floppy if you can't get it to fail. When I try it on the package-built `fd360-fat.bin`, I get:

[image]

When you get a failure, take a screenshot. I want to see if this is FAT12, 16 or 32, and the disk's exact FAT: configuration displayed at mount time.

Mellvik, 2020-03-19:
Thanks @ghaerr,

more testing (and many disks) shows that 1.2M disks are fine. Further, the hardware is not reliable on 360K disks, neither formatting, nor rd/wr. BTW, as it turns out, it was never a disk full problem. Sidenote: We should put df() on the TODO list.

Thus it is more likely that this is a hardware problem than anything else, so the issue may be closed. When/if more reliable 360K hw becomes available, I'll recheck.

—Mellvik

closed 2020-03-19 by ghaerr