ghaerr/elks #477: pipe/subshell crash

Mellvik, 2020-03-19:
In ash, the following pipe doesn't work, and yields unintelligible output, see attached screen dump:
# tar cf - bin |(cd /mnt/tmp; tar xvfp - )

[image]

ghaerr, 2020-03-19:
Hello @Mellvik,

Boy, you just won't stop with the bug finding, will you?? :)

Since there are many bugs in ELKS cmds, and many possibilities for what this might be (for instance `tar` itself might not be working and/or need more heap space), could I ask you to test this additionally by running the two halves as separate commands with an intermediate file (using file redirection?). This will determine whether it is a shell problem or a tar problem. 

I have found that a few ELKS commands fail because of lack of heap space, and spit out garbage or don't run. An easy temp fix for this is to use the `chmem` command and give it 65530 as the max heap. You can try that on tar.

Also, after trying the above, lets try dropping the `cd` within the subshell and using `tar tvfp` just to get a better feel for where the error is.

Thanks!

Mellvik, 2020-03-19:
@ghaerr,
well, no - there is no end to it ... or at least noe end in sight.
And thanks for 'landing it': Here's the verdict.

- subshells (ash) with multiple chained commands work, even a pipe into such subshell.
- tar doesn't write. And knowing the error message tar emits, it turns out to be embedded in the gobbledygook from my initial tar pipeline: Removing every other letter fra 1 (starting at 0), leaves the original tar message: Writing is not supported. 

Leaving that particular mystery for another day and time, we're left with an enhancement request for tar, nothing else.

--Mellvik

ghaerr, 2020-03-19:
> it turns out to be embedded in the gobbledygook from my initial tar pipeline: Removing every other letter fra 1 (starting at 0), leaves the original tar message: Writing is not supported. Leaving that particular mystery for another day and time

The 'mystery' is two simultaneous `tar` messages being mixed together by ELKS multitasking - 
'Writing is not supported', and 'Unknown tar flag'.

Since almost all of the ELKS commands need enhancements, and there's only you and I enhancing them presently, I would suggest just adding the idea of getting another `tar` from somewhere and porting it if you really want it, and closing this issue. Thanks!

Mellvik, 2020-03-19:
@ghaerr,

I totally agree. And noe that you have taken both /dev/tty, time() and clock/TZ, I'll put tar on my list.
After sash history and more().

Closing #477.

—Mellvik

closed 2020-03-19 by Mellvik

ghaerr, 2020-03-19:
> After sash history and more

The current `more` isn't that bad, I removed the '\<\<stdin>>' junk a couple commits ago. An easy fix that would make it much better would be to set the terminal in raw mode, so that \<space> works rather than requiring \<return>. You can use the `set_raw` and `set_normal` functions I added to `miniterm` to do that, if you want.

Mellvik, 2020-03-19:
Thanks for the hint, @ghaerr - I'll put that on my list.

My primary more()-changes have been to fix tty buffering ('more' appearing after you hit <cr>), cleanup multi-file input and file (non tty) output, in which case the filename is included (compatible with Linux more - and FWIW ancient unmore()).
It has been ready for a couple of weeks, just waiting for git anxiety to pass (kidding - the bisect was a really useful experience).

On my future list for more() is raw mode like you suggest, plus some simple commands, 1G, G, forward search (no regexp), no backward movement (except 1G) to keep it simple.

I'll create PR for it tomorrow, need comments on coding style & more.

—Mellvik