ghaerr/elks #481: Minor config adjustments needed

Mellvik, 2020-03-20:
When CONFIG_SOCKET is off (no networking support),
CONFIG_APP_KTCP should also be off. Otherwise https gets started and the resulting system is like molasses - until https is killed.

Also, elvis does not seem to make it into the final image. vi() is currently broken and not useable.

--Mellvik

Mellvik, 2020-03-20:
@ghaerr,

> 20. mar. 2020 kl. 15:34 skrev Gregory Haerr <notifications@github.com>:
> 
> 
> Otherwise https gets started and the resulting system is like molasses - until https is killed.
> 
> There is still a race condition bug at startup having to do with setting the controlling terminal with setsid which affects signal handling; I ended up adding (back) httpd to slow down the time period between init starting the rc.sys script and starting getty. The race manifests itself by signals (^C) never being generated from the TTY driver, no action when typed for any programs, and haven't been able to figure it out. I was wondering about the effect http might have by running. Thanks for the report, it will have to be removed. (It is likely buggy, its running in the background doing nothing should not affect system speed).
> 
> 
OK, makes sense now.
> At this point, it may be better for the rc.sys script to determine that either networking isn't present in the kernel, or that the ktcp start failed, and not run httpd and telnetd, than to not copy networking apps to the final image. I'll be trying to fix this using that approach first.
> 
> elvis isn't currently compiling, looks like its Makefile needs to be fixed, then it can easily be installed. Want to take a pass at that? It is in the 'Other apps not compiling' section of config.
> 
> 
Yes, I'll put it on my list.
> vi (xvi) works for me, except that its behavior is confusing when using '$' to move to end of line - the cursor goes one column to far, and must be moved back under a character. Can you describe other bugs that make it unusable?
> 
> 
I'm seen many scree/cursor addressing problems such as cursor not returning to 0 (left), moving to the wrong position and the line skew, which I have found unpredictable. Possibly curses bugs for all I know. Do you recommend trying to fix vi rather than spending time on elvis?

—Mellvik
>

ghaerr, 2020-03-20:
> Do you recommend trying to fix vi rather than spending time on elvis?

No. In general, unless small fixes (like `more`) its probably better to find working replacements.
`elvis` probably works well, and is still maintained (I think). I just stuck it in the non-compiling section during elkscmd cleanup. I think its a minor Makefile problem. You can then see how well it works compared to xvi.

ghaerr, 2020-03-22:
@Mellvik, the latest commits should fix the `httpd` problem you were having. If so, please close this PR, and you can submit a PR for a working elvis build when you have it.

closed 2020-03-22 by Mellvik