ghaerr/elks #494: Elvis enable/elvis VFS issue

Mellvik, 2020-03-26:
Attached diff enables elvis in ELKS builds (1440). There are curses/ANSI issues (in process), and a VFS issue that seems to be present in sash() only, see attached screenshot.

Elvis is 'work in progress' - right noe it seems more useable than vi (levee).

--Mellvik
[image]

[elvis.diff.gz](https://github.com/jbruchon/elks/files/4389876/elvis.diff.gz)

ghaerr, 2020-03-26:
@Mellvik, thanks for opening this. I'll commit the two outstanding PRs then look at this.
I must say very strange VFS error.

Why is it that every program you use brings out more bugs in ELKS?! Lol :)

Mellvik, 2020-03-26:
Hmmm, thanks @ghaerr, I keep asking myself the same question - and since I don’t like any of there answers (so far), let’s just keep it open. :-)

BTW - related subject, belonging somewhere else, but let me mention it so it doesn’t fall between the infamous chairs: wc() (on my list) needs a minor update to work in a pipe, just 2 lines to delete.

—Mellvik

ghaerr, 2020-03-26:
> wc() (on my list) needs a minor update to work in a pipe, just 2 lines to delete.

Already ran into this and solved it using `wc -` to read from stdin.

Mellvik, 2020-03-27:
wc():
Yes, that will do it. To me it's stil a bug though - unexpected behaviour from a typical filter. If you don't mind, I'd like to fix it.

--Mellvik

Mellvik, 2020-03-27:
@ghaerr: tested, works great - this is a milestone: having a screen editor on elks that actually works!
--Mellvik

closed 2020-03-27 by Mellvik