ghaerr/elks #501: time cmd not working

Mellvik, 2020-03-28:
Admittedly I haven't tested the new time command until now.

It reports 'Real 0m001s' regardless.

--M

ghaerr, 2020-04-03:
@Mellvik, This should be fixed by #503. Please test, thanks.

Mellvik, 2020-04-04:
Thank you @ghaerr, time now works - at some point I'll get to check it's accuracy.
That gives us (yes, belongs somewhere else, more numbers are coming) - on a 286/1.4M flp:
ls -l /bin: 7 sec.
cd /bin; time cp vi xx --- 28.5 secs (vi is 66k)
It's very slow.

--Mellvik

closed 2020-04-04 by Mellvik

ghaerr, 2020-04-04:
@Mellvik: thanks for testing. That is *very* slow.

I have a debug option in mind that would allow us to see why it takes 7 seconds to read /bin. If you turn on DEBUG_BLK to 1 in elks/include/debug.h and recompile ELKS from scratch, floppy block numbers will be displayed (and likely scrolled off) for disk I/O. That should allow you to see exactly which disk sectors (actually 1k blocks - 2 sectors) are read in order for the 'ls' to complete.

ghaerr, 2020-04-04:
Also @Mellvik is that slow speed on FAT or Minix floppy?

Mellvik, 2020-04-04:
Sorry for not pointing that out @ghaerr,
It's minix.

Numbers for more tests are coming. Hint: Right now Elks is 10x dos, 2x Venix (V7 fs) on physical hard disk copy. (Elks FAT to FAT).

-Mellvik