ghaerr/elks #502: possible bug - access system call

Mellvik, 2020-03-28:
While working on a new rm(), I've run into a problem with access, or at least that's what it looks like. 
Here's a code snippet - notice the segment commented out:
```
if ((buf.st_mode&S_IFMT) == S_IFDIR) {
                if(rflg) {
                        printf("skipping access\n");
                        /*
                        if (access(arg, 02) < 0) {
                                if (fflg==0)
                                        printf("%s not changed\n", arg);
                                errcode++;
                                return;
                        }
                        */
                        if(iflg && level!=0) {
                                printf("remove directory %s? ", arg);
                                if(!yes())
                                        return;
                        }
                        if((dirp = opendir(arg)) == NULL) {
                                printf("rm: cannot read %s?\n", arg);
                                exit(1);
                        }
                        while((dp = readdir(dirp)) != NULL) {
                                printf("inode %ld (%lX), name: %s/%s\n", dp->d_ino, dp->d_ino, arg, dp->d_name);
                                if(dp->d_ino != 0 && !dotname(dp->d_name)) {
                                        sprintf(name, "%s/%s", arg, dp->d_name);
                                        rm(name, fflg, rflg, iflg, level+1);
                                }
                        }
                        closedir(dirp);
                        if (dotname(arg))
                                return;
                        if (iflg) {
                                printf("rm: remove %s? ", arg);
                                if (!yes())
                                        return;
                        }
                        if (rmdir(arg) < 0) {
                                fprintf(stderr, "rm: ");
                                perror(arg);
                                errcode++;
                        }
                        return;
                }
                fprintf(stderr, "rm: %s is a directory\n", arg);
```
With the comment in place, the code works. When the comment is in place, it somehow changes the inode of the directory, which is subsequently busy and not removable by any means (rmdir etc). A reboot and the directory is normal - and removable by rmdir.

What am I missing? Is access() actually touching the inode, not just checking permissions?

--Mellvik

ghaerr, 2020-03-29:
Hello @Mellvik,

A quick look at elks/fs/open.c in sys_access() shows that you probably have found another kernel bug. It seems that 'namei' is called and doesn't release the inode in the case of successful completion (i.e. file exists).

After the call to namei, add 'if (!error) iput(inode);' and see whether that fixes the problem. If so, include that in your patch.

Also, in your code above, I think `rm` should continue processing after printing errors, rather than calling `exit`. There may be file permissions that don't permit removal but the rest of the files/dirs  should be processed. Overall, `rm` should return 1 if errors were present, and 0 otherwise.

Mellvik, 2020-03-30:
@ghaerr,
spot in, that fixed the problem. PR coming.

As to the exit(), it is my take the this is a very special case, thus the exit. If access() is OK, and diropen() fails, there is most likely a FS consistency problem or something else of serious character, better just bail out. All other errors along the way cause just a return.

--Mellvik

closed 2020-03-30 by Mellvik

ghaerr, 2020-03-30:
> As to the exit(), it is my take the this is a very special case, thus the exit. 

True, but the plan is to take these variations of simple commands and have them operate identically in `sash` (for small mods anyways). In those cases, (and I just fixed all others), calling `exit` causes the shell to exit, which isn't wanted. The decision to do this or not is related to the 'coreutils' idea of possibly sharing source between them at some point.

Mellvik, 2020-03-30:
OK @ghaerr,

that's a valid point. My rm() may be somewhat bloated for such use. I'll remove the exit() before the PR.

—Mellvik