ghaerr/elks #504: A couple BIOS HD issues

ghaerr, 2020-03-28:
Hello @tkchia,

During inspection of the MBR partition code, I uncovered an issue (shown in the boot messages) that I find strange. BIOSHD temporarily displays the CHS of hard drives (using the information from setup.S). The attached image shows a confirmed 63,16,63 CHS hard drive, but it is reporting the one cylinder short: 62,16,63. I have confirmed the problem is not in the setup.S code, it seems to be coming from the BIOS. Note that the calculated size of the /dev/bda flat partition is 62496, which is also using the 62 cylinders. This is booting from the `hd32-minix.bin` image, which is definitely being created with 63 cylinders.

Do you suppose this is a QEMU bug (I have not tested on anything else)? Since the geometry is wrong, it appears the QEMU also must be ignoring CHS when performing disk I/O, since everything works. Strange!
[image]

ghaerr, 2020-03-28:
The second issue, visible from the above screenshot, is that the genhd.c partition manager check is finding a partition in ELKS Minix boot sector code erroneously for partition 4. This also occurs with our FAT boot sector. This could possibly be safely ignored, but it would be nice to fix it if possible.

I tried adding code in 'add_partition' in genhd.c to double check the partition size against the whole image size, but found that caused problems with @Mellvik's 100MB MSDOS 6.22 hd image partition recognition, so I've left it out for now. The problem with that image is that the QEMU-reported BIOS information reported for that whole drive is smaller than the size of partition 1. (That may be why it won't boot on QEMU).

In any case, it seems we need to discuss what to do about current MINIX and FAT boot sectors being used on HD images, since ELKS is thinking they have partitions erroneously, and there is little partition signature information that can be reliably used.

We are only two steps away from being able to boot MBR images - an MBR boot is needed, and the VBR boot needs to get the starting sector offset from the partition at ES:SI. Another option would be to always use an MBR boot sector on any ELKS HD images, including flat ones.

tkchia, 2020-03-28:
Hello @ghaerr,

> but it is reporting the one cylinder short: 62,16,63. I have confirmed the problem is not in the `setup.S` code, it seems to be coming from the BIOS.

Interesting.  I just checked Ralf Brown's Interrupt List to see if it has anything to say about this.  And surprisingly, it does:

> > the maximum cylinder number reported in CX is usually two less than the total cylinder count reported in the fixed disk parameter table (see INT 41h,INT 46h) because early hard disks used the last cylinder for testing purposes; however, on some Zenith machines, the maximum cylinder number reportedly is three less than the count in the fixed disk parameter table.

> > for BIOSes which reserve the last cylinder for testing purposes, the cylinder count is automatically decremented

So perhaps QEMU, or more specifically the SeaBIOS firmware image used by QEMU, is simply replicating this hack.

Thank you!

ghaerr, 2020-03-28:
A third issue, also the subject of open issue #405, is what to do about the HW floppy driver (and also ATHD HW IDE driver).

Both drivers are not working. The floppy driver has the entire file '#if 0', because it is way out of date and produces tons of compilation errors.

The ATHD 'directhd.c' driver compiles, but doesn't work for reading any sectors. The ELKS boot sectors and other enhancements are now reliant on the BIOS disk driver, and there isn't enough kernel code space to include both BIOS and ATHD drivers in the kernel, without removing networking, even if it did work.

I make the suggestion that both "Experimental" choices be removed as configuration options, so as to not confuse users, like @ssieb in #405, and leave both `floppy.c` and `directhd.c` out of the build entirely, but leave the source files around just in case someone wants to do something with them at a later time.

ghaerr, 2020-03-28:
> Interesting. I just checked Ralf Brown's Interrupt List to see if it has anything to say about this. And surprisingly, it does

Amazing! Thanks for finding this information.

> So perhaps QEMU, or more specifically the SeaBIOS firmware image used by QEMU, is simply replicating this hack.

This poses a problem: ELKS tools could create and populate an HD image that has data on the last cylinder, causing execution/access issues while running on old BIOS's or QEMU. Should the tools subtract a cylinder during the 'mkfs' process? Currently, `mfs` isn't passed a cylinder count.

tkchia, 2020-03-28:
Hello @ghaerr,

> Should the tools subtract a cylinder during the 'mkfs' process?

I think another possibility would be to tweak the kernel (or `setup.S`) to work around the BIOS hack and deduce the hard drive's true geometry.  Although `int 0x13`, `ah = 0x08` reports only 62 cylinders, if you do a "read" or "verify" operation on cylinder 62, the operation will succeed --- so this may be one way to tell what is going on.  (I have tried this in a QEMU session with FreeDOS in drive A: and an ELKS image in drive C:.)

Thank you!

Mellvik, 2020-03-28:
@ghaerr, @thchia,
coming from the sidelines on this, I’d like to suggest accepting the drive data from the BIOS as bona fide. 

While irrelevant since ~1990, the ‘hidden' cylinders were important in the 80s, not so much for testing as for bad sector remapping. Drives always had bad sectors and the innermost tracks or cylinders were used as spares, transparent to the filesystem, and eventually (ATA, SCSI) transparent to the OS.
The XT MFM drives and the early AT IDE drives (like those in my Compaqs) still do bad sector remapping - so in order to work with ELKS, I suspect we have to keep it that way. 

FWIW, we’re not really lacking HD space …

—Mellvik

ghaerr, 2020-03-28:
> I’d like to suggest accepting the drive data from the BIOS as bona fide.
The XT MFM drives and the early AT IDE drives (like those in my Compaqs) still do bad sector remapping - so in order to work with ELKS, I suspect we have to keep it that way.

You make a good point, and extra determination of the 'real' cylinders isn't really necessary.

However, for "newly created" hard drive images ELKS will generate an improper I/O error when reading or writing almost full (within the last real cylinder), since the filesystem size is greater than the usable size. ELKS tools could decrease the generated filesystem size, but using mounted filesystems not created by ELKS could result in inability to access or change an existing file.

ghaerr, 2020-03-28:
Hello @Mellvik,

> I tried adding code in 'add_partition' in genhd.c to double check the partition size against the whole image size, but found that caused problems with @Mellvik's 100MB MSDOS 6.22 hd image partition [...]

On this other issue, here's the screen shot showing booting from floppy in QEMU with your original MSDOS 6.22 HD image as the second (hard) drive. Note how the CHS of 1023,3,50 multiplies out to match the flat `bda` image size of 153450, but that the first partition, which is MSDOS 6.22, shows a size of 204737, which is reasonable for your 100MB image (512 byte blocks).

I bring this up because I don't know whether your MSDOS 6.22 image is just broken, since it won't boot on QEMU, or whether this is some other problem. I'd like for you to try it on real hardware if possible, and then also see whether you can find a working QEMU-bootable MSDOS 6.22 image and send one over. Thanks!
[image]

Mellvik, 2020-03-28:
@ghaerr,
Makes sense, maybe we should test that. 

I’ll do some testing on real HW and other emulators tomorrow. I was trying to trick a CF drive into faking alternate CHS values a few weeks back (different context) and failed. 

—M

Mellvik, 2020-03-28:
@ghaerr,
I’ll check this one tomorrow too.
Remember that this exact image worked fine with an older QEMU on macOS. I tried to downgrade, but failed .. I’m pretty convinced we can get it to boot with the ‘right’ qemu config.

-M

ghaerr, 2020-03-29:
Hello @Mellvik and @tkchia,

I've found more information regarding the strange CHS values and `bda` size report on the boot display on the MSDOS 6.22 HD image boot above. I ran `fdisk` from ELKS on it and it showed the following partition information:

[image]

It seems QEMU might be using the MBR in order to setup the heads and sector values returned from BIOS, by using the END CHS values on the active partition. (3 heads and 50 sectors would be used to get to the end of active partition in the MBR). QEMU seems to force 1024 cylinders (underreported by 1), and CHS 1023, 3, 50 is reported by BIOS.

But what sort of drive has 3 heads?

If 4 heads are used, then size of `/dev/bda` would be 1024 * 4 * 50 = 204,800. This makes sense with the actual size of the drive which shown as `bda1` starting at sector 63 and 204,737 long = 204,800.

QEMU reports the proper head (16) and sector (63) count for ELKS `hd32-minix.bin` and `hd32-fat.bin`, both of which have no MBR.

I then found that @Mellvik used an older command line (not accepted in my newer QEMU version) of "cyls=203,heads=16,secs=63".  203 * 16 * 63 = 204,624, which is very close to the required minimum 204,800 sectors required. Using 204 cylinders would work (with 832 unused sectors in the last cylinder).

So, what is the real disk geometry for this apparently genuine HD copy of MSDOS 6.22: is it CHS 1024 x 4 x 50, or 204 x 16 x 63?  I suspect it's probably the latter, and QEMU is buggy using the MBR to automatically set CHS, and needs a way to directly specify CHS in the newer version.

Mellvik, 2020-03-29:
@ghaerr, @tkchia -

actually too bad this (qemu problem) is becoming sort of a project in itself.

I’ve spent some time on it this morning, here is the short version:
The image is fine, boots fine on virtualbox
Tha CHS values from the old QEMU command which you refer to, are the correct ones.
QEMU seems broken. One thing is that the ‘traditional’ ability to tell QEMU about the image is gone. Another is that the new ways are nonsensical - or maybe just buggy. If specifying raw format and CHS values, it says that raw cannot be combined with CHS data. Leaving out raw makes it say the opposite, then autoenable raw, and subsequently complain about the combination agin. I’m compiling an older version of QEMU in order to get around this. It seems faster than starting to look in the sources of the most recent  version.
The strange CHS values you get, are not so strange after all. 203-2-50 is the final sector of partition 1 which contains the filesystem (FAT). Without any other info available, fdisk assumes this is the last sector on the drive and guesses that this is the size of the drive. Why DOS ended the partition right here is anyone’s guess. I could have used partition magic to change that, but don’t think it would solve the qemu problem. On the other hand - this may indicate an fdisk (ELKS) bug, fdisk should query the BIOS for the parameters IMHO.
While writing, the compilation completed and the image boots fine with QEMU 2.8.1 (with the previously mentioned command line).

Back to the actual issue, it’s easy to see that the emulators have a hard time producing reasonable results for drive parameter querying. Old klunkers such as my Compaqs have a drive drive parameter table, probably in the bios, or possibly only in a setup program which write the params into NVRAM (aka CMOS). An index and a CHS value + write compensation track, landing track etc. While I believe this table was reasonably standardized at the time, it is not surprising that the EMUs lack such info. Thus not having the ability to specify CHD values (as is the case for QEMU right now) is a dealbreaker. There BIOS call cannot possibly produce reasonable results.
It may be reasonably assumed though that the results from the BIOS call are dependable on physical machines.

I’m getting BIOS disk parameter results from several physical machines separately, re. our discussion on where the drives actually end.

Let me know if you’d like the compiled QEMU 2.8.1.

—Mellvik

ghaerr, 2020-03-29:
Hi @Mellvik,

Thanks for the information, it's good to know the MSDOS image boots fine on VirtualBox and QEMU 2.81. It appears the areas where the QEMU CHS issues affect ELKS is in the disk size and partition size calculations, and the block request driver won't execute I/O requests outside the calculated flat or partition boundary. 

> Without any other info available, fdisk assumes this is the last sector on the drive and guesses that this is the size of the drive.

Actually it is QEMU that is guessing the size from the MBR data. `fdisk` just reads and displays the partition table from the MBR directly, it doesn't do any calculations on its own.

>  this may indicate an fdisk (ELKS) bug, fdisk should query the BIOS for the parameters IMHO.

ghaerr, 2020-03-29:
> this may indicate an fdisk (ELKS) bug, fdisk should query the BIOS for the parameters IMHO.

Actually, `fdisk` does an `ioctl` to ELKS kernel to retrieve the CHS values for the drive. These will be the same CHS values ELKS got (from the BIOS through QEMU) at startup.

I'll consider this first of the three items brought up in this issue closed, as it's not an ELKS bug. Thanks @Mellvik!

closed 2020-03-31 by ghaerr