ghaerr/elks #507: Wild processes and weird behaviour

Mellvik, 2020-03-30:
I ran into this when attempting to run getty on the serial line (from inittab). The approach was apparently not very smart, and getty is broken (on my list), but that's a different story.
What happens is (probably) that getty starts looping.
The first symptom that something is wrong is an ash() error message for all commands: 
`Cannot fork`
ash() does fork however, and the commands work just fine.
The other odd behaviour is from ps - reporting several zombies, garbage command lines and negative process numbers. Further - init (process 1) has an interesting command line 'umped'.

I'm not going to try to analyze this, but figured we have one or more bugs here. The inittab line is
`2:2345:respawn:/bin/getty /dev/ttysS0 9600`
See attached screen dump.

--Mellvik
[image]

ghaerr, 2020-03-30:
Well @Mellvik, leave it to you to try something new and immediately find more bugs!!

There are only 16 process slots and I suspect a loop where processes are being created rapidly and dying in the background. The negative process numbers is just the process number > 32767, and rapidly changing proc struct could account for garbage shown for some processes and lots of zombie states. The 'Cannot fork' indicates no free process slots.

I'm wondering how to duplicate this on QEMU over its virtual serial port, it looks like you're running ELKS on physical hardware and using a terminal emulator on MSDOS 6.22 on VirtualBox to connect to it?

Mellvik, 2020-03-30:
@ghaerr,
yes, that does indeed seem to be the case. :-) And this is not the only one. I needed to get sash to read a startup file (and decided to put  a 'sashrc' in /etc) in order to test a history pager, only to discover that putenv() is fundamentally broken. I'm on it right now.

getty:
I'm using virtualBox for the test, it's easier than QEMU for a lot of things because I don't have to remember all the command line options :-).
That said, I suspect you can repeat this on QEMU by simply connecting the virtual serial line to a file or pipe.

—Mellvik

ghaerr, 2020-03-30:
I wrote a 'putenv' 20 years ago for ELKS. I just removed it from elkscmd/lib thinking that putenv now worked, and, or course, you've found it doesn't!  Here it is.

[putenv.c.zip](https://github.com/jbruchon/elks/files/4403782/putenv.c.zip)

Mellvik, 2020-03-30:
Thanks @ghaerr,

Worked right away! I'm glad I mentioned the problem 'en passant' —

Which brings us to the next problem :-) :: This may well be a know issue though:

Changing files in libc/misc will cause the local out.a to be remade, but it doesn't make it into the build. Neither does a make clean in libc/.
A full top level make clean; make is necessary to get the changes into the image. Really good I was able to move the level environment to the mac (from the raspberrypi 2).

—Mellvik

ghaerr, 2020-03-30:
It's a little tricking with the libc functions, yes. Just replace putenv.c in whole with your patch, or I can do it in a separate PR if you wish. We will want to check behavior of other applications doing a putenv using a grep search for regression testing however.

ghaerr, 2020-03-31:
Got it figured out. `init` is cycling through getty's, the process numbers increment into negative territory, and `fork`, which is working correctly returns < 0 (but not -1). `ash` and too many userland programs only check 'fork() < 0' not -1.

If the /etc/inittab /dev/ttyS0 line is changed to leave out the 9600 baud rate, getty doesn't hang.

Looking at getty source, it's a minor nightmare in itself. I'll fix it up with decent debugging along with a kernel fix to not allocate negative PIDs and you can continue debugging.

PR #509 (2020-03-31): [kernel][cmds] Fix negative PIDs, baud rate on /etc/inittab line
Fixes #507.

Kernel no longer allocates negative process IDs.
/bin/init now properly processes lines containing optional baud rate after tty line.
Heavy cleanup of getty.c to enable console debug messages for debugging.

@Mellvik,
As usual, long story on all your findings. Turns out that `/bin/getty` never even got executed with the optional baud rate added after /dev/ttyS0. `/bin/init` was trying to open "/dev/ttyS0 9600" and failed, and immediately tried again until no process slots. After that was fixed, once getty finally ran [... cut]
closed 2020-03-31 by ghaerr