ghaerr/elks #515: getty/serial test report

Mellvik, 2020-04-03:
Summary: Close, but no cigar
Testing on physical HW & virtualbox, the latter turns out to lend itself well to serial testing: Serial line set up to listen to a TCP port, which can be connected to using netcat.

1. Serial comms work - `cat /etc/rc.d/rc.sys > /dev/ttyS0` does the expected, and read from serial line also works.
2. There is a problem with the serial driver or somewhere in the call chain from open() and down to physical, as we've seen before: `sash < /dev/ttyS0 > /dev/ttyS0 &` should work. When it does, we're almost there.

Mellvik, 2020-04-03:
UPDATE:
Some adjustments to getty, and **serial login is working** via inittab.

Still some minor issues (commands echoed twice), but seemingly - when started from inittab, the device gets opened only once, and we're OK.

Great work, @ghaerr !!

ghaerr, 2020-04-03:
> Some adjustments to getty, and serial login is working via inittab.

Good news! What kind of changes were made to getty?

A few explanations on your statements above:

> The error message is Cannot create /dev/ttyS0: error 16. 

Errno 16 is device busy, /dev/ttyS0 is protected against being opened twice.
Trying to run a shell with both 0 and 1 redirected isn't really the way to do it.

> There seem to be a stdio-problem and possibly a file descriptor inheritance problem in getty. The device in arg[1] is never opened by getty.

Long story short on this is that `/bin/init` opens the argument to `getty`. There is some special processing of /etc/inittab by 'init' and things aren't as simple as they seem. See 'sys_utils/init.c' for details.  We probably shouldn't change init processing at this point.

After the device is opened by init, it execs the inittab-specified program, usually getty, and now getty sets the termio struct as well as the baud rate from the command line (but the device is already opened).

> Starting getty from inittab 'hangs' (loops) the system. 

Is this fixed?

> Opening the device explicitly in getty and then replacing stdio 0,1,2 before the execv using dup2() works. 

`/bin/init` opens the device initially as stated above. I would like to see specifically what was required to get this working. 'init' does the same thing replacing 0,1,2 using dup2.

> I cannot figure out why dup2 is accepted but not the shell redirects.

That's the protection in the serial driver that prohibits multiple opens. 'dup2' just duplicates a file descriptor in the kernel without re-opening the device.

To summarize - the changes in the device open and dup2 calls you've made to get this to work may need to be changed in 'init', not 'getty'.

Mellvik, 2020-04-04:
@ghaerr,
it’s really encouraging being so close to having serial login - and thus full serial support - working.
Final physical machine testing is pending - surely just adjustments.

The testing / debugging has left some - from my perspective - interesting experiences, summarized below. 

> 3. apr. 2020 kl. 17:19 skrev Gregory Haerr <notifications@github.com>:
> 
> 
> Some adjustments to getty, and serial login is working via inittab.
> 
> Good news! What kind of changes were made to getty?
> 
When everything works, your fixed getty works perfectly. My adjustments are proposals, because as is, the serial/getty/inittab combination is impossible to debug: getty cannot be run from the command line (because of the single open policy, more on that below), and when run from init, it loops - unless everything on the hardware side is perfect. More on that too below.
> A few explanations on your statements above:
> 
> The error message is Cannot create /dev/ttyS0: error 16.
> 
> Errno 16 is device busy, /dev/ttyS0 is protected against being opened twice.
> Trying to run a shell with both 0 and 1 redirected isn't really the way to do it.
> 
I don’t agree. The unix way is allowing multiple opens and then requesting exclusivity when required (and if possible). Making serial tty different from this breaks reasonable expectations, breaks with history and complicates things (like this example). Further - for serial lines, multiple opens has been a requirement: Using the same line for inbound and outbound connections for example. Having SLIP ‘block’ getty while active would be such a situation. Also, being able to redirect from the shell like showed, has saved my day too many times to remember, in particular when working on serial lines. In other words, my suggestion is to change this.
> There seem to be a stdio-problem and possibly a file descriptor inheritance problem in getty. The device in arg[1] is never opened by getty.
>
> Long story short on this is that /bin/init opens the [...]
I got this eventually, and indeed this is the traditional unix way, while Linux seem to have deviated with mingetty. Anyway, my proposal for getty is to have it check its origin so to speak: Either use parent process id or compare the name of stdin to argv[1]. That way getty (and the environment around it) becomes debuggable. And it is a minor change.
> Starting getty from inittab 'hangs' (loops) the system.
> 
> Is this fixed?
> 
> 
Yes and no. For obvious reasons, getty is very sensitive to errors. Any open or read/write error (almost) will cause it to exit, making init loop and the system appears dead. There are probably more ways to fix this than I can think of, but I guess the ’normal’ way is to have init detect the looping and throttle. Or somehow make getty more robust...
> Opening the device explicitly in getty and then replacing stdio 0,1,2 before the execv using dup2() works.
>
> /bin/init opens the device initially as stated above. I would like to see specifically [...]
Maybe - I don’t know ELKS init, so I don’t have an opinion on that. Since I now know getty, that way seems simpler ...

Finally - sort of unrelated, but related anyway: elvis won’t run in a terminal window, complains about Screen too small. TERM is OK and /etc/termcap is available. I have that on my list, but if you immediately have a hunch about  the problem, let me know.

—Mellvik

ghaerr, 2020-04-04:
> When everything works, your fixed getty works perfectly. 
unless everything on the hardware side is perfect.

What exactly is not working with my last PR? Is it just that it can't be debugged easily when the hardware side isn't perfect?

> My adjustments are proposals, because as is, the serial/getty/inittab combination is impossible to debug

Agreed. I added a DEBUG option to both init and getty for this reason. Try turning them on (start with one at a time), it will help a lot. That was the only way I was able to see what was actually going on, finally.

> The unix way is allowing multiple opens and then requesting exclusivity when required (and if possible). 

Ok. The tty code could be changed on default to allow duplicate opens (at least for root), and the `ktcp` daemon could use O_EXCL on open to protect itself. I can add both those things.

> while Linux seem to have deviated with mingetty.

I'm not that familiar with mingetty, and ELKS has one too. I'd say we should probably stick with getty for now and just get the few things needed to get this working.

> Since I now know getty, that way seems simpler ...

Is there something that needs to be added to getty? What exactly?

Thanks for your testing, we're almost there!

ghaerr, 2020-04-04:
> elvis won’t run in a terminal window, complains about Screen too small. TERM is OK and /etc/termcap is available. 

What do you mean 'terminal window'? You mean from a /dev/ttyS0 line? Or from ELKS while connected to another system using 'miniterm'?

I suspect the reason may have to do with an `ioctl` to get the window size not being implemented on a non-/dev/ttyX line. Let me know more details and we'll figure it out.

ghaerr, 2020-04-05:
> Finally - sort of unrelated, but related anyway: elvis won’t run in a terminal window, complains about Screen too small. TERM is OK and /etc/termcap is available. I have that on my list, but if you immediately have a hunch about  the problem, let me know.

I spent some time tracking this down... `/bin/init` calls `getty` which calls `login`. Login sets the USER=, SHELL=, HOME= and TERM=ansi environment variables, then execs the shell specified in `/etc/passwd`. There is no other way that `elvis` gets the LINES/COLS. None of SIGWINCH nor the TIOCGWINSZ `ioctl` are implemented in ELKS. If login doesn't run, `elvis` won't work unless you set TERM=ansi explicitly beforehand.

Thus, I think your issue is that `elvis` is being run when the entire init->getty->login->shell sequence didn't happen. When it happens again, run `printenv` and take a look.

Mellvik, 2020-04-05:
@ghaerr,
This is a normal (full) login. The environ is set normally.
Let me know if there is anything specific to test to narrow it down.
Screenshot to be uploaded separately.
I don't think they're related, but FWIW: a stty problem report is coming separately.

--Mellvik

Mellvik, 2020-04-05:
BTW - this is from a Virtualbox connection via netcat. In my tribe language, this is a 'terminal window'. I haven't checked why we have the extra 'init's - coming back to that.
[image]

Mellvik, 2020-04-05:
@ghaerr, just some clarifications:

> When everything works, your fixed getty works perfectly.
> unless everything on the hardware side is perfect.
> 
> What exactly is not working with my last PR? Is it just that it can't be debugged easily when the hardware side isn't perfect?
> 
Getty works fine, there is nothing wrong with the PR. It’s just not helpful.
> My adjustments are proposals, because as is, the serial/getty/inittab combination is impossible to debug
>
> Agreed. I added a DEBUG option to both init and getty for this reason. Try turning them on [...]
Thanks - appreciated. 
> while Linux seem to have deviated with mingetty.
> 
> I'm not that familiar with mingetty, and ELKS has one too. I'd say we should probably stick with getty for now and just get the few things needed to get this working.
> 
I agree.
> Since I now know getty, that way seems simpler ...
> 
> Is there something that needs to be added to getty? What exactly?
> 
As suggested in the previous message, adding the ability to run getty from the command line turns it into a useful tool for debugging serial connections. A simple test of ppid != 1, or maybe just filename(stdin) != argv[1] and then open argv[1] does it. My version of getty does that - wrapped by the DEBUG #if (using the ppid!=0 method).
> Thanks for your testing, we're almost there!
> 
> 
Yes, this is very encouraging. 
> —Mellvik
> 
>

ghaerr, 2020-04-05:
> As suggested in the previous message, adding the ability to run getty from the command line turns it into a useful tool for debugging serial connections. 

Agreed. Please post your modified getty.c here, along with exactly how you want to run it for debugging serial connections and I'll look at it.
We need to be careful as there are two other serial line issues not debugged: the init recycling an elvis not running.

ghaerr, 2020-04-05:
For the elvis issue and your screenshot: Thanks, very strange.

Lets debug the multiple init processes first - this is definitely because an /etc/inittab process died or failed to start and init is re-running it. Please send over a screenshot of your /etc/inittab or post it. You might try running init in DEBUG mode first, to get a better idea of what is failing.

Mellvik, 2020-04-05:
[moved from #523]
@Mellvik:
> I'm not sure this is a problem, but it makes me suspicious.
> Serial login testing means among other things running ps(1) frequently. I'm noticing process numbers increasing fast while I'm not running many processes. Also, most of the time I see extra init processes in the ps list (see screen dump).

@ghaerr: 
>This is most definitely a problem. The reason you don't see anything is that I fixed the process IDs from going negative so the system continues without you noticing. The issue is exactly the same as before - init is waiting on an inittab child to exit, and restarting it... very quickly. I suspect this is your getty line, but don't know what your /etc/inittab looks like.

>I've added debugging to init so this can be traced - set DEBUG 1 in init.c and recompile, that will show the offending line and we can debug from there. There are lots of reasons that a getty line could be failing, I'm wondering whether /dev/ttyS0 is already in use, as that would cause the open to fail.

Mellvik, 2020-04-05:
That's the strange thing: I'm diagnosing this from the serial line login window. If you look closely at the screen clip you can see that there are logins (shells) @ two different terminals. Also, I'm using the distribution inittab unmodified with the exception of the comment removed for ttyS0.

I'll test more with init debug on.
-Mellvik

ghaerr, 2020-04-05:
Beware, 'init' may need to be debugged from the console (/dev/tty1). There is some code in main() that opens DEVTTY (tty1). I don't recommend changing that to run off /dev/ttyS0, as that will cause the open to fail due to multiple opens.

If you think that perhaps /dev/ttyS0 multiple opens being prohibited might be the cause of all this mess, comment out lines 258-259 in elks/arch/i86/drivers/char/serial.c - that will end the problem of serial line multiple opens.

Mellvik, 2020-04-06:
Thanks for the hint @ghaerr. Actually a useful combination - work on the terminal, messages on the console :-)

I'll make the serial.c change to eliminate that possibility.
Right now it looks like the runaway init is on physical HW only, not on VirtualBox (& serial via netcat).

—Mellvik

Mellvik, 2020-04-07:
@ghaerr:
>Please post your modified getty.c here, along with exactly how you want to run it for debugging serial connections and I'll look at it.

Here we go:
```
diff --git a/elkscmd/sys_utils/getty.c b/elkscmd/sys_utils/getty.c
index 4403e7e5..9060b7d5 100644
--- a/elkscmd/sys_utils/getty.c
+++ b/elkscmd/sys_utils/getty.c
@@ -1,5 +1,5 @@
 /*
- * elkscmd/sysutils/getty.c
+ * elkscmd/sys_utils/getty.c
  *
  * Copyright (C) 1998 Alistair Riddoch <ajr@ecs.soton.ac.uk>
  *
@@ -67,7 +67,7 @@
 
 char *	progname;
 char	Buffer[64];
-int	ch, col = 0, fd;
+int	ch, col = 0, fd, tty;
 
 void consolemsg(const char *str, ...)
 {
@@ -214,21 +214,48 @@ int main(int argc, char **argv)
     int n;
     speed_t baud = 0;
     struct termios termios;
+    pid_t ppid;
 
     progname = argv[0];
     signal(SIGTSTP, SIG_IGN);		/* ignore ^Z stop signal*/
 
-    if (argc < 2 || argc > 3) {
+    ppid = getppid();			/* if parent is init, usage() makes no sense */
+    if (ppid > 1 && (argc < 2 || argc > 3)) {
 	consolemsg("Usage: %s device [baudrate]\n", argv[0]);
 	exit(3);
     }
 
-    if (argc == 2) debug("'%s'\n", argv[1]);
-    else if (argc == 3) {
+    if (argc == 2) {
+	debug("info -- arg 2 is '%s'\n", argv[1]);
+    } else if (argc == 3) {
 	baud = atol(argv[2]);
 	debug("'%s' %ld\n", argv[1], baud);
     }
 
+    if (ppid > 1) {
+#if DEBUG
+    	tty = open(argv[1], O_RDWR);
+    	if (tty < 0) {
+		consolemsg("cannot open terminal %s\n", argv[1]);
+		exit(3);
+    	}
+
+    	debug("reopen %s\n", argv[1]);
+    	close(0); close(1); close(2); /* close inherited stdio */
+    	if (dup2(tty, STDIN_FILENO) != STDIN_FILENO ||  \
+		dup2(tty, STDOUT_FILENO) != STDOUT_FILENO ||  \
+		dup2(tty, STDERR_FILENO) != STDERR_FILENO) {
+		consolemsg("Cannot reopen stdio - error %d\n", errno);
+		exit(3);
+	}
+#else
+	fputs("Enable DEBUG to start getty from the command line\n", stderr);
+    	/*close(tty);		crashes getty */
+	exit(1);	
+
+#endif /* DEBUG */
+    }
+
     fd = open(ISSUE, O_RDONLY);
     if (fd >= 0) {
 	put('\n');
@@ -303,7 +330,10 @@ int main(int argc, char **argv)
 			    state(Host);
 			    break;
 			case 'L':			/* Line used */
-			    if (argc > 1) {
+			    if (argc > 1) {	/* this test is bull since 
+						 * we won't get here unless
+						 * argc > 1 
+						 */
 				ptr = rindex(argv[1],'/');
 				if (ptr == NULL)
 				    ptr = argv[1];
@@ -347,6 +377,7 @@ int main(int argc, char **argv)
 
     /* setup tty termios state*/
     baud = convert_baudrate(baud);
+    debug("setting baudrate %d\n", baud);
     if (tcgetattr(STDIN_FILENO, &termios) >= 0) {
         termios.c_lflag |= ISIG | ICANON | ECHO | ECHOE | ECHONL;
         termios.c_lflag &= ~(IEXTEN | ECHOK | NOFLSH);
@@ -369,7 +400,7 @@ int main(int argc, char **argv)
 	state("login: ");
 	n=read(STDIN_FILENO,Buffer,sizeof(Buffer)-1);
 	if (n < 1) {
-	    debug("read fail errno %d\n", errno);
+	    debug("read fail on stdin, errno %d\n", errno);
 	    if (errno != -EINTR)
 		exit(1);
 	    continue;
@@ -378,6 +409,7 @@ int main(int argc, char **argv)
 	while (n > 0)
 	    if (Buffer[--n] < ' ')
 		Buffer[n] = '\0';
+	debug("Calling login: %s\n", Buffer);
 	if (*Buffer) {
 	    char *nargv[3];
 
```

--Mellvik

ghaerr, 2020-04-07:
Thanks @Mellvik. 

Other than the debug statements to see what is going on, these mods are only to get getty to run from the shell, correct? 

No modifications are needed to run on ttyS0 from init?

Mellvik, 2020-04-07:
Yes, that's right @ghaerr.

There is also a message if started from the command line with DEBUG off, informing that DEBUG needs too be set in order to run like that.

—Mellvik

ghaerr, 2020-04-07:
You want the additional capability to run from the shell permanent, right?

I'll test it a bit, and create a PR for it. There's a couple small issues, and DEBUG shouldn't be required to run it from the shell, if it issues appropriate error messages, agree?

Mellvik, 2020-04-07:
Yes, I agree.
I misread your previous mail.
I can make that change if you'd like.

—M

ghaerr, 2020-04-07:
I'll put up a PR. There's still some bugs in your version, for instance, even though 'usage' is pointless when called from init, 'exit' still has be called. I want to review all the debug messages also, so that when we have to debug this again, it can be used!

I'm not completely happy with the debug messages from 'init' yet, but we're making progress. A system or console log would be nice, possibly combined with a kernel printk log using 'mesg', but we'll leave that for later.

Mellvik, 2020-04-07:
OK. And yes, a printk would be nice at some point.

—M

Mellvik, 2020-04-07:
New serial related issue, @ghaerr:
`ls -l /bin|more` works fine on the console, not via the serial line.
After the first page (first `--More--`), more(1) fails:
`more: : Bad file number `

Repeatable on physical and virtual (Virtuabox).

Buffering issue?

--Mellvik

ghaerr, 2020-04-07:
> After the first page (first --More--), more(1) fails:

IIRC, you added an explicit open of `/dev/tty` to more, and my implementation of that could be buggy. Try reading from stdin or 0 instead and see whether that helps. I'm not sure the open of /dev/tty should occur in more anyways, that's reserved normally for input that absolutely should not be redirected, like typing in passwords.

ghaerr, 2020-04-07:
> you added an explicit open of /dev/tty to more

Opening /dev/tty is definitely the problem. It is failing and there is no check for -1. Try removing that extra open and using fd '1' instead and test it on serial and console. We can't use 0 as I suggested in my last comment, as more may be taking input from a pipe. But if the output is redirected, it shouldn't paginate anyways. You may need to remove the error printf in that case. After testing that case and the normal use case on console and serial, submit a PR, thanks!

There is also my bug that /dev/tty doesn't work on serial, but I can't fix that until I can figure out how to get serial working on QEMU. I also can't figure out how to attach an ELKS image to VirtualBox nor serial on it either.

Mellvik, 2020-04-07:
@Ghaerr,

Yes, you're right, open /dev/tty does fail - but works in other cases. I'll take a look at the logic as well.

—M

ghaerr, 2020-04-08:
I added #526 that should allows getty to run from the shell.

Are there any other serial login problems other than elvis not running at this point? I'd like to close this and start new issues so things don't get too complicated. I have elvis on my list for when I can duplicate it.

Mellvik, 2020-04-08:
Only stty - which I'm working on right now.

We don't need to keep this issue open for that.

—Mellvik

ghaerr, 2020-04-08:
> We don't need to keep this issue open for that.

Ok - what is the difference between what you're working on for that and your separate stty issue #522?

ghaerr, 2020-04-08:
Oops - misread your comment. Go ahead and close this and we can keep #522 open, thanks.

ghaerr, 2020-04-09:
@Mellvik,

Now that I'm able to run ELKS from serial console and use it a bit, I've finally seen how the serial port is buggy (as referenced in #454 by @pawosm-arm). When regularly typing commands into the shell, I notice that on occasion it appears my keyboard misses a keystroke. What's really happening is that the serial driver is dropping an input character. Have you experienced this?

I don't know where the problem is yet, but this happens quite often when typing.

Mellvik, 2020-04-09:
That's great news, @ghaerr.
The answer is no, I have not experienced character loss at all. Not on virtualbox, not on physical. I'm running physical at 19200. 
That said, I have not pushed any limits. By experience, 19200 is the highest the klunkers can go w/o hw handshake,  but I'm assuming you're on virtual....

Typing never looses any input... (is there another process killing the system?)

-M

ghaerr, 2020-04-09:
> It sounds to me you have a qemu problem!!

I don't think this is a qemu problem. Yes, it could be another process is eating keystrokes, but this has never happened before, and it doesn't happen when typing the first character, only when typing quickly and only occasionally. You may not notice it if you don't type quickly, and at first, I thought I had not hit the keyboard hard enough, and just retyped the character.

I'm debugging all of it now, I think it's an interrupt servicing issue that also is the reason for #454 losing serial mouse data. We'll see.

Turns out `/bin/init` has a huge bug while parsing /etc/inittab - when a getty line happens to cross a 256-byte boundary in /etc/inittab, the bug shows up as firing off 3 serial getty's at the same time. It's verified and has been happening the whole time, which is why `ps` still showed processes running occasionally - these were failed getty's because the serial port couldn't be opened twice. I'm fixing all of that too.

I've got most of it all working, including the /dev/tty bug. But running `vi` prints screen size too small and crashes the system. Very strange!

ghaerr, 2020-04-10:
@Mellvik,

PR #532 fixes all issues brought up here. Please test, hopefully we can close this finally!

ghaerr, 2020-04-11:
Hello @Mellvik and @georgp24,

I just tested running the shell on the serial port (ttyS0) after the 'linenoise' enhancements were added with PR #536, and the shell prompt no longer displays on a new line when Enter is typed.
So, you'll need to set LINENOISE to 0 in ash/shell.h until this is debugged. I haven't had time to look at the behavior on /dev/tty1, will fully review but may not have time this weekend.

ghaerr, 2020-04-15:
Hi @Mellvik and @pawosm-arm,

Just after finishing the latest serial/tty enhancement in PR #553, I realized a big reason ELKS can't keep up with serial input - when in "raw" mode, `read` will never return the asked-for byte count, but instead always returns just a single byte (or nothing, for when there is no data). This means that the `mouse` program can't ever catch up to moving the mouse around a lot, nor can `cat` fill its 4096-byte buffer each read. Both `cat` and `mouse` (no pun intended, really) always get a single byte regardless of what they ask for. Thus, it is no surprise that the 512-byte kernel tty input queues drop data, as it is taking forever to get the received data from kernel to user programs.

I am still very interested in testing the new driver for "overrun" kernel messages, as that indicates the problem that the interrupt service routine can't empty the UART fast enough into the kernel input queue.

I hope to have another PR that correctly implements `read` shortly, which should increase throughput heavily.

Mellvik, 2020-04-15:
@ghaerr, FWIW - vi() has been working fine on serial since the 'screen too small' problem was solved.

--M

ghaerr, 2020-04-15:
> vi() has been working fine on serial since the 'screen too small' problem was solved.

Sounds like you're running a 25x80 emulation. PR #553 allows querying of larger window sizes; e.g. 60 x 90 window on using macOS Terminal. If the terminal emulator doesn't support the screen size (DSR) request, it reverts back to guessing 25x80.

Mellvik, 2020-04-15:
I do (25x80) - and thanks for the improvement.

—M

Mellvik, 2020-04-16:
init() related observations:

1. If a serial line is enabled in inittab and the device does not exist, init loops, seemingly continuously trying to open that line. There are no messages on the console. (Physical machine and VirtualBox).
2. If a serial line is enabled in inittab, and does exist, but is not available (example: VirtualBox, serial is opening a socket and waiting for a connect, from, say, netcat), init hangs. 

--Mellvik

Mellvik, 2020-04-16:
While continuing serial testing (which has involved some hardware debugging recently), I propose the following changes:
In irqtab.S - it may make sense to allocate irq5 to COM3 if the XT HD is not present:
```
--- a/elks/arch/i86/kernel/irqtab.S
+++ b/elks/arch/i86/kernel/irqtab.S
@@ -27,8 +27,12 @@ ds_kernel:
 #else
 #define MCD 0
 #endif
-#ifdef CONFIG_CHAR_DEV_RS
-#define MRS 0x18
+#if defined(CONFIG_CHAR_DEV_RS)
+# if defined(CONFIG_BLK_DEV_HD)
+# define MRS 0x18
+# else
+# define MRS 0x38 /* add 3rd COM line interrupt */
+# endif
 #else
 #define MRS 0
 #endif
@@ -127,8 +131,8 @@ _irq4:                      // COM1
        call    _irqit
        .byte   4
 #endif
-#ifdef CONFIG_BLK_DEV_HD
-_irq5:                 // XT HD
+#if defined(CONFIG_BLK_DEV_HD) || defined(CONFIG_CHAR_DEV_RS)
+_irq5:                 // XT HD or COM 3
        call    _irqit
        .byte   5
 #endif
```

Also, in serial.c the chip detection logic from wikibooks has a bug that prevents it for recognizing the 16550. It's worth noting that the 16550 chip has a FIFO, but its functionally unstable and cannot be used. So, IIR bit 7 indicates the presence of a FIFO, bit 6 indicates that the FIFO is useable (16550A and up), bit 5 means 16570.
I propose the following diff to take care of this:
```
--- a/elks/arch/i86/drivers/char/serial.c
+++ b/elks/arch/i86/drivers/char/serial.c
@@ -62,7 +62,6 @@ static struct serial_info ports[NR_SERIAL] = {
 static char irq_port[NR_SERIAL] = { 3, 1, 0, 2 };
 
 static unsigned int divisors[] = {
-    0,                         /*  0 = B0      */
     2304,                      /*  1 = B50     */
     1536,                      /*  2 = B75     */
     1047,                      /*  3 = B110    */
@@ -115,12 +114,14 @@ static int rs_probe(register struct serial_info *sp)
 
     /* then read FIFO status*/
     status = inb_p(sp->io + UART_IIR);
-    if (status & 0x40) {
-       if (status & 0x80) {            /* FIFO enabled*/
-           if (status & 0x20)          /* 64 byte FIFO enabled*/
-               type = ST_16750;
-           else type = ST_16550A;
-       } else type = ST_16550;
+    //printk("stat 0x%02x ", status);
+    if (status & 0x80) {               /* FIFO available*/
+       if (status & 0x20)              /* 64 byte FIFO enabled*/
+           type = ST_16750;
+       else if (status & 0x40)         /* 16 byte FIFO OK */
+           type = ST_16550A;
+        else 
+           type = ST_16550;            /* Non-functional FIFO */
     } else {
        /* no FIFO, try writing arbitrary value to scratch reg*/
        outb_p(0x2A, sp->io + UART_SCR);
@@ -128,6 +129,7 @@ static int rs_probe(register struct serial_info *sp)
            type = ST_16450;
        else type = ST_8250;
     }
+    //printk("\n");
     sp->flags = SERF_EXIST | type;
 
     /*
@@ -367,7 +369,7 @@ static int rs_open(struct tty *tty)
 #define UART_FCR_ENABLE_FIFO4  (UART_FCR_ENABLE_FIFO | 0x40)
     /* enable FIFO with 8 byte trigger */
 #if FIFO
-    if ((port->flags & SERF_TYPE) > ST_16450)
+    if ((port->flags & SERF_TYPE) > ST_16550)
        outb_p(UART_FCR_ENABLE_FIFO8, port->io + UART_FCR);
 #endif
```
Finally, the marking on the chips may be 16550 even if it's really a 16550A, which makes the chip detection logic more crucial.

--Mellvik

pawosm-arm, 2020-04-16:
> In irqtab.S - it may make sense to allocate irq5 to COM3 if the XT HD is not present:

Nah, IRQ lines is a precious resource on every XT. And even if XT-IDE I'm using inside of both of my XT's does not claim any IRQ line, my Turbo-XT has all of them occupied as such:
IRQ 0 - timer
IRQ 1 - kbd
IRQ 2 - network card
IRQ 3 - 8250 COM2:
IRQ 4 - 8250 COM1:
IRQ 5 - SoundBlaster card
IRQ 6 - FDD
IRQ 7 - LPT1:
Fortunately still, Paradise VGA also does not claim any IRQ.

For me, having hardware flow control is now the main feature missing in serial port driver.

Mellvik, 2020-04-16:
@pawosm-arm,

the proposed change doesn't in any way interfere with your requirements. It just changes the default config of IRQ5 from (effectively) 'unused' to 'something'. That 'something' is - incidentally - in line with your priorities: serial flow control. 

-M

pawosm-arm, 2020-04-16:
That only shows how rigid current hardware specification options are. I guess a kind of source ports configuration table in kernel config (similar to the one that can be filled for floppy disk drives when BIOS autodetect is disabled) would be more beneficial.

Mellvik, 2020-04-16:
Which brings us right back to the discussion in #372.

ghaerr, 2020-04-16:
> In irqtab.S - it may make sense to allocate irq5 to COM3 if the XT HD is not present:

@Mellvik - is that the change you had to make to get COM3 to work? And did the solve the problem of getting that card to work with your original COM1 card?

BTW, "XT hard disk support", known now as "Experimental HW IDE hard disk support" is entirely broken. What I don't know is whether the BIOS uses IRQ 5 when asked to do hard disk I/O.

> a kind of source ports configuration table specifying per-serial-port I/O ports and IRQs in kernel config would be more beneficial.

Agreed that since COM3, XT HD, and possibly future sound card may use IRQ 5, the irqtab.S modification shouldn't assume COM3 IRQ 5 when CONFIG_CHAR_DEV_RS is set. I'm thinking that a new config option specifying where to allocate IRQ 5  would be a better way to go.

Is there a need to reallocate the other IRQs in @pawosm-arm's list? Allowing a general re-allocation of any IRQ could be a lot of work, it seems only IRQ 5 is used for multiple devices generally.

Mellvik, 2020-04-16:
@ghaerr,

> 16. apr. 2020 kl. 17:55 skrev Gregory Haerr <notifications@github.com>:
> 
> 
> In irqtab.S - it may make sense to allocate irq5 to COM3 if the XT HD is not present:
> 
> @Mellvik <https://github.com/Mellvik> - is that the change you had to make to get COM3 to work? And did the solve the problem of getting that card to work with your original COM1 card?
> 
> 
Yes, it did (I had a system board conflict too, long story). Now 3 serial ports run fine except for the known performance issues.
> BTW, "XT hard disk support", known now as "Experimental HW IDE hard disk support" is entirely broken. What I don't know is whether the BIOS uses IRQ 5 when asked to do hard disk I/O.
> 
> 
I'm experimenting in order to determine this right now. The system feels slower when IRQ 5 is reallocated, but the clunkers are so slow anyway, that it's hard to determine. More on that later.
> a kind of source ports configuration table specifying per-serial-port I/O ports and IRQs in kernel config would be more beneficial.
>
> Agreed that since COM3, XT HD, and possibly future sound card may [...]
Agree. There are many more or less esoteric uses for IRQ 5 and others that may be accommodated that way. 
> Is there a need to reallocate the other IRQs in @pawosm-arm <https://github.com/pawosm-arm>'s list? Allowing a general re-allocation of any IRQ could be a lot of work, it seems only IRQ 5 is used for multiple devices generally.
> 
> 
>

ghaerr, 2020-04-16:
> the chip detection logic from wikibooks has a bug that prevents it for recognizing the 16550. It's worth noting that the 16550 chip has a FIFO, but its functionally unstable and cannot be used. 

Interesting. Can you point me to other information I can read about this?

>  So, IIR bit 7 indicates the presence of a FIFO, bit 6 indicates that the FIFO is useable (16550A and up), bit 5 means 16570.

Did you test this change, or are both just working from theory here? It seems the wikibooks article was written by a knowledgeable guy, and he used a tricky method, rather than the normal bit settings, for his detection algorithm. 

Are we certain that the 16550 FIFO doesn't work? That's easily changed by testing for "> ST_16550", but we have yet to show that turning on the FIFO makes any difference. Correct?

What happens if you enable the FIFO on the 16550, anything bad? Presently, no serial driver code depends on the proper recognition after just enabling the FIFO, so I don't know how much this matters.

ghaerr, 2020-04-16:
> If a serial line is enabled in inittab and the device does not exist, init loops

Fixing this has been on my list for awhile. I'll take a pass at fixing it by having `init` not respawn again if it fails to open the device before running `getty`. 

> If a serial line is enabled in inittab, and does exist, but is not available (example: VirtualBox, serial is opening a socket and waiting for a connect, from, say, netcat), init hangs.

This is trickier, since there is no ELKS protection from serial ports used by multiple processes, although that may not be the problem. We could make the same change as above, to not respawn if the line open fails in init. I've never seen init hang, only loop repeatedly. You'll have to test with VB, I don't know where exactly init is hanging.

Finally, a good reason for init opening the device prior to calling getty! This would basically happen once at system startup, and allow an /etc/inittab for both tty1 and ttyS0 (and ttyS1)?

Mellvik, 2020-04-16:
I agree, the wikibooks article is great, he's just somewhat unclear on this issue (in addition to the bug in the code which is an easy oversight). He says 'FIFO enabled but not functioning' - this is the 16550. Knowing this, and following the logic in the code, you see that it will never hit the 16550, because he's testing on bit 6, not bit 7.

Here's a quote from the 16550 manual that came with one of my serial cards:
"The primary difference between these two parts (16550, 16550A) is in the operation of the FIFOs. The NS16550 will sometimes transfer extra characters when the CPU reads the RX FIFO. Due to the asynchronous nature of this failure, there is no workaround and the NS16550 should NOT be used in FIFO mode. The NS16550A has no problems operating in the FIFO mode and should be used on all new designs. 
The programmer should not the difference in the function of bit 6 in the Interrupt Identification Register (IIR). This bit is permanently at logical 0 in the NS16550. In the 16550A this bit will be set to 1 when the FIFOs are enabled."

The serial card I am using currently has two chips marked 16550, the identify (per the above) as 16550A. I have yet to test the operation of the FIFO, it's on the agenda.

—Mellvik

ghaerr, 2020-04-16:
@Mellvik - Thanks for the interesting explanation. Lets plan on using your modified patch for the chip detection algorithm and when to enable the FIFO. I say 'modified' because I notice that the 'B0' entry was deleted in divisors[] and that will definitely break the baud rate setting code.

> The serial card I am using currently has two chips marked 16550, the identify (per the above) as 16550A. I have yet to test the operation of the FIFO, it's on the agenda.

After fully testing the chip detection and FIFO, please submit a PR, thanks!

pawosm-arm, 2020-04-16:
> Is there a need to reallocate the other IRQs in @pawosm-arm's list? Allowing a general re-allocation of any IRQ could be a lot of work, it seems only IRQ 5 is used for multiple devices generally.

I guess IRQ 2 has similar story of being used for multiple devices.

ghaerr, 2020-04-16:
> I guess IRQ 2 has similar story of being used for multiple devices.

I'll come up with something for IRQ 5 as well as IRQ 2. Lets move this IRQ discussion and comments over to #372 as there's lots going on around serial ports here.

Mellvik, 2020-04-17:
@ghaerr -
Sorry about the  `B0`, an experiment got interrupted (pun intended).

-M

Mellvik, 2020-04-21:
@ghaerr,
more serial testing today. IMHO the chip detection code is solid and FIFO can be enabled. I've tested threshold levels 8 and 14, both work fine.
```
+    if ((port->flags & SERF_TYPE) > ST_16550)
+       outb_p(UART_FCR_ENABLE_FIFO14, port->io + UART_FCR);
```
I've made two more changes along the way, neither have any effect on the character dropping, but fixes nevertheless:
```
@@ -473,7 +481,7 @@ int rs_init(void)
     do {
        if (sp->tty != NULL) {
            printk("ttyS%d at 0x%x, irq %d is a%s\n", ttyno,
-                      sp->io, sp->irq, serial_type[sp->flags & 0x3]);
+                      sp->io, sp->irq, serial_type[sp->flags & SERF_TYPE]);
        }
```
```
 /* Flush input fifo */
+// resetting FIFO mode clears all FIFOs
+#define flush_input_fifo(pt) outb_p(0, pt->io + UART_FCR)
+#if 0
 static void flush_input_fifo(register struct serial_info *sp)
 {
     int i = MAX_RX_BUFFER_SIZE;
@@ -96,8 +100,9 @@ static void flush_input_fifo(register struct serial_info *sp)
        inb_p(sp->io + UART_RX);
     } while (--i && (inb_p(sp->io + UART_LSR) & UART_LSR_DR));
 }
+#endif
```
The docs clearly states that setting FCR bit 0, clears the FIFOs.

I'm still working on the character drop problem - we're getting there. A few of interesting tidbits from the trenches: 
- Other system activities (such as disk/floppy I/O) are likely part og the problem, but input (paste) into `cat > /dev/null` and `cat > /tmp/xx` give similar results (significant character loss). 
- There are no more data overrun messages to the console, even when most of the input data get lost.
- On the other hand, pre-running (buffering) drastically improve serial performance (using ash, see below): Pasting into `stty -echo; cat > xx; stty echo` gives 50% character loss @ first run, almost 0 @ second run (12 bytes lost out of 12k). That's on a 16450, no fifo, @ 19200bps, probably as good as it gets w/o handshake.
- FIFO works, but there is no improvement (I changed the local buffer to 16).
- sash screwed me for a while (embarrassing) with it's handling of redirections and pipes: `cat /etc/inittab > /dev/null` takes about 19 seconds to execute (the `time`command reports 0,8 secs btw). This is on a 386/20.
- sash seems to take significantly longer to exec commands than ash. This is on my list for further verification.
- In fact dropping `sash` for `ash` decreased serial line character losses significantly (10% in general), which doesn't make much sense. This is the most recent version of ash btw, w/linenoise.
- Like before, the line speed (19200 and below) does affect character loss, but not much. 
- There is no significant difference between the NEW and OLD irq code.

Conclusion: IMHO the serial code is as good as it gets until we do a rewrite using a fully interrupt driven model, which is required to implement HW flow control anyway.

-M

Mellvik, 2020-04-21:
@ghaerr,
I just discovered (from the NS application note for the 16550a): the command to enable FIFO must have bit 0 set. Missed that one, need to do more testing in the morning, but it doesn't change the conclusion.
M

ghaerr, 2020-04-21:
> I just discovered (from the NS application note for the 16550a): the command to enable FIFO must have bit 0 set.

It does set bit 0: `#define UART_FCR_ENABLE_FIFO8   (UART_FCR_ENABLE_FIFO | 0x80)`.

> IMHO the chip detection code is solid and FIFO can be enabled. I've tested threshold levels 8 and 14, both work fine.

Can you upload a patch of serial.c with the changes? Thanks! I will sort through it and add the FIFO detection code, the changed FIFO enable code, and look at your other changes. I'm not yet sure about replacing the flush input fifo code, since that code is run with chips without FIFOs as well.

> There are no more data overrun messages to the console, even when most of the input data get lost.

This is very big news! That means that the kernel interrupt routine has the capability to keep up with input data when FIFOs are enabled. Very good news. I need to explain the problem of overall throughput, and why we are still a ways away from improving that, which I will do below.

> There is no significant difference between the NEW and OLD irq code.

Good: it was the intent to change nothing when moving the IRQs to be configurable. There is still more work to be done for NE2K driver, but otherwise it is now lots easier to configure different hardware. (I assume you've noticed we had to turn off serial COM3 IRQ5 by default, it also was default OFF beforehand as well).

> FIFO works, but there is no improvement (I changed the local buffer to 16).

There is improvement by not seeing data overruns, correct? You won't be seeing any full hardware-to-cat-file improvements, those are a different issue entirely.

> Like before, the line speed (19200 and below) does affect character loss, but not much.

When we say "character loss" here, I assume you mean that "cat /dev/ttyS0 > file" loses characters. More important is whether there are any data overruns at 19200. I will explain why there is non-overrun "character loss" shortly. So we need to keep the two problems separate.

> sash screwed me for a while (embarrassing) with it's handling of redirections and pipes: cat /etc/inittab > /dev/null takes about 19 seconds to execute (the timecommand reports 0,8 secs btw). This is on a 386/20.

I'll bet you don't realize that `sash` doesn't support file redirection? I made it work behind the scenes and `sash` execs `sh` whenever one types a command that uses '>', '<', '|', '&' etc. So heck yeah it's going to be very slow starting up. `sash` is supposed to be small, and does not support much. I felt it better to achieve seamless compatibility between `sash` and `sh` than to rewrite `sash` to become `sh`, since we don't really need two shells!

> sash seems to take significantly longer to exec commands than ash. This is on my list for further verification.

`sash` will run its long list of internal commands without hitting the disk at all (e.g. ls, cp, chown, etc, see elkscmd/APPS for details). You can run the /bin version of the command by typing '/bin/ls', for instance. The actual `exec` of an external command should be quite quick, except for when seamless integration kicks in and `sh` needs to load. In that case, ELKS still has to read in the `sh` data segment, the code segment is shared, so its a little quicker than starting `sh` all on its own. I hope to have a buffer info program sometime in the future that will show buffer usage, and that may allow `sh` to be loaded from RAM instead of disk.

> (buffering) drastically improve serial performance (using ash, see below): Pasting into stty -echo; cat > xx; stty echo gives 50% character loss @ first run, almost 0 @ second run (12 bytes lost out of 12k). That's on a 16450, no fifo, @ 19200bps, probably as good as it gets w/o handshake.

Sounds like things are improving, and my last PR allowed for reading more characters from the 512-byte TTY buffer into `cat`s 4096 buffer with each read, it sounds like that is working. 

Here is a quick overview of how serial data gets from hardware to disk. Keep in mind that each of these, except the interrupt processing, is subject to being rescheduled from the CPU and some other task running. And the faster the baud rate, the more time is spent in the interrupt routine rather than running programs:

- serial interrupt occurs when FIFO trigger level of serial characters received. This can be set to 1, 4, 8 or 14. If the FIFO is set to > 1, then FIFO trigger, or 2 characters of serial framing timeout must occur before an interrupt is generated. ELKS interrupt routine reads up to 10 characters from FIFO into the 512-byte TTY character queue. If for any reason a program is not reading the TTY queue fast enough, then it will drop characters when the interrupt routine force-adds more.
- A program issues a `read` call on the serial port, with a buffer length (4096 for cat). What happens next depends on the many variables settable using `termios` on the serial tty.
- The TTY driver will suspend a `read` and reschedule the processor until various conditions are met, OR, if some met, will return less than the number of characters asked for. In the version before my last PR, at most ONE character was EVER returned from the TTY driver. In the new version after the PR, it uses VMIN and VTIME to determine (in somewhat complex fashion), how many characters to return. The old default for VMIN was 0, which meant that it basically polled the serial driver, and possibly returned 1 character, no matter how many were ready.
[... cut ...]
Options to play with would be setting VMIN to 256 or more and VTIME to 1, but would be best done in a special `cat` program that also did the `stty` calls directly.

Another thought is creating a mechanism to allocate the TTY queue buffer dynamically, and set it much higher, if data transfers need to happen at hight rates and the kernel throughput can't be achieved. But the kernel throughput needs to be improved as well. 

The `ktcp` and SLIP processing routines have max packet sizes, and I believe these are all smaller than 512 bytes.

Mellvik, 2020-04-22:
Thanks for this thorough walkthru, @ghaerr. It has been a while since I did low level serial communication software and device drivers. And yes, we're getting there.

Some clarifications below.

> I just discovered (from the NS application note for the 16550a): the command to enable FIFO must have bit 0 set.
> 
> It does set bit 0: #define UART_FCR_ENABLE_FIFO8 (UART_FCR_ENABLE_FIFO | 0x80).
> 
Sorry, I misread that one!
> IMHO the chip detection code is solid and FIFO can be enabled. I've tested threshold levels 8 and 14, both work fine.
> 
> Can you upload a patch of serial.c with the changes? Thanks! I will sort through it and add the FIFO detection code, the changed FIFO enable code, and look at your other changes.
> 
Will do. I suspect my version is quite a bit behind yours.
> I'm not yet sure about replacing the flush input fifo code, since that code is run with chips without FIFOs as well.
> 
I don't think this is an issue. When FIFO is not enabled the 'buffer' is emptied by a single inb().
> There are no more data overrun messages to the console, even when most of the input data get lost.
>
> This is very big news! That means that the kernel interrupt routine has the capability to keep up [...]
To me this is bad news, although I suspect when it comes down to it, we're in agreement.
> There is no significant difference between the NEW and OLD irq code.
>
> Good: it was the intent to change nothing when moving the IRQs to be configurable. There is still more work to be done for NE2K [...]
I have, this is good. I believe - in order to save memory - it's may be good idea to make the number of serial lines a config option. More than 2 ports is rare, and the freed buffers and data structures may be better used somewhere else.
> FIFO works, but there is no improvement (I changed the local buffer to 16).
>
> There is improvement by not seeing data overruns, correct? You won't be seeing any full hardware-to-cat-file [...]
Yes, I'm talking about lost data...
> sash screwed me for a while (embarrassing) with it's handling of redirections and pipes: cat /etc/inittab > /dev/null takes about 19 seconds to execute (the timecommand reports 0,8 secs btw). This [...]
I am well aware of the characteristics of sash and their motivation, which I completely support. The reason I'm reporting this is that the effects seem out of scale, and (to me) some of them are hard to explain. Also it seems like the time command is broken, reporting 0.8s real time when real time is more like 20s. 
Actually, some times it feels like the system just stops dead in the tracks for - say - 5+ seconds, then continues. And this happens only when using sash. Doesn't make sense. Also, it doesn't make sense to me that sash takes longer to exec a random (external) command than ash. Sidenote: I'm doing most of the testing from a different room than the physical machine - continuously banging the serial line, so I don't hear floppy accesses. Thus the need for more verification.
> (buffering) drastically improve serial performance (using ash, see below): Pasting into stty -echo; cat > xx; stty echo gives 50% character loss @ first run, almost 0 @ second run (12 bytes lost out [...]
Thanks again for the detailed walkthrough! If I read you correctly, I disagree with your conclusion. From my perspective, cat is a perfect test because it tests the entire system. Further, a system that loses data w/o flagging it, is fundamentally broken. The way I see it, we have two problems that need to be fixed:
1) A broken (unreliable, possibly too complicated?) data path between the input (character) buffers and the client program, maybe even the other way (output) too.
2) A low level driver that doesn't support flow control and doesn't report (and record) errors (such as overruns, framing errors, …)
From my perspective, if a system is getting more than it can handle, the only place data loss is acceptable, is at the ingress point, where it can be observed and handled. The rest of the path must be 100% reliable all the time.
> Options to play with would be setting VMIN to 256 or more and VTIME to 1, but would be best done in a special cat program that also did the stty calls directly.
>
> Another thought is creating a [...]
With header compression (CSLIP), the packet size slightly less than 1K, I'm unsure whether it can be 'tuned' much lower …

Back to the two main problems above:
I suggest we share the 'burden'. Among a million other things it seems you are in the process of fixing the path inside the system already. I couldn't do that in any reasonable time. But I can rewrite parts of the driver, making it fully interrupt driven w/ hard and soft flow control. And I have the setup to test it.
> 
—Mellvik

Mellvik, 2020-04-22:
Diff from the test version of serial.c:
```
diff --git a/elks/arch/i86/drivers/char/serial.c b/elks/arch/i86/drivers/char/serial.c
index 054c7775..7797c13e 100644
--- a/elks/arch/i86/drivers/char/serial.c
+++ b/elks/arch/i86/drivers/char/serial.c
@@ -12,7 +12,8 @@
 #include <arch/io.h>
 
 #define NEW            1       /* set =0 for old driver code if needed*/
-#define FIFO           0       /* set =1 to enable new FIFO code for testing*/
+#define FIFO           1       /* set =1 to enable new FIFO code for testing*/
+#define NEWPROBE       1       /* Use new probe code */
 
 #if defined (CONFIG_CHAR_DEV_RS) || defined (CONFIG_CONSOLE_SERIAL)
 
@@ -62,7 +63,7 @@ static struct serial_info ports[NR_SERIAL] = {
 static char irq_port[NR_SERIAL] = { 3, 1, 0, 2 };
 
 static unsigned int divisors[] = {
-    0,                         /*  0 = B0      */
+    0,                         /*  0 = B0      */
     2304,                      /*  1 = B50     */
     1536,                      /*  2 = B75     */
     1047,                      /*  3 = B110    */
@@ -88,6 +89,9 @@ static unsigned int divisors[] = {
 #define        RS_IALLMOSTEMPTY        (    INQ_SIZE / 4)
 
 /* Flush input fifo */
+// resetting FIFO mode clears all FIFOs
+#define flush_input_fifo(pt) outb_p(0, pt->io + UART_FCR)
+#if 0
 static void flush_input_fifo(register struct serial_info *sp)
 {
     int i = MAX_RX_BUFFER_SIZE;
@@ -96,8 +100,9 @@ static void flush_input_fifo(register struct serial_info *sp)
        inb_p(sp->io + UART_RX);
     } while (--i && (inb_p(sp->io + UART_LSR) & UART_LSR_DR));
 }
+#endif
 
-#if NEW
+#if NEWPROBE
 static int rs_probe(register struct serial_info *sp)
 {
     int status, type;
@@ -115,12 +120,13 @@ static int rs_probe(register struct serial_info *sp)
 
     /* then read FIFO status*/
     status = inb_p(sp->io + UART_IIR);
-    if (status & 0x40) {
-       if (status & 0x80) {            /* FIFO enabled*/
-           if (status & 0x20)          /* 64 byte FIFO enabled*/
-               type = ST_16750;
-           else type = ST_16550A;
-       } else type = ST_16550;
+    if (status & 0x80) {               /* FIFO available*/
+       if (status & 0x20)              /* 64 byte FIFO enabled*/
+           type = ST_16750;
+       else if (status & 0x40)         /* 16 byte FIFO OK */
+           type = ST_16550A;
+        else 
+           type = ST_16550;            /* Non-functional FIFO */
     } else {
        /* no FIFO, try writing arbitrary value to scratch reg*/
        outb_p(0x2A, sp->io + UART_SCR);
@@ -227,8 +233,6 @@ static void update_port(register struct serial_info *port)
 
     //FIXME: update lcr from parity and word termios values
 
-    clr_irq();
-
     /* Set the divisor latch bit */
     outb_p(port->lcr | UART_LCR_DLAB, port->io + UART_LCR);
 
@@ -238,8 +242,6 @@ static void update_port(register struct serial_info *port)
 
     /* Clear the divisor latch bit */
     outb_p(port->lcr, port->io + UART_LCR);
-
-    set_irq();
 }
 
 /* WARNING: Polling write function */
@@ -265,7 +267,7 @@ void rs_irq(int irq, struct pt_regs *regs, void *dev_id)
     struct ch_queue *q;
     int i, j, status;
     char *io;
-    unsigned char buf[10];
+    unsigned char buf[16];
 
     i = 0;
     sp = &ports[(int)irq_port[irq - 2]];
@@ -278,7 +280,7 @@ void rs_irq(int irq, struct pt_regs *regs, void *dev_id)
        //if (status & UART_LSR_DR)                     /* Receiver buffer full? */
            do {
                buf[i++] = inb_p(io + UART_RX);         /* Read received data */
-           } while ((inb_p(io + UART_LSR) & UART_LSR_DR) && i < 10);
+           } while ((inb_p(io + UART_LSR) & UART_LSR_DR) && i < 16);
        //}
     //} while (!(inb_p(io + UART_IIR) & UART_IIR_NO_INT) && i < 10);
 
@@ -367,7 +369,7 @@ static int rs_open(struct tty *tty)
 #define UART_FCR_ENABLE_FIFO4  (UART_FCR_ENABLE_FIFO | 0x40)
     /* enable FIFO with 8 byte trigger */
 #if FIFO
-    if ((port->flags & SERF_TYPE) > ST_16450)
+    if ((port->flags & SERF_TYPE) > ST_16550)
        outb_p(UART_FCR_ENABLE_FIFO8, port->io + UART_FCR);
 #endif
 
@@ -473,7 +475,7 @@ int rs_init(void)
     do {
        if (sp->tty != NULL) {
            printk("ttyS%d at 0x%x, irq %d is a%s\n", ttyno,
-                      sp->io, sp->irq, serial_type[sp->flags & 0x3]);
+                      sp->io, sp->irq, serial_type[sp->flags & SERF_TYPE]);
        }
        sp++;
     } while (++ttyno < NR_SERIAL);

```

ghaerr, 2020-04-22:
Thanks for the diff. I'll update serial.c and submit a PR to synchronize where we're both at.

> It's may be good idea to make the number of serial lines a config option. More than 2 ports is rare, and the freed buffers and data structures may be better used somewhere else.

Ok, good idea. This may be best done for the time being in ports.h, since that file would also need modification to properly allocate COM3/COM4. I'll look at it.

> From my perspective, if a system is getting more than it can handle, the only place data loss is acceptable, is at the ingress point, where it can be observed and handled. The rest of the path must be 100% reliable all the time.

All UNIX and Linux systems drop data in this way, when/if the TTY queue buffer overflows. There isn't anything the system can do about it reasonably, since this is at interrupt time.

> With header compression (CSLIP), the packet size slightly less than 1K, I'm unsure whether it can be 'tuned' much lower …

I had thought that SLIP packets were 256 bytes, but looking at `ktcp`, I see they that slip MTU is 1064 bytes. Given this, and your reasonable comment that the system should be 100% reliable, I propose adding variable size TTY queues. That would allow a slow system with a FIFO chip to be 'tuned' so that the TTY queue takes up the slack between the interrupts and complete data throughput. This will allow you to fix the problem, providing the data rate doesn't exceed the hardware interrupt processing overhead.

> A low level driver that doesn't support flow control and doesn't report (and record) errors (such as overruns, framing errors, …)

I'm not sure what the point would be of recording overruns, we report them now. Once a single report is seen, the baud rate must be dropped since there's nothing else ELKS/we can do to help other than using a larger FIFO. I plan on setting the largest FIFO trigger in the next PR, BTW.

> But I can rewrite parts of the driver, making it fully interrupt driven w/ hard and soft flow control. And I have the setup to test it.

Great. I wouldn't suggest "fully interrupt driven" at this point, since that's just going to add to the interrupt overhead. I'll submit a PR shortly, then go ahead and add hardware flow control. Enabling interrupts for the MCR register (modem status interrupt) only would be a good start, and the interrupt routine can manipulate a flag in the serial_info struct that `rs_write` can use to continue transmitting or sleep using `wake_up` etc. To keep things simple, just use the 'CRTSCTS' bit in termios.c_cflag to enable HW flow control processing, which can be set by `miniterm` etc after you get it working.

> Actually, some times it feels like the system just stops dead in the tracks for - say - 5+ seconds, then continues. And this happens only when using sash. Doesn't make sense. 

I have found the memory corruption error I've been working on for two days. Turns out any program that calls `grgetnam` to get a group name badly corrupted memory. This includes the `chgrp` command, and `ls -l`. Since `sash` has an internal `ls`, running `ls -l` corrupted sash memory. That could be the problem, we'll have to see. 

> Also it seems like the time command is broken, reporting 0.8s real time when real time is more like 20s.

Keep testing. Lets keep time or sash bugs in another issue, please open if you find more information, after my PR memory fix.

[13 more replies]

closed 2020-07-19 by Mellvik