ghaerr/elks #516: ELKS on FAT root filesystem now seems to hang inside /bin/init

tkchia, 2020-04-04:
With the latest commit, booting ELKS on a FAT filesystem seems to hang at startup:
```
FAT: 1440k, fat12 format
VFS: Mounted root (msdos filesystem).
```
Logs from QEMU suggest that `/bin/init` is running, but is not producing any output for some reason.

Removing `/bin/init` from the ELKS image gives a different hang:
```
FAT: 1440k, fat12 format
VFS: Mounted root (msdos filesystem).
Unable to open /dev/tty1 (error 2)
No init - running /bin/sh
```
Thank you!

ghaerr, 2020-04-04:
Hmmm, not getting that to duplicate over here. Make sure to rm .config; make defconfig.

> Unable to open /dev/tty1 (error 2)

This sounds like the /dev FAT magic may not be running, it requires /dev to have been created within the first two root entries. Can you please send me the output of 'minfo' on that image (unsorted dir listing)?

Are you creating the images using "cd image; make images" or setting the extra images in the build?

Can you send me a screenshot? I just rebuilt from scratch and booted from both 'fd1440-fat.bin' and 'hd32mbr-fat.bin' boot and run (see attached).

[image]

ghaerr, 2020-04-04:
Hello @tkchia,

The problem is surely the FAT /dev not enabling, thus neither `init` nor `sh`
can output anything. FYI, look at elks/fs/msdos/inode.c, line 170.
If you are using your own FAT disk, make sure it has a /dev entry in the rootdir,
and increase the search count on line 170 to test. This can be relaxed, but doing so
would render files within preexisting /dev directories inaccessible, and also increase
mount disk I/O if extended beyond 15.

tkchia, 2020-04-05:
Hello @ghaerr,

The command sequence I used was
```
rm -f .config && make defconfig && make menuconfig && make clean && make
```

It seems the glitch was because `CONFIG_IMG_DEV` (`image/config.in`) was not enabled for the FAT filesystem output when I did the `make menuconfig`.  This caused the makefile in `image/` not to create the `/dev/` directory at all.  If I fix that, then the FAT image does work.

I will submit a PR soon.

Thank you!

closed 2020-04-05 by tkchia