ghaerr/elks #522: stty problems

Mellvik, 2020-04-05:
While testing serial login, stty gets a real rundown. Most isses are repeatable on the console, some may have nothing to do with stty, originating further down the chain.

- `stty sane` screws up, anything typed echoes as question marks. Possibly changes character length or parity into something non sensible.
- `stty erase <some char>` works, but has consequences
-- no visible delete (echoe still set)
-- Backspace (^H) still deletes visibly, but not actually
-- When erase is back to ^H, visual delete is OK again, and stops at the beginning of the line.
- [Enhancement request] Literal next character
- [minor - and this may be a shell issue] deleting invisible characters on the command line cause the cursor backspaces beyond the beginning of the line.
- The line kill character has no effect

ghaerr, 2020-04-05:
> stty sane screws up, anything typed echoes as question marks.

Try `stty -raw` or `stty cooked` (same thing). This should solve your problem.

Not sure why stty has both cooked and sane options... I can fix 'sane' to be the same as cooked but perhaps we should do more research on it, as to why they're different in the first place.

All the other problems you mention are issues with the tty driver (of which serial is a part of),
and require extensive coding. I'm not sure we need any of these other options, do we?

ghaerr, 2020-04-05:
Update: I've fixed 'sane' mode as well as stty erase <somechar>. I'll put up a PR.

> -- Backspace (^H) still deletes visibly, but not actually

That's normal, since ^H isn't a delete character, but still echoed, including past the beginning of the line.

> When erase is back to ^H, visual delete is OK again, and stops at the beginning of the line.

That is now the behavior of the fixed stty erase, but with any character set as the erase character.

> [minor - and this may be a shell issue] deleting invisible characters on the command line cause the cursor backspaces beyond the beginning of the line.

This will likely remain, as the tty driver can't make assumptions about what might be displayed or not on the [remote, serial, network, etc] display. The tty driver is now fixed for printable erase characters, but other than that assumes a non-erase character is printable.

> The line kill character has no effect

Not sure its worth implementing this.

> [Enhancement request] Literal next character

Not sure where this might be implemented, likely a shell enhancement not in the tty driver.

Mellvik, 2020-04-06:
> Update: I've fixed 'sane' mode as well as stty erase . I'll put up a PR.

Thanks @ghaerr,

this helps a lot. There are still a few tweaks to be made in stty and possibly the serial driver. If it's ok with you, I'd like to experiment with the combination, implement the lnext character, and propose the changes in a PR.

Related, but really a different issue: sash has an (minor) INTR problem. If you type ^C at the prompt, you get a new prompt as expected. If you type a couple of chars at the prompt, then ^C, you get a newline, nothing else. Type a newline and the shell tries to execute whatever was typed before the INTR. The symptom is entirely repeatable across virtual, physical, serial, console.

—Mellvik

ghaerr, 2020-04-06:
>  If you type a couple of chars at the prompt, then ^C, you get a newline, nothing else. Type a newline and the shell tries to execute whatever was typed before the INTR.

`sash` catches SIGINTR and prints a newline in the signal callback function. That function would have to also be modified to reset the input buffer, if that's the expected behavior. Is this what `ash` does?

> I'd like to experiment with the combination, implement the lnext character,

Looks like `stty` already implements lnext, so a mod to the tty driver will be all that should be needed. Remember to check that the line isn't in raw mode so that the slip drivers don't break. What is the typical intended use of lnext?

Mellvik, 2020-04-06:
@ghaerr,

> 6. apr. 2020 kl. 16:42 skrev Gregory Haerr <notifications@github.com>:
> 
> ﻿
> If you type a couple of chars at the prompt, then ^C, you get a newline, nothing else. Type a newline and the shell tries to execute whatever was typed before the INTR.
> 
> sash catches SIGINTR and prints a newline in the signal callback function. That function would have to also be modified to reset the input buffer, if that's the expected behavior. Is this what ash does?
> 
Yes, ash and - my experience - other unix/linux shells. The thing is, with linekill nonfunctional, which is not unusual these days, INTR is the new linekill.
> I'd like to experiment with the combination, implement the lnext character,
>
> Looks like stty already implements lnext, so a mod to the tty driver will be all that should be needed. Remember to check [...]
I guess most regular users couldn't care less, but lnext is invaluable when you need to know what is really coming down the line (and switching to raw is just too time consuming for that particular situation), and in those rare cases - script or whatever, you really need to input 0x3 (binary) or 0x9 (^I) to a program (you can do incredible things with lnext and regexp). Another debugging tool for the command line/serial world. Needless to say, 35+ years later I'm still using my ^V (lnext) weekly. International keyboard input is complicated. And FWIW, I'm not in development.
BTW, I do not plan to do anything with SLIP in the foreseeable future. Someone has to need it first...

-- Mellvik

ghaerr, 2020-04-06:
Sounds good. Take a pass at enhancing the tty driver for lnext.

>  I do not plan to do anything with SLIP in the foreseeable future. Someone has to need it first...

Yes. I was just saying that when adding lnext to ntty.c make sure that you check that the line is not in raw mode before processing the lnext character, since `ktcp` currently supports TCP/IP over serial.

closed 2020-04-15 by ghaerr