ghaerr/elks #523: Runaway init?

Mellvik, 2020-04-05:
I'm not sure this is a problem, but it makes me suspicious.
Serial login testing means among other things running ps(1) frequently. I'm noticing process numbers increasing fast while I'm not running many processes. Also, most of the time I see extra init processes in the ps list (see screen dump). [Ehnancement request to ps: Sorted by process id, reporting real tty if it exists.]

Serial and console are both logged in, mostly idle, normal response most of the time. No filesystems mounted except root (floppy, minix).

--Mellvik

ghaerr, 2020-04-05:
> I'm not sure this is a problem, but it makes me suspicious.
Serial login testing means among other things running ps(1) frequently. I'm noticing process numbers increasing fast while I'm not running many processes. Also, most of the time I see extra init processes in the ps list (see screen dump).

This is most definitely a problem. The reason you don't see anything is that I fixed the process IDs from going negative so the system continues without you noticing. The issue is exactly the same as before - `init` is waiting on an inittab child to exit, and restarting it... very quickly. I suspect this is your getty line, but don't know what your /etc/inittab looks like.

I've added debugging to init so this can be traced - set DEBUG 1 in init.c and recompile, that will show the offending line and we can debug from there. There are lots of reasons that a getty line could be failing, I'm wondering whether /dev/ttyS0 is already in use, as that would cause the open to fail.

ghaerr, 2020-04-05:
@Mellvik, this problem is directly related to your open serial line issue #515. It would be easier to close this and keep the discussion in one place, thanks!

Mellvik, 2020-04-05:
Moving to #515

closed 2020-04-05 by Mellvik