ghaerr/elks #527: Command history in the shell?

georgp24, 2020-04-08:
Bash keeps a command history in a buffer which you can use by hitting the Up and Down arrow keys. Is this disabled in ELKS or just not available?

Mellvik, 2020-04-08:
Hi @georgp24,
yes, wouldn't that be great? It's doable but not really in line with the purpose of sash: small, efficient, simple. 

'classic' csh history, which is half way there in sash right now, works great when you get used to it. Actually, for many purposes it's a lot more capable than arrow key history. Some of us grew up with it, and cannot live without it.
Check it out - maybe you like it. Hint: start off with `alias h history` and expand the history list to say 40-50 lines `setenv HISTORY 40` in /root/.sashrc.

—Mellvik

ghaerr, 2020-04-08:
> Bash keeps a command history in a buffer which you can use by hitting the Up and Down arrow keys. Is this disabled in ELKS or just not available?

ELKS is running an early version of `ash`, not `bash`, and command history is not implemented.
You're welcome to try porting a version of bash to ELKS, but I suspect it will be far too large to fit in 64k.

closed 2020-04-10 by ghaerr